freifunk-gluon/gluon #1064: Gluon 2016.2.3: TL-WA801NDv3 Region Code required

oszilloskop, 2017-03-09:
We got an issue with Gluon 2016.2.3 (factory image) in combination with a new TL-WA801NDv3 router.

The factory firmware (version unknown) has canceled the update flash process with following error message: "Error code: 18005 (file was incorrect)"

Unfortunately, no further information is available.

Adorfer, 2017-03-09:
probably another ETSI300328-routerlockdown. 

is there an "(EU)" on the boilerplate?

oszilloskop, 2017-03-09:
I got following information update:
The router works fine with a Gluon 2016.2.3 factory image flashed via TFTP.

NeoRaider, 2017-03-10:
I've extracted the necessary data from the stock firmware images and will push an update soon.

NeoRaider, 2017-03-10:
There is a fix in my LEDE staging tree at https://git.lede-project.org/?p=lede/neoraider/staging.git;a=summary now, it would be great if someone could test that this yields images flashable from both EU and US firmwares.

NeoRaider, 2017-05-30:
Still nobody who can check this? Would it help if I provide a test image?

rotanid, 2018-02-03:
closing as there hasnt been any feedback.

closed 2018-02-03 by rotanid