freifunk-gluon/gluon #1069: PoE Passthrough option falsely available Ubnt Loco M2 (XM)

dersch81, 2017-03-13:
v2016.2.3-2-g242e6361

[image]

NeoRaider, 2017-03-14:
This is likely the result of using the wrong image: using the normal NanoStation image on the Loco will mostly work, but it will make the system think it has two Ethernet ports.

After a NanoStation image was installed once, you need to upgrade to the correct image (ubiquiti-nanostation-loco-m2) *without keeping the configuration* to get rid of incorrect settings.

dersch81, 2017-03-14:
I've checked it several times but really took the loco m2 FW of my community. I attach it for double checking but also will talk to our firmware builder.

[gluon-ffda-0.9.4-ubiquiti-nanostation-loco-m2.zip](https://github.com/freifunk-gluon/gluon/files/840585/gluon-ffda-0.9.4-ubiquiti-nanostation-loco-m2.zip)

NeoRaider, 2017-03-14:
Please post the contents of `/tmp/sysconfig/board_name` and `/etc/config/system`.

dersch81, 2017-03-14:
Ok i will do. But it takes a couple of days until the node is reachable for me

dersch81, 2017-03-21:
Here you go:

`cat /etc/config/system

config system
        option timezone 'CET-1CEST,M3.5.0,M10.5.0/3'
        option hostname '64283-in-vagranti-nord-west-ladida'

config timeserver 'ntp'
        option enabled '1'
        option enable_server '0'
        list server 'ntp.services.ffda.io'

config gpio_switch 'gpio_switch_poe_passthrough'
        option name 'PoE Passthrough'
        option gpio_pin '8'
        option value '0'

config led 'led_rssilow'
        option name 'RSSILOW'
        option sysfs 'ubnt:red:link1'
        option trigger 'rssi'
        option iface 'rssid_wlan0'
        option minq '1'
        option maxq '100'
        option offset '0'
        option factor '13'

config led 'led_rssimediumlow'
        option name 'RSSIMEDIUMLOW'
        option sysfs 'ubnt:orange:link2'
        option trigger 'rssi'
        option iface 'rssid_wlan0'
        option minq '26'
        option maxq '100'
        option offset '-25'
        option factor '13'

config led 'led_rssimediumhigh'
        option name 'RSSIMEDIUMHIGH'
        option sysfs 'ubnt:green:link3'
        option trigger 'rssi'
        option iface 'rssid_wlan0'
        option minq '51'
        option maxq '100'
        option offset '-50'
        option factor '13'

config led 'led_rssihigh'
        option name 'RSSIHIGH'
        option sysfs 'ubnt:green:link4'
        option trigger 'rssi'
        option iface 'rssid_wlan0'
        option minq '76'
        option maxq '100'
        option offset '-75'
        option factor '13'
`

`/tmp/sysinfo# cat board_name
bullet-m
`

NeoRaider, 2017-03-23:
The `gpio_switch_poe_passthrough` section is created by the script `/etc/board.d/03_gpio_switches`, which only adds it when the board name is *nanostation-m*. Even though you are using the correct image now, you must have had installed the NanoStation one at least once in the past. Reinstalling the firmware **without keeping your configuration** will solve the issue.

closed 2017-03-23 by NeoRaider