freifunk-gluon/gluon #1076: dir860l b1: mesh is degrading with increasing uptime

mweinelt, 2017-03-19:
The mesh connectivity on the D-Link DIR860L B1 is, at least on 2.4 GHz, degrading with increased uptime.

In this cycle it lost mesh connectivity after 6d12h.

```
BusyBox v1.25.1 () built-in shell (ash)

     _________
    /        /\      _    ___ ___  ___
   /  LE    /  \    | |  | __|   \| __|
  /    DE  /    \   | |__| _|| |) | _|
 /________/  LE  \  |____|___|___/|___|                      lede-project.org
 \        \   DE /
  \    LE  \    /  -----------------------------------------------------------
   \  DE    \  /    Reboot (17.01-SNAPSHOT, r3276+11-4a405ac8f9)
    \________\/    -----------------------------------------------------------

root@64289-rocket-cloud:/# dmesg
[    0.000000] Linux version 4.4.52 (hexa@build.ffda.io) (gcc version 5.4.0 (LEDE GCC 5.4.0 r3276+11-4a405ac8f9) ) #0 SMP Sat Mar 11 05:15:18 2017
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is D-Link DIR-860L B1
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 08000000 @ 00000000 (usable)
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] On node 0 totalpages: 32768
[    0.000000] free_area_init_node: node 0, pgdat 804890e0, node_mem_map 81000000
[    0.000000]   Normal zone: 256 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 32768 pages, LIFO batch:7
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] PERCPU: Embedded 10 pages/cpu @8110c000 s8416 r8192 d24352 u40960
[    0.000000] pcpu-alloc: s8416 r8192 d24352 u40960 alloc=10*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 32512
[    0.000000] Kernel command line: console=ttyS0,57600 rootfstype=squashfs,jffs2
[    0.000000] PID hash table entries: 512 (order: -1, 2048 bytes)
[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes)
[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Writing ErrCtl register=000100d0
[    0.000000] Readback ErrCtl register=000100d0
[    0.000000] Memory: 124440K/131072K available (3711K kernel code, 204K rwdata, 856K rodata, 208K init, 256K bss, 6632K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] NR_IRQS:256
[    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
[    0.000000] sched_clock: 32 bits at 100 Hz, resolution 10000000ns, wraps every 21474836475000000ns
[    0.010000] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
[    0.070000] pid_max: default: 32768 minimum: 301
[    0.070000] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.080000] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.090000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.090000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.090000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.090000] CPU1 revision is: 0001992f (MIPS 1004Kc)
[    0.180000] Synchronize counters for CPU 1: done.
[    0.190000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.190000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.190000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[... 449 lines cut ...]
          Tx-Power: 18 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: unknown
          Bit Rate: unknown
          Encryption: unknown
          Type: nl80211  HW Mode(s): 802.11nac
          Hardware: 14C3:7662 14C3:7662 [Generic MAC80211]
          TX power offset: unknown
          Frequency offset: unknown
          Supports VAPs: yes  PHY name: phy0

mesh1     ESSID: unknown
          Access Point: 00:00:00:00:00:00
          Mode: Mesh Point  Channel: 1 (2.412 GHz)
          Tx-Power: 19 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: unknown
          Bit Rate: unknown
          Encryption: unknown
          Type: nl80211  HW Mode(s): 802.11bgn
          Hardware: 14C3:7662 14C3:7662 [Generic MAC80211]
          TX power offset: unknown
          Frequency offset: unknown
          Supports VAPs: yes  PHY name: phy1
```

I will leave the device hanging in this state for a few days, just in case some additional debug output can be captured.

mmalte, 2017-03-30:
I've got an 860l B1 runnning with the Lede Based Gluon Master. I'll disable the LAN-Mesh to try to reproduce.

mweinelt, 2017-05-14:
This problem seems limited to the 2.4 GHz Radio. I see both mesh and client beacons on 5 GHz.

mweinelt, 2017-08-07:
After ~13 days uptime the connectivity broke away.

```
root@64289-rocket-cloud:/# iw dev
phy#1
        Interface client1
                ifindex 18
                wdev 0x100000003
                addr 92:8d:78:23:af:1c
                ssid darmstadt.freifunk.net
                type AP
                channel 1 (2412 MHz), width: 20 MHz, center1: 2412 MHz
                txpower 19.00 dBm
        Interface mesh1
                ifindex 17
                wdev 0x100000002
                addr 96:8d:78:23:af:1c
                type mesh point
                channel 1 (2412 MHz), width: 20 MHz, center1: 2412 MHz
                txpower 19.00 dBm
phy#0
        Interface client0
                ifindex 20
                wdev 0x3
                addr 90:8d:78:23:af:1e
                ssid darmstadt.freifunk.net
                type AP
                channel 44 (5220 MHz), width: 20 MHz, center1: 5220 MHz
                txpower 18.00 dBm
        Interface mesh0
                ifindex 19
                wdev 0x2
                addr 92:8d:78:23:af:1e
                type mesh point
                channel 44 (5220 MHz), width: 20 MHz, center1: 5220 MHz
                txpower 18.00 dBm
root@64289-rocket-cloud:/# iw dev mesh1 station dump
Station aa:69:23:85:e1:11 (on mesh1)
        inactive time:  80 ms
        rx bytes:       48939625908
        rx packets:     349676967
        tx bytes:       380122141
        tx packets:     4043524
        tx retries:     2034539
        tx failed:      9823
        rx drop misc:   673293
        signal:         -75 [-83, -75] dBm
        signal avg:     -74 [-83, -74] dBm
        tx bitrate:     78.0 MBit/s MCS 12
        rx bitrate:     1.0 MBit/s
        expected throughput:    15.563Mbps
        mesh llid:      11865
        mesh plid:      21358
        mesh plink:     ESTAB
        mesh local PS mode:     ACTIVE
        mesh peer PS mode:      ACTIVE
        mesh non-peer PS mode:  ACTIVE
        authorized:     yes
        authenticated:  yes
        associated:     yes
        preamble:       long
        WMM/WME:        yes
        MFP:            no
        TDLS peer:      no
        DTIM period:    2
        beacon interval:100
        short slot time:yes
        connected time: 1486706 seconds
Station 42:f7:31:6f:6c:c9 (on mesh1)
        inactive time:  80 ms
[... 251 lines cut ...]
10:53:13.121026 BAT 42:f7:31:6f:6c:c9: BCAST, orig da:ff:61:00:03:05, seq 2834615, IP6 fe80::d8ff:61ff:fe00:304 > ff02::1:ff61:8de0 ICMP6 neighbor solicitation, who has 2001:67c:2ed8:6100:e1cb:bfe:7561:8de0, length 72
10:53:13.121238 BAT aa:69:23:85:e1:11: BCAST, orig da:ff:61:00:02:05, seq 8067081, IP6 fe80::d8ff:61ff:fe00:204 > ff02::1:ff6c:aa7f ICMP6 neighbor solicitation, who has 2001:67c:2ed8:6100:d73:d87:456c:aa7f, length 72
10:53:13.121464 BAT aa:69:23:85:e1:11: BCAST, orig da:ff:61:00:03:05, seq 2834615, IP6 fe80::d8ff:61ff:fe00:304 > ff02::1:ff61:8de0 ICMP6 neighbor solicitation, who has 2001:67c:2ed8:6100:e1cb:bfe:7561:8de0, length 72
10:53:13.121678 BAT aa:69:23:85:e1:11: BCAST, orig da:ff:61:00:02:05, seq 8067082, IP6 fe80::d8ff:61ff:fe00:204 > ff02::1:ffc0:9fa1 ICMP6 neighbor solicitation, who has 2001:67c:2ed8:6100:5888:5b74:dcc0:9fa1, length 72
10:53:13.121891 BAT aa:69:23:85:e1:11: BCAST, orig 62:f1:f6:6c:0c:24, seq 63495, IP6 fd01:67c:2ed8:6100:acce:62ab:8a1e:ebab > ff02::1:ff00:104 ICMP6 neighbor solicitation, who has fe80::d8ff:61ff:fe00:104, length 72
10:53:13.122104 BAT aa:69:23:85:e1:11: BCAST, orig da:ff:61:00:03:05, seq 2834616, IP6 fe80::d8ff:61ff:fe00:304 > ff02::1:ff65:8b68 ICMP6 neighbor solicitation, who has 2001:67c:2ed8:6100:957e:c4a9:1b65:8b68, length 72
10:53:13.122324 BAT aa:69:23:85:e1:11: BCAST, orig da:ff:61:00:03:05, seq 2834615, IP6 fe80::d8ff:61ff:fe00:304 > ff02::1:ff61:8de0 ICMP6 neighbor solicitation, who has 2001:67c:2ed8:6100:e1cb:bfe:7561:8de0, length 72
10:53:13.122538 BAT 96:8d:78:23:af:1c: BCAST, orig da:ff:61:00:03:05, seq 2834615, IP6 fe80::d8ff:61ff:fe00:304 > ff02::1:ff61:8de0 ICMP6 neighbor solicitation, who has 2001:67c:2ed8:6100:e1cb:bfe:7561:8de0, length 72
10:53:13.784990 BAT 96:8d:78:23:af:1c: BCAST, orig da:ff:61:00:02:05, seq 8067089, IP6 fe80::d8ff:61ff:fe00:204 > ff02::1:ff1e:55bd ICMP6 neighbor solicitation, who has 2001:67c:2ed8:6100:34ba:5c95:fe1e:55bd, length 72
10:53:13.785986 BAT 96:8d:78:23:af:1c: BCAST, orig da:ff:61:00:04:05, seq 5848757, IP6 fe80::d8ff:61ff:fe00:404 > ff02::1:ff01:596c ICMP6 neighbor solicitation, who has fd01:67c:2ed8:6100:10eb:bc03:f301:596c, length 72
10:53:13.786219 BAT 42:f7:31:6f:6c:c9: BCAST, orig da:ff:61:00:02:05, seq 8067089, IP6 fe80::d8ff:61ff:fe00:204 > ff02::1:ff1e:55bd ICMP6 neighbor solicitation, who has 2001:67c:2ed8:6100:34ba:5c95:fe1e:55bd, length 72
10:53:13.786432 BAT 42:f7:31:6f:6c:c9: BCAST, orig da:ff:61:00:04:05, seq 5848757, IP6 fe80::d8ff:61ff:fe00:404 > ff02::1:ff01:596c ICMP6 neighbor solicitation, who has fd01:67c:2ed8:6100:10eb:bc03:f301:596c, length 72
10:53:13.786646 BAT aa:69:23:85:e1:11: BCAST, orig da:ff:61:00:02:05, seq 8067089, IP6 fe80::d8ff:61ff:fe00:204 > ff02::1:ff1e:55bd ICMP6 neighbor solicitation, who has 2001:67c:2ed8:6100:34ba:5c95:fe1e:55bd, length 72
10:53:13.786858 BAT aa:69:23:85:e1:11: BCAST, orig da:ff:61:00:04:05, seq 5848757, IP6 fe80::d8ff:61ff:fe00:404 > ff02::1:ff01:596c ICMP6 neighbor solicitation, who has fd01:67c:2ed8:6100:10eb:bc03:f301:596c, length 72
[... cut ...]
10:53:13.787491 BAT aa:69:23:85:e1:11: BCAST, orig da:ff:61:00:02:05, seq 8067090, IP6 fe80::d8ff:61ff:fe00:204 > ff02::1:ff80:7004 ICMP6 neighbor solicitation, who has 2001:67c:2ed8:6100:99da:2d49:9b80:7004, length 72
10:53:13.787720 BAT 96:8d:78:23:af:1c: BCAST, orig da:ff:61:00:02:05, seq 8067090, IP6 fe80::d8ff:61ff:fe00:204 > ff02::1:ff80:7004 ICMP6 neighbor solicitation, who has 2001:67c:2ed8:6100:99da:2d49:9b80:7004, length 72
10:53:1^C61ff:fe00:504 > ff02::1:ff12:ee3f ICMP6 neighbor solicitation, who has 2001:67c:2ed8:6100:dccc:afb3:f012:ee3f, length 72
```

1.1~20170720
v2017.1-59-g582d0961

Restarting & updating to a newer release now.

bobster-galore, 2018-01-04:
Is that solved with a newer release?

rotanid, 2018-01-05:
probably not, but @Codefetch might be working on an updated driver backport of mt76

mweinelt, 2018-02-28:
@NeoRaider At the last developer meeting you offered to provide a branch that uses the same mac address allocation behaviour for both radios on this device. This is a friendly reminder.

NeoRaider, 2018-03-08:
@mweinelt Branch: debug-mac-addresses (completely untested)

mweinelt, 2018-03-09:
Nope, 2.4 GHz radio is completely unusable. 5 GHz otoh works flawlessly.

mweinelt, 2018-04-27:
Opened an issue against mt76

https://github.com/openwrt/mt76/issues/168

mweinelt, 2018-11-26:
Things have come a long way and improved so much that I consider removing the BROKEN flag soon.

CodeFetch, 2018-12-01:
That's great news! I think we should wait a little longer before removing the broken flag as we have other devices with that chipset, too and we should really be sure it works if we add new chipset support. If someone has one of those devices, they now know it should be stable and can verify it. 
I'm going to test the Digineo AC1200 Pro in a crowded environment.
Thank you very much for you effort!

CodeFetch, 2018-12-01:
@rotanid Yes, it is a WG3526 branded by Digineo. Other interesting devices are the Xiaomi Mi WiFi Mini and the Xiaomi Mi WiFi R3G, but they need to be flashed with a Xiaomi development firmware, then registered online to get SSH access and then flashed via SSH. It would be nice if someone could write a tool that automates this process. Anyway this is great news as I assumed nbd would not finish fixing mesh support and we already considered collecting funds for fixing these issues.

rotanid, 2018-12-01:
it seems someone is already funding him to do this, otherwise he wouldnt invest as much time in it, just watch the commitlog of the mt76 repository...

CodeFetch, 2018-12-01:
He only receives funding for AP/STA, but not Mesh functionality as far as I know. If we have a Mesh-only issue, chances are bad it is going to be fixed if it is a big effort.

mweinelt, 2019-02-12:
I opened up a new issue on mt76, the whole 2.4 GHz radio is dying again after some days of uptime. The device had been running fine for nearly two months, but the OpenWrt bump on january 15th introduced new issues.

https://github.com/openwrt/mt76/issues/246

mweinelt, 2019-03-09:
I'm satisfied by nbds progress on the driver to close this issue again. The radios now, instead of logging MCU message timeouts, are simply being reset, which makes them quite resilient.

Fixed after 2086bed7cc33eb03606dc2d243554ef4d5838604.

closed 2019-03-09 by mweinelt