freifunk-gluon/gluon #1079: strange non-deterministic networking on some nodes

kpanic23, 2017-03-23:
We are noticing strange network lockups on our nodes.
The nodes somehow lose network connection on layer 3, are not pingable, no ssh, nothing. But they can still be pinged via "batctl ping" on layer 2.
There seems to be no system behind which node is affected, the amount of clients, traffic, everything seems to be quite random. 
One of my own nodes at home was affected twice, with two days in between. I just cycled the power both times, and have never had a problem since and before.
Strangely, I have seen NTP packets _from_ these nodes, while being affected, to our NTP servers.
The affected nodes appear in the map (hopglass) as online, but no message since X.
They get listed in the list of disappeared nodes, but with green color.
On the map itself, they are shown twice, once online and once offline.

Since there is no chance to connect to those nodes remotely, neither of them have a serial console attached and one can't anticipate which node is affected next, there is unfortunatelly no chance of getting any logs.
We have noticed this behaviour since January, when we first experimented with hopglass. On our old alfred-based meshviewer map they were listed as plain offline, so it never occured to us they were still pingable vie batctl. So I suppose this problem exists quite a bit longer.

rotanid, 2017-03-23:
we only have one node where this problem re-appears often.
the issue isn't related to hopglass-server or respondd or alfred, it's just that the neighbor-nodes still report they see the broken node, as layer2 still works.

rotanid, 2017-03-24:
@kpanic23 did you see this issue on nodes that have VPN-link? i did not.
i suspect that this is the same as #605 or at least similar to it.
my workaround script probably doesn't catch it because it does assume the connection is great if a batctl ping to the gateway succeeds.

T-X, 2017-03-24:
For someone capable of reproducing the issue frequently on a specific node: Could you assign a static IPv4 address on br-client and check whether both IPv4 and IPv6 or just one of these protocol families fails? A long(er) time ping run to compare packetlosses of IPv4 vs. IPv6 could be interesting, too.

Also, could affected people tell their Gluon version and link their site.conf?

rotanid, 2017-03-24:
as i don't have easy access to a node that often has the issue, we have to count on @kpanic23 .
regarding gluon version, they and we are both using 2016.2.4, our site.conf is there:
https://github.com/tecff/site-ffa/blob/stable/site.conf
(technically the issue happend with 2016.2.3 - the .4 update took place today)

kpanic23, 2017-03-24:
@rotanid I don't think it's the same as #605 , because it also happens on nodes which are solely meshing via cable. I'd rather think it might be something in batman-adv.
[image]

Our site(s) are here: https://github.com/ff3l/site-ff3l/tree/2016.2

The main problem in trying to debug this, is, you really don't have the slightest idea which node is affected next. As I have written, one of my nodes (which unfortunatelly is an WA860RE, so no quick opening and connecting a serial interface) had this problem twice, and after power-cycling the last time that particular node has been running flawlessly for months. I don't have the slightest idea either how to provoke this behaviour or how to predict where it will happen next. Frankly, I'm totally clueless.

rotanid, 2017-03-24:
ok, didn't know about cable-mesh nodes.
we have a node where it happens often, but we also have no possibility to add a console.

kb-light, 2017-03-26:
I maybe had a similar issue, the node (Ubiquiti Outdoor+) was powered up but after some time it was shown as offline (meshviewer). It has mesh-on-wan and 11s-mesh enabled. Via link-local on wan interface I was able to get a ssh session. It seems the bat0 interface was down, after a reboot everything is fine.

rotanid, 2017-03-27:
were you able to look at logread/dmesg via the link-local ssh connection?

rotanid, 2017-03-27:
maybe related to b7eeef9b04b44a70b2a953c4efe35a3fdceba2db ?

kpanic23, 2017-03-27:
```
root@ff3l-GU-Haltingen-Offloader:~# ping6 fe80::62e3:27ff:fee7:56ce%br-mesh_lan
PING fe80::62e3:27ff:fee7:56ce%br-mesh_lan (fe80::62e3:27ff:fee7:56ce%br-mesh_lan): 56 data bytes
^C
--- fe80::62e3:27ff:fee7:56ce%br-mesh_lan ping statistics ---
11 packets transmitted, 0 packets received, 100% packet loss

root@ff3l-GU-Haltingen-Offloader:~# batctl p 60:e3:27:e7:56:ce
PING 60:e3:27:e7:56:ce (02:1b:28:67:eb:53) 20(48) bytes of data
20 bytes from 60:e3:27:e7:56:ce icmp_seq=1 ttl=50 time=0.12 ms
20 bytes from 60:e3:27:e7:56:ce icmp_seq=2 ttl=50 time=0.12 ms
20 bytes from 60:e3:27:e7:56:ce icmp_seq=3 ttl=50 time=0.11 ms
^C--- 60:e3:27:e7:56:ce ping statistics ---
3 packets transmitted, 3 received, 0% packet loss
rtt min/avg/max/mdev = 0.110/0.115/0.119/0.004 ms
root@ff3l-GU-Haltingen-Offloader:~# 
```
[image]

NeoRaider, 2017-03-27:
@kpanic23 Your ping6 command looks wrong, the IP address used on the lower layers (br-wan/br-mesh_lan) should look completely different from the primary MAC address, so this can't work.

kpanic23, 2017-03-27:
Oops... I've copied the wrong lines while desperately trying to communicate on any interface. Nevertheless, you can't ping that node on any interface with any IP address. It seems to be completely deaf.

T-X, 2017-03-27:
Why I'm especially interested in IPv4 vs. IPv6 comparisons: I saw that your latest site version adds the gluon-ebtables-segment-mld package and statistics regarding IPv4 vs. IPv6 could be an indicator whether your issues might be somehow related to that package. That is, if IPv4 works and IPv6 doesn't then it is quite likely an issue regarding bridge multicast snooping. If both fail, then it has nothing to do with bridge multicast snooping.

Maybe you could just add static IPv4 addresses to some random nodes you have access to and run ping/ping6 in the background for a few days, while doing a "tcpdump 'icmp or icmp6' capture?

Are all your nodes involved running Gluon >= 2016.2 (in particular the nodes you are trying to ping from/to)? Are your gateway nodes involving bridges on top of bat0 (and if yes, do they have multicast_router=2 for bat0 set)?

kpanic23, 2017-03-28:
@T-X Yes, the gluon-ebtables-segment-mld package is added since we use 2016.2.1
Our gateways do indeed use bridges on top of bat[0-32], and no, they have multicast_router set to 1
Might that be the problem? Shall I change that setting to 2? Or maybe build a new firmware omitting that package?
Only 9 of our nodes are currently running gluon <2016.2, the majority uses 2016.2.4
(See statistics on https://map.ff3l.net)

T-X, 2017-03-28:
Ah, okay, the missing multicast_router = 2 could be a probable cause, yes! Your map server is running on some such gateway? Did you also run the ping6 tests from your gateway?

For clarification: multicast_router=2 forces all multicast traffic onto this bridge port, no matter if it detected a multicast router or multicast listener behind it. This is needed now because with the segment-mld package we shrink the horizon of the bridge concerning multicast listeners (by refusing to forward IGMP/MLD into the mesh). Similar to the unicast MAC address in the bridge, batman-adv takes over this role in between now (and distributes multicast listener status in a way more efficient than IGMP/MLD, via the reactive batman-adv Translation Table).

- multicast_router=1 -> autodetect a multicast router
- multicast_router=2 -> force port to be a multicast router port
- multicast_router=0 -> never mark this port as a multicast router port

A multicast router port means, that all multicast traffic will be copied here, even if no multicast listener was detected.

(hrm, even after explaining it a few times on IRC I still suck at explaining this, and I guess people will continue to stumble upon this... I'll try to add one more sentence to the package description)

rotanid, 2017-03-28:
we have a similar setup, most nodes run >=2016.2.4.
at the moment, there's only one node showing the issue but that didn't change after doing "echo 2 > /sys/class/net/bat0/brport/multicast_router"

T-X, 2017-03-29:
Okay. If the IPv4 vs. IPv6 ping is too difficult for you for now, here is an easier way we could check first: a) what does "ip neigh" say, is the IPv6 address resolved fine or is it already stuck at address resolution? If it is stuck at address resolution, next to a missing entry you should see "ICMPv6 Neighbor Solicitations" instead of ICMPv6 echo requests with tcpdump - if that's the case could you check whether setting an ipv6 address <-> MAC address entry manually helps?

kpanic23, 2017-03-29:
@T-X I have tested pinging from my client, the gateways and one of the other nodes in the local mesh. No change.
Should multicast_router=2 be set on the bat interface or on the bridge containing the bat interface? 
Also, where shall I run "ip neigh"? And where tcpdump? On gluon there is no "ip neigh". On the gateway it dumps me a list of thousands of IPv4/MAC addresses. Nothing gets stuck.

Also, I don't understand why a setting on the gateway disrupts IPv6 connectivity in the local mesh (which should even work without any connection to any gateway)

lcb01a, 2017-03-29:
Hi,

I have a node with the very same issue, but I managed to connect to it via the link local address of ibss0.2 (We use vlans ontop of the ibss interfaces).
The node seems to not apply the IP-Addresses to br-client and bat0 is not part of the bridge:

```
root@Hinterhof-LinkesZentrum-Neu:~# ip a show dev br-client
15: br-client: <BROADCAST,MULTICAST> mtu 1500 qdisc noop
    link/ether e6:5f:e0:41:97:2a brd ff:ff:ff:ff:ff:ff
```

```
root@Hinterhof-LinkesZentrum-Neu:~# brctl show
bridge name     bridge id               STP enabled     interfaces
br-mesh_lan             7fff.8ec7ee4ea314       no              eth0
br-wan          7fff.8ec7ee4ea310       no              eth1
br-client               8000.e65fe041972a       no              client0
```

```
root@Hinterhof-LinkesZentrum-Neu:~# batctl if
ibss0.2: active
primary0: active
br-wan: active
```

Since bat0 is not part of br-client it explains why the node is not reachable with anything but `batctl ping`

A reboot usually fixes this and it also only occurs randomly after a reboot of a node.
I suspect that some interface is not yet ready when its being configured. I'm trying to find more on the issue.

kpanic23, 2017-04-10:
Might commit d452a7c2cf1c0da4e034666a50dc0e7aa9ddc592 fix that issue?

NeoRaider, 2017-04-10:
@kpanic23 No, that one just caused kernel crashs. I think this issue might be a netifd issue, possibly the same as #905.

NeoRaider, 2017-04-12:
Please test with the current master (e45c30330d2eede2cec39c3082ad5e6bdc40a6ba or later), I've refactored the batman-adv interface management to make it much more robust.

kpanic23, 2017-04-20:
@NeoRaider Nailed it!
That really seems to have fixed the problem. At the moment we have 24 nodes, that had been constantly locking up before, running a current master (ad9878b), and we have had not a single lockup so far. A few of the nodes, e.g. https://map.freifunk-3laendereck.net/#!v:m;n:60e327e756ce are rebooting by themselves once or twice a day, but that's waaaay better than locking up.
Actually I'm thinking about signing that master as new stable firmware ;)

closed 2017-04-21 by NeoRaider