freifunk-gluon/gluon #1086: [RFC] add Hidden SSID option to site.conf

Nurtic-Vibe, 2017-04-06:
It would be helpful to have an option to mark some SSIDs as hidden via site configuration, especially in case of mesh SSIDs.

NeoRaider, 2017-04-06:
Hiding SSID works by not sending out beacons, and instead having stations probe for known networks. In IBSS and 11s networks, there are no stations, but only equal peers finding each other through beacons; thus, hiding the SSID is not possible.

As hiding the AP SSIDs doesn't make much sense, and it's impossible for other VIF types, I don't think such an option would be useful.

closed 2017-04-06 by NeoRaider

FreifunkUFO, 2017-04-06:
there is no possibility to hide adhoc-mesh. some gluon-users are hiding that wifi with a strange SSID-Name.
but, one possibility would be nice: to add wpa2 to the adhoc-mesh. so normal users cant login/ see it anymore as an open wifi.

hiding the Client-AP-Network doenst make much sense.

NeoRaider, 2017-04-06:
I fear adding encryption would have a negative effect on mesh performance; also, there are a lot of incompatibilities and general breakage in the different drivers regarding encryption of non-AP networks.

I tried to get encryption running for an 11s network with OpenWrt using ath9k a while ago, and I couldn't make it work for some reason; I don't plan to invest more time into this at the moment.

Regarding IBSS meshing: I consider this a legacy technology, and don't think we need to add new features to it. Gluon will eventually drop IBSS support altogether, after we've further developed our migration mechanisms.

rubo77, 2017-04-06:
Also worth mention is, that hidden SSIDs are dangeroous, because they server a possible attack vector:

https://www.heise.de/ct/hotline/SSID-Broadcast-besser-an-1081978.html

viisauksena, 2017-04-07:
there is also superlimited benefit in encrypting mesh traffic which is
than  streamed via nearby nodes in open wifi - if you dont try to secure
this connections at all. which would be hard to implement, given the
limited resources and limitless variety of nodes and end-user-devices