freifunk-gluon/gluon #1088: Gluon (Lede) ipq806x and Archer C2600

seligenthaler, 2017-04-07:
Hardware: TP-Link ARCHER C2600 v1.1
Gluon: v2016.2-243-g6a0ca58

If you reboot the router there is a shutdown with all LED going off.
You must use the power switch to bring it back.

Error in config- mode wifi:
>500 Interner Serverfehler
>Entschuldigung, auf dem Server ist ein unerwarteter Fehler aufgetreten.
>Failed to execute dispatcher target for entry '/admin/wifi-config'.
>The called action terminated with an exception:
>/lib/gluon/web/model/admin/wifi-config.lua:19: bad argument #1 to 'txpwrlist' (string expected, got nil)

After the config is done and after rebooting (manualy s.a.) the router comes up and is accessible by SSH.
Alfred and batman did not work.  (lack of bat0)

The client0 (5GHz) is on purpose manualy disabled.
In (/sys/class/net) the virtual bat0-Device and primary0-Device are missing.
The  bat0-Device section in /etc/config/network is available.

Just for testing and information:
You can use the command "batctl if add client1" (does this make sense?)
This will create an temporary virtual Device bat0
Then you can start alfred
/etc/init.d/alfred start
>/etc/init.d/alfred: waiting 30 secs for br-client address...
>/etc/init.d/alfred: starting alfred
>/etc/init.d/alfred: starting batadv-vis

any suggestions?

documents attached:  network wireless logdata
[c2600.zip](https://github.com/freifunk-gluon/gluon/files/905766/c2600.zip)

belzebub40k, 2017-04-07:
I also recognized this with the mvebu target. I assume it is because the device does not support IBSS or 802.11s but I didn't had time to look deeper into the issue yet.

belzebub40k, 2017-04-08:
Found the problem for the Linksys WRT1200AC (mvebu). It addresses the radios by its PCIe path in `wireless.radio(0|1).path`

```
# uci show wireless | grep .path
wireless.radio0.path='soc/soc:pcie-controller/pci0000:00/0000:00:01.0/0000:01:00.0'
wireless.radio1.path='soc/soc:pcie-controller/pci0000:00/0000:00:02.0/0000:02:00.0'
```

To find the corresponding PHY gluon searches in `/sys/devices/ + wireless.radio(0|1).path + /ieee80211/phy*` but this is wrong for this kind of path. instead of `/sys/devices/` it needs needs to prepend `/sys/devices/platform/`.

I already have a working fix for my device. If @seligenthaler confirms that the Archer C2600 has similar values in `wireless.radio(0|1).path` I will open a PR.

@NeoRaider Do you see any other locations in Gluon where this could cause problems?

seligenthaler, 2017-04-08:
@belzebub40k 
You are right, same problem.
Here are the results from the C2600

wireless.radio(0|1).path:
>root@su-sgb-talsperre-bone:~# uci show wireless | grep .path
>wireless.radio0.path='soc/1b500000.pci/pci0000:00/0000:00:00.0/0000:01:00.0'
>wireless.radio1.path='soc/1b700000.pci/pci0001:00/0001:00:00.0/0001:01:00.0'

Wrong path:
>root@su-sgb-talsperre-bone:/sys/devices# ls -l
>drwxr-xr-x   15 root     root             0 Jan  1  1970 platform
>drwxr-xr-x     6 root     root             0 Jan  1  1970 system
>drwxr-xr-x   10 root     root             0 Jan  1  1970 virtual

Correct path
>root@su-sgb-talsperre-bone:/sys/devices/platform/soc# ls -l | grep 1b
>drwxr-xr-x    3 root     root             0 Jan  1  1970 1b500000.pci
>drwxr-xr-x    3 root     root             0 Jan  1  1970 1b700000.pci
>drwxr-xr-x    2 root     root             0 Jan  1  1970 1bb00000.syscon

seligenthaler, 2017-04-08:
sorry accidently closed

seligenthaler, 2017-04-08:
Path exists:
/sys/devices/platform/soc/1b500000.pci/pci0000:00/0000:00:00.0/0000:01:00.0/ieee80211/phy0
and
/sys/devices/platform/soc/1b700000.pci/pci0001:00/0001:00:00.0/0001:01:00.0/ieee80211/phy1

seligenthaler, 2017-04-09:
@belzebub40k 
Your PR
./gluon-web-wifi-config/luasrc/lib/gluon/web/model/admin/wifi-config.lua:

The same I found here:
./gluon-core/luasrc/usr/lib/lua/gluon/util.lua

Can you confirm this?

belzebub40k, 2017-04-09:
Updated PR

seligenthaler, 2017-04-09:
@belzebub40k 
@NeoRaider 
I can confirm that ther error in config-mode (wlan) on C2600 is gone with the solution in PR 1089.
The boot-problem ist still there and no bat0-device.
[logdata.zip](https://github.com/freifunk-gluon/gluon/files/908286/logdata.zip)

NeoRaider, 2017-04-09:
@seligenthaler, did you completely reset your configuration while upgrading to a firmware containing #1089?

One issue I found is the `option 'disabled' '1'` in your */etc/config/wireless*. if the issue persists even after a clean installation of a new version, manually run */lib/gluon/upgrade/200-wireless* to see if it produces an error message.

Your last log output also contains a segmention fault; if you have further issues, please attach the output of *dmesg* in addition to *logread*, so we can debug that.

seligenthaler, 2017-04-09:
@NeoRaider 
When I use the _sysupgrade_ fuction the device gets bricked.
Then only the the power-led flashes irregular.
Solution -> TFTP gluon factory restore (bootloader).

I used _sysupgrade -n_ witch leads to the config-mode. 
In config-mode the device "takes it time" to reboot, sometimes you must save the config again. There is no direct reboot as if you use the _reboot_ command.

In the last log I used the WLAN config to prove that this works now, i disabled the 5GHz. Thats the  `option 'disabled' '1'`  you mentioned.

_/lib/gluon/upgrade/200-wireless_ runs without output
I will do a new sysupgrade -n with no changes in WLAN 
(only inserting the ssh-key,node-name, email and disable mesh-vpn)

The update process:
root@su-sgb-talsperre-bone:/tmp# sysupgrade -n gluon-su-su-v2016.2.lede-sgb-exp-2.9-tp-link-archer-c2600-sysupgrade.bin
Sending TERM to remaining processes ... ntpd dnsmasq respondd ubusd askfirst sh sh ntpd
Sending KILL to remaining processes ... askfirst
Switching to ramdisk...
Performing system upgrade...
Unlocking os-image ...
Unlocking rootfs ...

Writing from <stdin> to os-image ...

Writing from <stdin> to rootfs ...
Upgrade completed
Rebooting system...
umount: can't unmount /dev/pts: Resource busy
umount: can't unmount /dev: Resource busy
umount: can't unmount /tmp: Resource busy

The device goes "off" (no led)

### FF-Node:   su-sgb-talsperre-bone
### Node mac:  ec:08:6b:9c:ec:37
### Datum:     Sun Apr  9 20:20:02 CEST 2017
### Uptime:    20:20:02 up 17 min,  load average: 0.36, 0.25, 0.14
### Hardware:  TP-Link Archer C2600
### Branch:    experimental
### Gluon:     v2016.2-244-g13c61d9+
### Firmware:  v2016.2.lede-sgb-exp-2.9

root@su-sgb-talsperre-bone:~# /lib/gluon/upgrade/200-wireless
root@su-sgb-talsperre-bone:~#

Logdata:
[C2600.zip](https://github.com/freifunk-gluon/gluon/files/908450/C2600.zip)

seligenthaler, 2017-04-09:
New firmware 247
```
### FF-Node:   su-sgb-talsperre-bone
### Node mac:  ec:08:6b:9c:ec:37
### Datum:     Sun Apr  9 23:52:37 CEST 2017
### Uptime:    23:52:37 up 1 min,  load average: 0.47, 0.19, 0.07
### Hardware:  TP-Link Archer C2600
### Branch:    experimental
### Gluon:     v2016.2-247-g9e487f3+
### Firmware:  v2016.2.lede-sgb-exp-2.10
```
(Addup PR 1089)

No bat0.

After web-config it takes 2minutes for the reboot to start, power-led going off.  It needs no second save-click (I was to impatient)  
Maybe I have to wait more after a normal sysupgrade (without the -n). I will test that later.

Here are the new log-files:
[c2600-247.zip](https://github.com/freifunk-gluon/gluon/files/908597/c2600-247.zip)

NeoRaider, 2017-04-09:
The reasion you have no bat0 is that neither of the mesh interfaces is working (the 5GHz one because the ath10k firmware doesn't support 11s; no idea what is wrong with the 2.4GHz one, "ip: SIOCSIFFLAGS: Invalid argument" isn't really helping). Could you add the output of `iw phy`?

seligenthaler, 2017-04-09:
@NeoRaider 
Should I try  `GLUON_ATH10K_MESH ?= ibss`  ?
Here comes phy:
[phydata.zip](https://github.com/freifunk-gluon/gluon/files/908630/phydata.zip)

NeoRaider, 2017-04-09:
`GLUON_ATH10K_MESH = ibss` won't help. I'm not sure if there are 11s-capable firmwares for QCA99X0 at all.

I still don't have an idea why the 2.4GHz mesh interface doesn't work.

seligenthaler, 2017-04-10:
@NeoRaider 
@belzebub40k 

Yes, thanks both of you.
ibss was the hint I needed.  I changed it in site.mk.
The C2600 comes up connects the first time to all my other nodes.
To gluon 2016.2.5 and 2016.2.x-247 nodes.
Will there be a solution for 11s? Maybe later?

That's my info script:

```
### FF-Node:   su-sgb-talsperre-bone
### FF-Code:   su-su
### Node mac:  ec:08:6b:9c:ec:37
### Datum:     Mon Apr 10 01:56:01 CEST 2017
### Uptime:    01:56:01 up 8 min,  load average: 0.56, 0.34, 0.19
### Hardware:  TP-Link Archer C2600
### Branch FW: experimental
### Branch akt:experimental
### Gluon:     v2016.2-247-g9e487f3+
### Firmware:  v2016.2.lede-sgb-exp-2.10
### TX client0:  20 dBm ( 100 mW)
### TX client1:  20 dBm ( 100 mW)
### Momentane Clients:  0
### Gesammt FF-Clients: 460
########################################
MemTotal:         480788 kB
MemFree:          394308 kB
MemAvailable:     381200 kB
Flash:           23.9M    748.0K     23.2M   3% /overlay
########################################
fda0:747e:ab29:2241:ee08:6bff:fe9c:ec37
fe80::ee08:6bff:fe9c:ec37
########################################
Mesh on LAN:    "up": false
Mesh on WAN:    "up": false
Station d2:74:f5:66:25:09 (on mesh1)
        mesh plink:     ESTAB
Station 76:c4:d8:ce:0b:29 (on mesh1)
        mesh plink:     ESTAB
Station 7a:df:0a:ca:3d:25 (on mesh1)
        mesh plink:     ESTAB
Station de:cf:6c:81:b3:31 (on mesh1)
        mesh plink:     ESTAB
########################################
[B.A.T.M.A.N. adv 2017.0, MainIF/MAC: primary0/0a:5a:47:6b:98:1b (bat0/ec:08:6b:9c:ec:37 ]
  Router            ( TQ) Next Hop          [outgoingIf]  Bandwidth
  04:57:a0:90:57:10 ( 27) 7a:df:0a:ca:3d:25 [     mesh1]: 92.0/92.0 MBit
* 04:57:a0:90:57:40 (206) 7a:df:0a:ca:3d:25 [     mesh1]: 92.0/92.0 MBit
########################################
Locally retrieved addresses (from bat0) announced via TT (TTVN: 2):
       Client         VID Flags    Last seen (CRC       )
 * ec:08:6b:9c:ec:37   -1 [.P....]   0.000   (0x796bbf2d)
 * ec:08:6b:9c:ec:37    0 [.P....]   0.000   (0xdf03bac6)
root@su-sgb-talsperre-bone:~#

```

seligenthaler, 2017-04-10:
Info to the reboot problem.
This seems to be an issue in lede.

From the lede project forum:

>When I reboot from the interface, the router just goes dead - no lights flashing or lit, every time. If I then pull the power and reconnect, it comes up fine. 
> 
>You problably have a version 1.1 instead of a version 1.0.
>Version 1.1 needs a patch that is not in Lede v17.01.0.

RubenKelevra, 2017-06-30:
With GLUON_ATH10K_MESH ?= ibss the 5 GHz Mesh don't work, 2,4 GHz Mesh work (but not very fast or reliable ...

But the machine is running very well for me, no reboot so far.

But both WIFI-LEDs are off with Gluon 2017.1.1.

RubenKelevra, 2017-06-30:
@NeoRaider 

I saw this fix on the lede-master, which seems to be a issue on this device:

https://github.com/lede-project/source/commit/189239ade353636f9e2788cd0467bcca97bb9324

Without this fix the pre-cal images are clearly not found:

```[  153.141741] ath10k driver, optimized for CT firmware, probing pci device: 0x40.
[  153.142645] ath10k_pci 0001:01:00.0: enabling bus mastering
[  153.143187] ath10k_pci 0001:01:00.0: pci irq msi oper_irq_mode 2 irq_mode 0 reset_mode 0
[  153.270836] ath10k_pci 0001:01:00.0: Direct firmware load for ath10k/pre-cal-pci-0001:01:00.0.bin failed with error -2
[  153.270877] ath10k_pci 0001:01:00.0: Falling back to user helper
[  153.386065] firmware ath10k!pre-cal-pci-0001:01:00.0.bin: firmware_loading_store: map pages failed
[  153.386504] ath10k_pci 0001:01:00.0: Direct firmware load for ath10k/fwcfg-pci-0001:01:00.0.txt failed with error -2
[  153.394033] ath10k_pci 0001:01:00.0: Falling back to user helper
[  153.448356] firmware ath10k!fwcfg-pci-0001:01:00.0.txt: firmware_loading_store: map pages failed
[  153.448617] ath10k_pci 0001:01:00.0: qca99x0 hw2.0 target 0x01000000 chip_id 0x003b01ff sub 168c:0002
[  153.456224] ath10k_pci 0001:01:00.0: kconfig debug 1 debugfs 1 tracing 0 dfs 1 testmode 0
[  153.467468] ath10k_pci 0001:01:00.0: firmware ver 10.4.1.00030-1 api 5 features no-p2p crc32 d2901e01
[  153.473715] ath10k_pci 0001:01:00.0: failed to fetch board data for bus=pci,vendor=168c,device=0040,subsystem-vendor=168c,subsystem-device=0002 from ath10k/QCA99X0/hw2.0/board-2.bin
[  153.482873] ath10k_pci 0001:01:00.0: board_file api 1 bmi_id N/A crc32 7e56fd07
[  154.591535] ath10k_pci 0001:01:00.0: 10.4 wmi init: vdevs: 16  peers: 528  tid: 132
[  154.591564] ath10k_pci 0001:01:00.0: msdu-desc: 1424  skid: 32
[  154.671763] ath10k_pci 0001:01:00.0: htt-ver 2.2 wmi-op 6 htt-op 4 cal file max-sta 512 raw 0 hwcrypto 1
```

But if I just move the files from Gluon 2017.1 (firmware compiled with `GLUON_ATH10K_MESH ?= 11s`) both wifi-devices cannot be used at all:

```[   18.121274] ath10k driver, optimized for CT firmware, probing pci device: 0x40.
[   18.127841] ath10k_pci 0001:01:00.0: enabling device (0140 -> 0142)
[   18.134384] ath10k_pci 0001:01:00.0: enabling bus mastering
[   18.134830] ath10k_pci 0001:01:00.0: pci irq msi oper_irq_mode 2 irq_mode 0 reset_mode 0
[   18.278678] ath10k_pci 0001:01:00.0: Direct firmware load for ath10k/fwcfg-pci-0001:01:00.0.txt failed with error -2
[   18.278735] ath10k_pci 0001:01:00.0: Falling back to user helper
[   18.336165] firmware ath10k!fwcfg-pci-0001:01:00.0.txt: firmware_loading_store: map pages failed
[   18.337954] ath10k_pci 0001:01:00.0: qca99x0 hw2.0 target 0x01000000 chip_id 0x003b01ff sub 168c:0002
[   18.344209] ath10k_pci 0001:01:00.0: kconfig debug 1 debugfs 1 tracing 0 dfs 1 testmode 0
[   18.357709] ath10k_pci 0001:01:00.0: firmware ver 10.4.1.00030-1 api 5 features no-p2p crc32 d2901e01
[   18.382093] ath10k_pci 0001:01:00.0: failed to fetch board data for bus=pci,vendor=168c,device=0040,subsystem-vendor=168c,subsystem-device=0002 from ath10k/QCA99X0/hw2.0/board-2.bin
[   18.382196] ath10k_pci 0001:01:00.0: board_file api 1 bmi_id N/A crc32 7e56fd07
[   20.458458] ath10k_pci 0001:01:00.0: unable to read from the device
[   20.458483] ath10k_pci 0001:01:00.0: could not execute otp for board id check: -110
[   20.463523] ath10k_pci 0001:01:00.0: failed to get board id: -110
[   22.478476] ath10k_pci 0001:01:00.0: unable to read from the device (-110)
[   22.478499] ath10k_pci 0001:01:00.0: could not read board ext data addr (-110)
[   22.484233] ath10k_pci 0001:01:00.0: could not push board ext data (-110)
[   22.491493] ath10k_pci 0001:01:00.0: failed to download board data: -110
[   22.498295] ath10k_pci 0001:01:00.0: failed to run otp: -110
[   22.505066] ath10k_pci 0001:01:00.0: could not init core (-110)
[   22.510857] ath10k_pci 0001:01:00.0: could not probe fw (-110)
```

Also interesting:

With `GLUON_ATH10K_MESH ?= 11s` the ath10k_pci (without the patch mentioned above) is always complaining about the mesh-point stuff:

`[   23.597657] ath10k_pci 0001:01:00.0: must load driver with rawmode=1 to add mesh interfaces`

NeoRaider, 2017-07-01:
lede-project/source@189239a will probably need other changes to be backported first. More useful than testing Gluon would be to check if the LEDE master runs properly (i.e. the Wifi is working); as long as that is not the case, backporting doesn't make much sense.

The last issue is obvious: no 11s-capable ath10k firmware for the QCA99X0 exists, so `GLUON_ATH10K_MESH = 11s` simply can't do anything useful. We don't know if QCA will ever release one...

RubenKelevra, 2017-07-01:
@NeoRaider yeah I took a look at it, it looks like we need parts of these ones:

https://github.com/lede-project/source/commit/0a85a2fe615861a7822dad6a4f3d8fa0099f5d55
https://github.com/lede-project/source/commit/cc189c0b7fa015978b04bb663a75b1da726376b5
https://github.com/lede-project/source/commit/48ae1f535acc25e53967e8fc3862bef947abbaee

I cherry picked them for my builder and gonna try them.

RubenKelevra, 2017-07-01:
I've noticed that the blinking pattern of the config mode doesn't work ... it first interferes a long time (like 1-2 minutes) with the standard booting battern, means, both patterns merge together to a unsteady blinking.

After 1-2 minutes the boot seems to be finished and the power ligth constantly lit up, while in config mode.

RubenKelevra, 2017-07-01:
With all those patches it remains on this:

```
[   18.621273] ath10k driver, optimized for CT firmware, probing pci device: 0x40.
[   18.627883] ath10k_pci 0001:01:00.0: enabling device (0140 -> 0142)
[   18.634379] ath10k_pci 0001:01:00.0: enabling bus mastering
[   18.634820] ath10k_pci 0001:01:00.0: pci irq msi oper_irq_mode 2 irq_mode 0 reset_mode 0
[   18.770787] ath10k_pci 0001:01:00.0: Direct firmware load for ath10k/pre-cal-pci-0001:01:00.0.bin failed with error -2
[   18.770840] ath10k_pci 0001:01:00.0: Falling back to user helper
[   23.641505] ath10k_pci 0001:01:00.0: Direct firmware load for ath10k/fwcfg-pci-0001:01:00.0.txt failed with error -2
[   23.641564] ath10k_pci 0001:01:00.0: Falling back to user helper
[   23.724847] firmware ath10k!fwcfg-pci-0001:01:00.0.txt: firmware_loading_store: map pages failed
[   23.726640] ath10k_pci 0001:01:00.0: qca99x0 hw2.0 target 0x01000000 chip_id 0x003b01ff sub 168c:0002
[   23.732919] ath10k_pci 0001:01:00.0: kconfig debug 1 debugfs 1 tracing 0 dfs 1 testmode 0
[   23.746634] ath10k_pci 0001:01:00.0: firmware ver 10.4.1.00030-1 api 5 features no-p2p crc32 d2901e01
[   23.770697] ath10k_pci 0001:01:00.0: failed to fetch board data for bus=pci,vendor=168c,device=0040,subsystem-vendor=168c,subsystem-device=0002 from ath10k/QCA99X0/hw2.0/board-2.bin
[   23.770804] ath10k_pci 0001:01:00.0: board_file api 1 bmi_id N/A crc32 7e56fd07
[   25.848455] ath10k_pci 0001:01:00.0: unable to read from the device
[   25.848479] ath10k_pci 0001:01:00.0: could not execute otp for board id check: -110
[   25.853522] ath10k_pci 0001:01:00.0: failed to get board id: -110
[   27.868466] ath10k_pci 0001:01:00.0: unable to read from the device (-110)
[   27.868490] ath10k_pci 0001:01:00.0: could not read board ext data addr (-110)
[   27.874224] ath10k_pci 0001:01:00.0: could not push board ext data (-110)
[   27.881490] ath10k_pci 0001:01:00.0: failed to download board data: -110
[   27.888285] ath10k_pci 0001:01:00.0: failed to run otp: -110
[   27.895107] ath10k_pci 0001:01:00.0: could not init core (-110)
[   27.900836] ath10k_pci 0001:01:00.0: could not probe fw (-110)
```

For both interfaces while compiling with `GLUON_ATH10K_MESH ?= 11s` while the firmware is in place:
```
# find / -name "*pre-cal-pci-0001:01:00.0.bin"

/lib/firmware/ath10k/pre-cal-pci-0001:01:00.0.bin
```
As well as the board-2.bin

```
[   24.211262] ath10k driver, optimized for CT firmware, probing pci device: 0x40.
[   24.217914] ath10k_pci 0001:01:00.0: enabling device (0140 -> 0142)
[   24.224372] ath10k_pci 0001:01:00.0: enabling bus mastering
[   24.224814] ath10k_pci 0001:01:00.0: pci irq msi oper_irq_mode 2 irq_mode 0 reset_mode 0
[   24.360773] ath10k_pci 0001:01:00.0: Direct firmware load for ath10k/pre-cal-pci-0001:01:00.0.bin failed with error -2
[   24.360827] ath10k_pci 0001:01:00.0: Falling back to user helper
[   29.129704] ath10k_pci 0001:01:00.0: Direct firmware load for ath10k/fwcfg-pci-0001:01:00.0.txt failed with error -2
[   29.129740] ath10k_pci 0001:01:00.0: Falling back to user helper
[   29.169623] firmware ath10k!fwcfg-pci-0001:01:00.0.txt: firmware_loading_store: map pages failed
[   29.170785] ath10k_pci 0001:01:00.0: qca99x0 hw2.0 target 0x01000000 chip_id 0x003b01ff sub 168c:0002
[   29.177484] ath10k_pci 0001:01:00.0: kconfig debug 1 debugfs 1 tracing 0 dfs 1 testmode 0
[   29.189460] ath10k_pci 0001:01:00.0: firmware ver 10.4.3-ct-fW-009-cd072c1 api 5 features peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,set-special-CT crc32 9440ad0c
[   29.215425] ath10k_pci 0001:01:00.0: failed to fetch board data for bus=pci,vendor=168c,device=0040,subsystem-vendor=168c,subsystem-device=0002 from ath10k/QCA99X0/hw2.0/board-2.bin
[   29.215847] ath10k_pci 0001:01:00.0: board_file api 1 bmi_id N/A crc32 7e56fd07
[   31.288467] ath10k_pci 0001:01:00.0: unable to read from the device
[   31.288493] ath10k_pci 0001:01:00.0: could not execute otp for board id check: -110
[   31.293535] ath10k_pci 0001:01:00.0: failed to get board id: -110
[   33.308443] ath10k_pci 0001:01:00.0: unable to read from the device (-110)
[   33.308466] ath10k_pci 0001:01:00.0: could not read board ext data addr (-110)
[   33.314203] ath10k_pci 0001:01:00.0: could not push board ext data (-110)
[   33.321471] ath10k_pci 0001:01:00.0: failed to download board data: -110
[   33.328264] ath10k_pci 0001:01:00.0: failed to run otp: -110
[   33.335085] ath10k_pci 0001:01:00.0: could not init core (-110)
[   33.340836] ath10k_pci 0001:01:00.0: could not probe fw (-110)
```
For both interfaces while compiling with `GLUON_ATH10K_MESH ?= ibss` while the firmware is in place here too.

I've also flashed the binary LEDE image from the dev-snapshot from yesterday (June the 30th):

```
[   14.456287] ath10k_pci 0001:01:00.0: enabling device (0140 -> 0142)
[   14.456394] ath10k_pci 0001:01:00.0: enabling bus mastering
[   14.456945] ath10k_pci 0001:01:00.0: pci irq msi oper_irq_mode 2 irq_mode 0 reset_mode 0
[   14.640823] ath10k_pci 0001:01:00.0: Direct firmware load for ath10k/pre-cal-pci-0001:01:00.0.bin failed with error -2
[   14.640870] ath10k_pci 0001:01:00.0: Falling back to user helper
[   20.465653] ath10k_pci 0001:01:00.0: qca99x0 hw2.0 target 0x01000000 chip_id 0x003b01ff sub 168c:0002
[   20.465689] ath10k_pci 0001:01:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 1
[... cut ...]
[   20.545580] ath10k_pci 0001:01:00.0: board_file api 2 bmi_id 1:2 crc32 08fa09f2
[   21.811260] ath10k_pci 0001:01:00.0: htt-ver 2.2 wmi-op 6 htt-op 4 cal pre-cal-file max-sta 512 raw 0 hwcrypto 1
[   21.901891] ath: EEPROM regdomain: 0x0
[   21.901901] ath: EEPROM indicates default country code should be used
[   21.901908] ath: doing EEPROM country->regdmn map search
[   21.901921] ath: country maps to regdmn code: 0x3a
[   21.901931] ath: Country alpha2 being used: US
[   21.901939] ath: Regpair used: 0x3a
```

NeoRaider, 2017-07-01:
@RubenKelevra The other patches you picked are against the official ath10k in the mac80211 package, but Gluon uses the ath10k-ct driver instead.

RubenKelevra, 2017-07-01:
Ah! This makes sense. :)

rotanid, 2018-01-19:
as this issue is a mixture of different issues, please open separate ones - but only for issues that are NOT present when testing the LEDE/OpenWrt upstream images. if those have the issue, too, please use their bugtracker to report the issues.

Casey1979, 2018-01-25:
I did a short Test and the Archer C 2600 V1.0 works in 11s Mesh so far without Problems, the only thing that needs to be done after compiling and installing is to install the latest Candelatech HTT Firmware instead of the Qualcomm 9980 Wifimodule Firmware.
Tested Firmware: http://www.candelatech.com/downloads/ath10k-10-4/ath10k-fw-beta/firmware-5-ct-non-commercial-full-htt-mgt.bin
Working Router: https://map.hamburg.freifunk.net/#!v:m;n:f4f26d37b770

rotanid, 2018-01-26:
wow, impressive!
though i don't think we can use or recommend this firmware here, at least the name of the file looks to me like we can't include this by default, there might be Gluon users with commercial usecases, no?

rotanid, 2018-01-26:
wow, impressive!
though i don't think we can use or recommend this firmware here, at least the name of the file looks to me like we can't include this by default, there might be Gluon users with commercial usecases, no?
maybe you can try again with the latest community firmware instead of the non-commercial?

Casey1979, 2018-01-26:
Rebooted with this community firmware and still looks ok http://www.candelatech.com/downloads/ath10k-10-4/ath10k-fw-beta/firmware-5-ct-full-htt-mgt-community.bin

rotanid, 2018-01-26:
did you also try firmwares that are already bundled with LEDE/OpenWrt? or does it only work with this very new "beta" fw?
if they are not, we won't include them in Gluon, you would have to take care of integration into OpenWrt, first.

Casey1979, 2018-01-26:
The latest QC firmware used by gluon for the 9980 modules is still 10.4.1.00030-1 which does not work.
Which exact versions of the alternative candelatech firmware are bundled with lede is unknown to me and irrelevant for this demonstration. The point was to demonstrate that it could be fixed with not more than a swap of the firmware for the 9980 wifi modules.
In http://www.candelatech.com/downloads/ath10k-10-4/ath10k-fw-beta/release_notes_5.txt you can see that release 9 of the ct firmware contains some mesh related fixes so it would probably work with that too.
But as gluon uses the ath10k-ct driver for this router and i wanted to have the bugfixes of CT beta 10 also i have chosen that firmware in the HTT version and it worked on first try.
The standard CT version is also untested.
You need to either provide a link or the exact version of firmware you want to get tested.

rotanid, 2018-01-26:
@Casey1979 as i don't have the device, i won't invest much more time.
if you want better support for it in Gluon and OpenWrt, i guess you would have to invest time.

Casey1979, 2018-01-27:
Seems to work Out of the Box if targets/ipq806x is modified as follows:

```
ATH10K_PACKAGES='-kmod-ath10k kmod-ath10k-ct -ath10k-firmware-qca99x0 ath10k-firmware-qca99x0-ct'

# TP-Link

device tp-link-archer-c2600 C2600
packages 

$ATH10K_PACKAGES
```
```
[  174.413768] ath10k driver, optimized for CT firmware, probing pci device: 0x40.
[  174.414387] ath10k_pci 0000:01:00.0: enabling device (0140 -> 0142)
[  174.420025] ath10k_pci 0000:01:00.0: enabling bus mastering
[  174.420491] ath10k_pci 0000:01:00.0: pci irq msi oper_irq_mode 2 irq_mode 0 reset_mode 0
[  174.550173] ath10k_pci 0000:01:00.0: Direct firmware load for ath10k/pre-cal-pci-0000:01:00.0.bin failed with error -2
[  174.550223] ath10k_pci 0000:01:00.0: Falling back to user helper
[  174.601314] firmware ath10k!pre-cal-pci-0000:01:00.0.bin: firmware_loading_store: map pages failed
[  174.601761] ath10k_pci 0000:01:00.0: Direct firmware load for ath10k/cal-pci-0000:01:00.0.bin failed with error -2
[  174.609327] ath10k_pci 0000:01:00.0: Falling back to user helper
[  225.869105] ath10k_pci 0000:01:00.0: Direct firmware load for ath10k/fwcfg-pci-0000:01:00.0.txt failed with error -2
[  225.869148] ath10k_pci 0000:01:00.0: Falling back to user helper
[  225.915712] firmware ath10k!fwcfg-pci-0000:01:00.0.txt: firmware_loading_store: map pages failed
[  226.654629] ath10k_pci 0000:01:00.0: qca99x0 hw2.0 target 0x01000000 chip_id 0x003b01ff sub 168c:0002
[  226.654664] ath10k_pci 0000:01:00.0: kconfig debug 1 debugfs 1 tracing 0 dfs 1 testmode 0
[  226.664798] ath10k_pci 0000:01:00.0: firmware ver 10.4.3-ct-fW-009-cd072c1 api 5 features peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,set-special-CT crc32 9440ad0c
[  226.671384] ath10k_pci 0000:01:00.0: failed to fetch board data for bus=pci,vendor=168c,device=0040,subsystem-vendor=168c,subsystem-device=0002 from ath10k/QCA99X0/hw2.0/board-2.bin
[  226.692255] ath10k_pci 0000:01:00.0: board_file api 1 bmi_id N/A crc32 7e56fd07
[  227.791501] ath10k_pci 0000:01:00.0: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[  227.791557] ath10k_pci 0000:01:00.0: msdu-desc: 2500  skid: 32
[  227.866955] ath10k_pci 0000:01:00.0: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0'
[  227.867565] ath10k_pci 0000:01:00.0: wmi print 'free: 33200 iram: 14692 sram: 16912'
[  228.014660] ath10k_pci 0000:01:00.0: htt-ver 2.2 wmi-op 6 htt-op 4 cal file max-sta 32 raw 0 hwcrypto 1
[  228.128978] ath: EEPROM regdomain: 0x0
[  228.129001] ath: EEPROM indicates default country code should be used
[  228.129018] ath: doing EEPROM country->regdmn map search
[  228.129041] ath: country maps to regdmn code: 0x3a
[  228.129059] ath: Country alpha2 being used: US
[  228.129078] ath: Regpair used: 0x3a
[  228.136932] ath10k driver, optimized for CT firmware, probing pci device: 0x40.
[  228.137955] ath10k_pci 0001:01:00.0: enabling device (0140 -> 0142)
[  228.143287] ath10k_pci 0001:01:00.0: enabling bus mastering
[  228.143881] ath10k_pci 0001:01:00.0: pci irq msi oper_irq_mode 2 irq_mode 0 reset_mode 0
[  228.339181] ath10k_pci 0001:01:00.0: Direct firmware load for ath10k/pre-cal-pci-0001:01:00.0.bin failed with error -2
[  228.339215] ath10k_pci 0001:01:00.0: Falling back to user helper
[  228.548311] firmware ath10k!pre-cal-pci-0001:01:00.0.bin: firmware_loading_store: map pages failed
[  228.548544] ath10k_pci 0001:01:00.0: Direct firmware load for ath10k/cal-pci-0001:01:00.0.bin failed with error -2
[  228.556167] ath10k_pci 0001:01:00.0: Falling back to user helper
[  284.261993] ath10k_pci 0001:01:00.0: Direct firmware load for ath10k/fwcfg-pci-0001:01:00.0.txt failed with error -2
[  284.262027] ath10k_pci 0001:01:00.0: Falling back to user helper
[  284.299955] firmware ath10k!fwcfg-pci-0001:01:00.0.txt: firmware_loading_store: map pages failed
[  284.300119] ath10k_pci 0001:01:00.0: qca99x0 hw2.0 target 0x01000000 chip_id 0x003b01ff sub 168c:0002
[  284.307821] ath10k_pci 0001:01:00.0: kconfig debug 1 debugfs 1 tracing 0 dfs 1 testmode 0
[  284.319371] ath10k_pci 0001:01:00.0: firmware ver 10.4.3-ct-fW-009-cd072c1 api 5 features peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,set-special-CT crc32 9440ad0c
[  284.325272] ath10k_pci 0001:01:00.0: failed to fetch board data for bus=pci,vendor=168c,device=0040,subsystem-vendor=168c,subsystem-device=0002 from ath10k/QCA99X0/hw2.0/board-2.bin
[  284.346263] ath10k_pci 0001:01:00.0: board_file api 1 bmi_id N/A crc32 7e56fd07
[  285.446244] ath10k_pci 0001:01:00.0: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[  285.446270] ath10k_pci 0001:01:00.0: msdu-desc: 2500  skid: 32
[  285.522355] ath10k_pci 0001:01:00.0: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0'
[  285.522985] ath10k_pci 0001:01:00.0: wmi print 'free: 33200 iram: 14692 sram: 16912'
[  285.687812] ath10k_pci 0001:01:00.0: htt-ver 2.2 wmi-op 6 htt-op 4 cal file max-sta 32 raw 0 hwcrypto 1
[  285.801953] ath: EEPROM regdomain: 0x0
[  285.801960] ath: EEPROM indicates default country code should be used
[  285.801966] ath: doing EEPROM country->regdmn map search
[  285.801975] ath: country maps to regdmn code: 0x3a
[  285.801982] ath: Country alpha2 being used: US
[  285.801988] ath: Regpair used: 0x3a
```

rotanid, 2018-01-28:
@Casey1979 that sounds good, have you done short speedtests over the 2,4ghz and 5ghz 11s mesh?
if those were OK in your eyes, i would create a pull request with the changes - and would ask you to test that a last time before we merge it.

Casey1979, 2018-01-28:
I haven't done wireless performancetests yet as i'm still fighting against deauth-, replay- and blockackoutofwindow-attacks on all channels i am trying to use, especially if it has ssids used by freifunk, only topped of by -60dBm noise at times which is all in part logged by professional accesspoints.
Done by the same crazy neigbour that crashed the wr841nv9 and wr1043v2 since the middle of 2015. The reason to this is that they still need the channels for an insane amount of 2.4 and 5G av transmitters.
Because of that ongoing cold war these tests may take some more thinking, time and preparation.
It's a good environment to uncover bugs, but a bad to actually use the hardware as it is meant to be.

Casey1979, 2018-02-07:
Iperf2 on the nodes via 5GHz 11s HT20 channel 165 meshlink between the c2600 and a wdr4900 transfered up to 100mbit/s the wdr4900 cpu was maxed out during the test.
2.4GHz performance is still untested but should perform the same as it uses exactly the same type of chip but the 2.4GHz meshlinks reached uptimes of over 10 days even under these very bad conditions.

rotanid, 2018-03-12:
@Casey1979 please try out the branch with your suggested change:
https://github.com/freifunk-gluon/gluon/tree/c2600

Casey1979, 2018-03-16:
The c2600 ran flawless and the gluon c2600 branch works as expected but it needs to be remembered that i just tested the V1.0 Hardwareversion and 11s mesh, the V1.1 and ibss still needs testing from someone else.

```
root@TL-ARCHER-C2600:~# iw phy
Wiphy phy1
        max # scan SSIDs: 16
        max scan IEs length: 195 bytes
        max # sched scan SSIDs: 0
        max # match sets: 0
        max # scan plans: 1
        max scan plan interval: -1
        max scan plan iterations: 0
        Retry short limit: 7
        Retry long limit: 4
        Coverage class: 0 (up to 0m)
        Device supports AP-side u-APSD.
        Available Antennas: TX 0xf RX 0xf
        Configured Antennas: TX 0xf RX 0xf
        Supported interface modes:
                 * IBSS
                 * managed
                 * AP
                 * AP/VLAN
                 * monitor
                 * mesh point
                 * P2P-client
                 * P2P-GO
                 * P2P-device
        Band 1:
                Capabilities: 0x19ef
                        RX LDPC
                        HT20/HT40
                        SM Power Save disabled
                        RX HT20 SGI
                        RX HT40 SGI
                        TX STBC
                        RX STBC 1-stream
                        Max AMSDU length: 7935 bytes
                        DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: 8 usec (0x06)
                HT TX/RX MCS rate indexes supported: 0-31
                VHT Capabilities (0x339b79b2):
                        Max MPDU length: 11454
                        Supported Channel Width: neither 160 nor 80+80
                        RX LDPC
                        short GI (80 MHz)
                        TX STBC
                        SU Beamformer
                        SU Beamformee
                        MU Beamformer
                        MU Beamformee
                        RX antenna pattern consistency
                        TX antenna pattern consistency
                VHT RX MCS set:
                        1 streams: MCS 0-9
                        2 streams: MCS 0-9
                        3 streams: MCS 0-9
                        4 streams: MCS 0-9
                        5 streams: not supported
                        6 streams: not supported
                        7 streams: not supported
                        8 streams: not supported
                VHT RX highest supported: 0 Mbps
                VHT TX MCS set:
                        1 streams: MCS 0-9
                        2 streams: MCS 0-9
                        3 streams: MCS 0-9
                        4 streams: MCS 0-9
                        5 streams: not supported
[... 116 lines cut ...]
                        * 5620 MHz [124] (27.0 dBm) (radar detection)
                        * 5640 MHz [128] (27.0 dBm) (radar detection)
                        * 5660 MHz [132] (27.0 dBm) (radar detection)
                        * 5680 MHz [136] (27.0 dBm) (radar detection)
                        * 5700 MHz [140] (27.0 dBm) (radar detection)
                        * 5720 MHz [144] (disabled)
                        * 5745 MHz [149] (14.0 dBm)
                        * 5765 MHz [153] (14.0 dBm)
                        * 5785 MHz [157] (14.0 dBm)
                        * 5805 MHz [161] (14.0 dBm)
                        * 5825 MHz [165] (14.0 dBm)
        valid interface combinations:
                 * #{ managed } <= 16, #{ AP, mesh point } <= 16, #{ IBSS } <= 1,
                   total <= 16, #channels <= 1, STA/AP BI must match, radar detect widths: { 20 MHz (no HT), 20 MHz, 40 MHz, 80 MHz }

        HT Capability overrides:
                 * MCS: ff ff ff ff ff ff ff ff ff ff
                 * maximum A-MSDU length
                 * supported channel width
                 * short GI for 40 MHz
                 * max A-MPDU length exponent
                 * min MPDU start spacing
        Device supports VHT-IBSS.

```

cfstras, 2018-03-17:
Have been running the 2017.1.4 release for a few weeks, and noticed frequent wifi disconnects (5 and 2.4). Also sometimes, our two laptops can't find the beacons, while the iPhones claim to be in wifi. Reboot (or wifi down &&wifi) fixes that.
Will recompile on the c2600 branch and report back.

Casey1979, 2018-03-17:
Your wifiproblems maybe caused by the old 9980 firmware used by gluon missing at least 6 months of bugfixes and improvements while i mostly tested 2 months old firmware for the last weeks until i switched to the c2600 branch with the same old beta 9 modulefirmware.
Which hardwareversion and mesh method are you testing?

cfstras, 2018-03-17:
V1.1, no meshing for now.
Still fiddling with the update, something broke. Gonna do a full restore instead of sysupdate next.

cfstras, 2018-03-18:
nope... wasn't the sysupgrade. Flashed fresh image.
It boots ok (still with the hard-reboot bug though), and gets an IP. SSIDs are advertised, but authentication fails. dmesg and link to site-config attached.
https://gist.github.com/cfstras/38ab6eea38f016b15f6e79f37d59cc86

Casey1979, 2018-03-18:
That error is not relevant and normal ath10k behaviour, see this ath10k example from a QCA988X module in a Archer C7V2 working with lede
```
[   11.166832] ath10k_pci 0000:01:00.0: pci irq legacy oper_irq_mode 1 irq_mode 0 reset_mode 0
[   11.574140] ath10k_pci 0000:01:00.0: Direct firmware load for ath10k/pre-cal-pci-0000:01:00.0.bin failed with error -2
[   11.585019] ath10k_pci 0000:01:00.0: Falling back to user helper
[   11.714364] firmware ath10k!pre-cal-pci-0000:01:00.0.bin: firmware_loading_store: map pages failed
[   11.999255] ath10k_pci 0000:01:00.0: Direct firmware load for ath10k/QCA988X/hw2.0/firmware-6.bin failed with error -2
[   12.010142] ath10k_pci 0000:01:00.0: Falling back to user helper
[   12.078514] firmware ath10k!QCA988X!hw2.0!firmware-6.bin: firmware_loading_store: map pages failed
[   12.434791] ath10k_pci 0000:01:00.0: qca988x hw2.0 target 0x4100016c chip_id 0x043202ff sub 0000:0000
[   12.444173] ath10k_pci 0000:01:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 1
[   12.457262] ath10k_pci 0000:01:00.0: firmware ver 10.2.4-1.0-00037 api 5 features no-p2p,raw-mode,mfp,allows-mesh-bcast crc32 a4a52adb
[   12.502285] ath10k_pci 0000:01:00.0: Direct firmware load for ath10k/QCA988X/hw2.0/board-2.bin failed with error -2
[   12.512895] ath10k_pci 0000:01:00.0: Falling back to user helper
[   12.579736] firmware ath10k!QCA988X!hw2.0!board-2.bin: firmware_loading_store: map pages failed
[   12.608502] ath10k_pci 0000:01:00.0: board_file api 1 bmi_id N/A crc32 bebc7c08
[   13.706884] ath10k_pci 0000:01:00.0: htt-ver 2.1 wmi-op 5 htt-op 2 cal file max-sta 128 raw 0 hwcrypto 1
```

rotanid, 2018-03-19:
@Casey1979 my change only builds if 11s is used, so ibss testing is not necessary.
also, the v1.1 seems to be the same hardware es the v1.0 from what i found online, so your tests should be enough.
if your tests are OK *without* the firmware update, we can merge it - but i don't have the time to also backport the ath10k firmware, you would have to provide a proper patch if that is necessary.

Casey1979, 2018-03-19:
I am confident enough for V1.0 hardware to give it a go as it is in the c2600 branch, but i would rather wait until cfstras, RubenKelevra, seligenthaler or some other user gives his ok too.
When it comes to new ct-firmware, that seems to not be possible anymore since march without an ath10k-ct driverbackport too, so id rather leave it as it is.
I have a backup of a working february firmware set though should someone see the need to recheck arising problems with newer ct-firmware.

cfstras, 2018-03-20:
I‘ce had some problems wirh the radios being disabled by default and only now figured you need to set radioX.disabled=0 explicitly. Is there a way to set this in he image?
The last thing I had put on was c2600~1, so without the proposed change. Will test again with c2600 later this week.

rotanid, 2018-03-20:
that sounds weird, there is nothing special to this image which could disable the radios.
maybe you disabled them in an older flash or you did it in the config mode?
please re-flash *without* keeping old settings (using the branch i linked above) and report what happened.

cfstras, 2018-03-30:
Did a few more flashes and always either got a brick with all lights constant-on, or the scenario I had previously described.

rotanid, 2018-03-30:
maybe there is something wrong with your configuration (site.conf) or build environment as @Casey1979 didn't report this problem.

Casey1979, 2018-03-31:
If you gave it enough time to reboot itself after flashing (may take many minutes sometimes) i really dont know what went wrong, but maybe its the hardwarerevision 1.1 still having unsolved problems in the used lede release.
For me with revision 1.0 it worked out of the box but you really have to give it the time it needs sometimes when flashing. Just to test give it up to 15 minutes before you do a powercycle.
Alternatively do a clean flash of the latest lede trunk and upgrade to gluon from there to see if it makes a difference.

cfstras, 2018-03-31:
I always gave it at least 30 minutes.. sometimes it would then „work“ after hard-rebooting, as if it was stuck somewhere in the install.
Right now I just need a wifi router, so I went back to stock for now :/
maybe I’ll get more time in a few weeks, but at this point I’m considering buying somehing with better support and better wifi (aruba? ubiquiti?)

rotanid, 2018-04-23:
let's reopen until there's non-BROKEN support.
at the moment, @kpanic23 is testing the device with a serial console attached

kpanic23, 2018-04-24:
Okay, I have now tested multiple firmwares. Here are my discoveries:

I started with a 2017.1.5 build. Device is booting, crashes on reboot. Client wifi seems to be working, 802.11s mesh isn't. On the status page there's not even mesh listed, only the VPN uplink.
[2017.1.5.log](https://github.com/freifunk-gluon/gluon/files/1941830/2017.1.5.log)

Then I tried building a gluon-next image. Device seemed to lock up after reboot in config mode. After a power cycle the device was bricked. Strangely, using the FFDA firmware seemed to boot, albeit mesh not working, as in above. 
After connecting the serial console, I noticed that the device is spending about 5 minutes "erasing all blocks after the end marker" on the first boot using my firmware, thus delaying everything, the device seems to be locked up. When powercycling in that stage, flash seems to get corrupted, hence the device gets stuck in bootloop.
[next.log](https://github.com/freifunk-gluon/gluon/files/1941829/next.log)

After that, @rotanid suggested trying to flash the latest OpenWrt Snapshot, which I did:
[snapshot.log](https://github.com/freifunk-gluon/gluon/files/1941828/snapshot.log)

Then he told me of his branch (https://github.com/freifunk-gluon/gluon/tree/c2600) using alternative firmware for the ath10k driver (ath10k-firmware-qca99x0-ct), and I cherrypicked his changes onto the latest gluon-next, which seems to run flawlessly, even reboots are working:
[next-alt-fw.log](https://github.com/freifunk-gluon/gluon/files/1941827/next-alt-fw.log)

Then @rotanid suggested applying the patch to gluon master, which seems to run great, albeit suffering from the crash on reboot issue:
[master-alt-fw.log](https://github.com/freifunk-gluon/gluon/files/1941826/master-alt-fw.log)

rotanid, 2018-04-24:
to summarize: flashing takes a very long time and only the gluon next branch, based on openwrt snapshots, is working without problems so far.
if someone wants support for this device in Gluon 2018.1 he/she would have to fix this reboot issue that @kpanic23 observed.

kpanic23, 2018-04-24:
Don't get me wrong, flashing works normally. But on the first boot, it takes quite some time, about 300sec or 5min, to erase the unused flash portion behind the image:
`[   19.117808] jffs2_scan_eraseblock(): End of filesystem marker found at 0x10000`
`[   19.117925] jffs2_build_filesystem(): unlocking the mtd device...`
`[   19.124022] done.`
`[   19.131040] jffs2_build_filesystem(): erasing all blocks after the end marker... `
`[  323.758488] done.`

If you configure the node via the config mode web interface and hit "Save and reboot",the reboot does not occur until that process has finished. If you impatiently cycle the power in that state, the flash seems to get corrupted, resulting in a boot loop.

Casey1979, 2018-05-06:
I upgraded from the c2600 branch to the latest gluon master from 05.05.2018, the C2600 V1.0 needed its 5 to 6 minutes for the sysupgrade and worked flawless afterwards. Tried a reboot and it came back online as expected.

Then i tried release 10 of the ath10k-ct firmware and at first the router came back online but couldnt start the wifimodules, after a hardreboot it came back without problems and wifi modules online, probably some incompatibility between the old and new wifi firmware on softreboots.
Just wanted to confirm that release 10 of the at10k-ct firmware is usuable again if needed.

Casey1979, 2018-05-06:
No, the ath10k-ct firmware is still a prerequisite in every case as the Qualcomm Firmware hasn't been changed for the last 3 years. Maybe i should have mentioned that manual change again.

rotanid, 2018-05-06:
@Casey1979 but the ath10k-firmware-qca99x0-ct included in Gluons master branch is sufficient, you did not need to update it manually from an external source?

Casey1979, 2018-05-06:
@rotanid Yes i had no issues with the gluon master and the compiled in ct-firmware (like in the c2600branch). It worked out of the box with the only thing a little bit unpleasant being the long time it needs to reboot after upgrading probably because of the jffs erasing.
If there are still issues for others they might be caused by v1.1 hardware then.

rotanid, 2018-05-06:
@Casey1979 thanks for your work, i have merged the c2600 branch to include the target/device by default for builds with 11s meshing enabled.
1531571a7e03c505035f4c0b1e7cdb5a5f623d9b

[2 more replies]

closed 2018-05-06 by rotanid