freifunk-gluon/gluon #1091: WDR4300 mesh0 (2.4ghz) does not come up

christf, 2017-04-11:
reproducibly mesh0 does not come up. ifup mesh_radio0 produces
Fri Feb 10 07:14:07 2017 daemon.notice netifd: radio0 (2942): ip: SIOCSIFFLAGS: No error information
Fri Feb 10 07:14:07 2017 daemon.notice netifd: radio0 (2942): Interface 1 setup failed: IFUP_ERROR
in logread.

client0, client1 and mesh1 are up.
setting beacon_int on every wifi-iface in /etc/config/wireless does not change the symptoms 
what can I try to investigate this?

rgbiernat, 2017-04-28:
I can confirm this bug in 2016.2.4. Unfortunately I don't know for sure when it started.

https://map.freifunk-uelzen.de/#!v:m;n:14cc205e5890

I had to swap it with a wr1043 as it was the central uplink.

I can physically access that device as it is only 5km away. How can I help?

Gammelknup, 2017-04-28:
seems to work at gluon-v2016.2.5 with ibss mesh at 2.4 ghz: 
https://map.kbu.freifunk.net/#!v:m;n:e894f63f4bbe

another example at v2016.2.5-1-g03fc7b8 with ibss mesh at 2.4 ghz:
https://map.kbu.freifunk.net/#!v:m;n:e894f6cd3bf6

so maybe fixed in 2016.2.5 or 802.11s related

ambassador86, 2017-04-28:
We have no problems with 802.11s on 2016.2.5 in Chemnitz: https://meshviewer.chemnitz.freifunk.net/#!v:m;n:647002d32160
https://meshviewer.chemnitz.freifunk.net/#!v:m;n:647002e2953e
https://meshviewer.chemnitz.freifunk.net/#!v:m;n:6466b3a23b40

NeoRaider, 2017-04-28:
This bug should occur in master only (and we haven't been able to reproduce it properly yet). If you see something similar, make sure it produces the same log messages, otherwise it is likely a different issue, for example a variant of #1079 (fixed in master)).

closed 2018-11-26 by mweinelt