freifunk-gluon/gluon #1099: Ubiquiti LiteBeam M5 / NanoBeam M5

RubenKelevra, 2017-04-16:
We discovered those two devices which are pretty cheap and have nice buildin antennas for P-t-P (Litebeam) and as AP/Meshpoint (NanoBeam).

Is Gluon support possible for those two?

NeoRaider, 2017-04-18:
If these are AirMax "XM" devices, the bullet image should work. If they are "XW", LEDE support is missing at the moment.

RubenKelevra, 2017-04-18:
Thanks!

We got one LiteBeam M5 installed with stock firmware but it's reachable, so I'll try the Bullet firmware, if we got the right hardware.

Currently we're not using any NanoBeams but they look promising as nanostation-replacement, that's why I asked for them too.

Anything we can do if we got the wrong hardware for the Bullet firmware to help the LEDE development?

FreifunkUFO, 2017-04-20:
its XW. but there is no downgrade-"stock"-firmware to avoid the "new stock firmware bugs" :-(
flashing with freifunk-firmware is working, but you cant save anything.

very crazy discussions at freifunk-forum, so people say its bad to think about openwrt/lede support :-(

https://forum.freifunk.net/t/litebeam-m5-praxistest/12735

NeoRaider, 2017-04-20:
Hmm yes, the missing flash unlock support, I need to look into that. What flash chip does the LiteBeam use?

FreifunkUFO, 2017-04-24:
have a look: 

..dmesg 
[    0.000000] Linux version 4.4.50 (buildbot@builds-02.infra.lede-project.org) (gcc version 5.4.0 (LEDE GCC 5.4.0 r3101-bce140e) ) #0 Mon Feb 20 17:13:44 2017
[    0.000000] MyLoader: sysp=8c84edad, boardp=85b59785, parts=a58521a0
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001974c (MIPS 74Kc)
[    0.000000] SoC: Atheros AR9342 rev 2
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 04000000 @ 00000000 (usable)
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] No valid device tree found, continuing without
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] On node 0 totalpages: 16384
[    0.000000] free_area_init_node: node 0, pgdat 803d04d0, node_mem_map 81000000
[    0.000000]   Normal zone: 128 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 16384 pages, LIFO batch:3
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 16256
[    0.000000] Kernel command line:  board=UBNT-LOCO-XW mtdparts=spi0.0:256k(u-boot)ro,64k(u-boot-env)ro,7552k(firmware),256k(cfg)ro,64k(EEPROM)ro console=ttyS0,115200 rootfstype=squashfs,jffs2 noinitrd
[    0.000000] PID hash table entries: 256 (order: -2, 1024 bytes)
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Writing ErrCtl register=00000000
[    0.000000] Readback ErrCtl register=00000000
[    0.000000] Memory: 60352K/65536K available (3071K kernel code, 165K rwdata, 412K rodata, 312K init, 205K bss, 5184K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS:51
[    0.000000] Clocks: CPU:535.000MHz, DDR:400.000MHz, AHB:200.000MHz, Ref:40.000MHz
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7144898866 ns
[    0.000010] sched_clock: 32 bits at 267MHz, resolution 3ns, wraps every 8027976190ns
[    0.008312] Calibrating delay loop... 266.64 BogoMIPS (lpj=1333248)
[    0.091155] pid_max: default: 32768 minimum: 301
[    0.096214] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.103270] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.113601] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.125386] NET: Registered protocol family 16
[    0.131661] MIPS: machine is Ubiquiti Loco M XW
[    0.367902] clocksource: Switched to clocksource MIPS
[    0.374557] NET: Registered protocol family 2
[    0.380300] TCP established hash table entries: 1024 (order: 0, 4096 bytes)
[    0.387736] TCP bind hash table entries: 1024 (order: 0, 4096 bytes)
[    0.394569] TCP: Hash tables configured (established 1024 bind 1024)
[    0.401454] UDP hash table entries: 256 (order: 0, 4096 bytes)
[    0.407689] UDP-Lite hash table entries: 256 (order: 0, 4096 bytes)
[    0.414704] NET: Registered protocol family 1
[    0.419423] PCI: CLS 0 bytes, default 32
[    0.424247] futex hash table entries: 256 (order: -1, 3072 bytes)
[    0.430881] Crashlog allocated RAM at address 0x3f00000
[    0.453157] squashfs: version 4.0 (2009/01/31) Phillip Lougher
[    0.459431] jffs2: version 2.2 (NAND) (SUMMARY) (LZMA) (RTIME) (CMODE_PRIORITY) (c) 2001-2006 Red Hat, Inc.
[    0.472778] io scheduler noop registered
[    0.476953] io scheduler deadline registered (default)
[    0.482742] Serial: 8250/16550 driver, 16 ports, IRQ sharing enabled
[    0.492674] console [ttyS0] disabled
[    0.516596] serial8250.0: ttyS0 at MMIO 0x18020000 (irq = 11, base_baud = 2500000) is a 16550A
[    0.525793] console [ttyS0] enabled
[    0.533217] bootconsole [early0] disabled
[    0.547394] m25p80 spi0.0: found mx25l6405d, expected m25p80
[    0.553239] m25p80 spi0.0: mx25l6405d (8192 Kbytes)
[    0.558230] 5 cmdlinepart partitions found on MTD device spi0.0
[... 134 lines cut ...]
dev:    size   erasesize  name
mtd0: 00040000 00010000 "u-boot"
mtd1: 00010000 00010000 "u-boot-env"
mtd2: 00760000 00010000 "firmware"
mtd3: 00140000 00010000 "kernel"
mtd4: 00620000 00010000 "rootfs"
mtd5: 003f0000 00010000 "rootfs_data"
mtd6: 00040000 00010000 "cfg"
mtd7: 00010000 00010000 "EEPROM"
root@LEDE:~# cat /tmp/sysinfo/*
loco-m-xw
Ubiquiti Loco M XW
root@LEDE:~# uname -a
Linux LEDE 4.4.50 #0 Mon Feb 20 17:13:44 2017 mips GNU/Linux
root@LEDE:~# cat /etc/openwrt_*
DISTRIB_ID='LEDE'
DISTRIB_RELEASE='17.01.0'
DISTRIB_REVISION='r3205-59508e3'
DISTRIB_CODENAME='reboot'
DISTRIB_TARGET='ar71xx/generic'
DISTRIB_ARCH='mips_24kc'
DISTRIB_DESCRIPTION='LEDE Reboot 17.01.0 r3205-59508e3'
DISTRIB_TAINTS='no-all'
r3205-59508e3
root@LEDE:~#

FreifunkUFO, 2017-04-24:
logread 

Mon Feb 20 17:13:48 2017 kern.notice kernel: [    0.000000] Linux version 4.4.50 (buildbot@builds-02.infra.lede-project.org) (gcc version 5.4.0 (LEDE GCC 5.4.0 r3101-bce140e) ) #0 Mon Feb 20 17:13:44 2017
Mon Feb 20 17:13:48 2017 kern.debug kernel: [    0.000000] MyLoader: sysp=8c84edad, boardp=85b59785, parts=a58521a0
Mon Feb 20 17:13:48 2017 kern.info kernel: [    0.000000] bootconsole [early0] enabled
Mon Feb 20 17:13:48 2017 kern.info kernel: [    0.000000] CPU0 revision is: 0001974c (MIPS 74Kc)
Mon Feb 20 17:13:48 2017 kern.info kernel: [    0.000000] SoC: Atheros AR9342 rev 2
Mon Feb 20 17:13:48 2017 kern.info kernel: [    0.000000] Determined physical RAM map:
Mon Feb 20 17:13:48 2017 kern.info kernel: [    0.000000]  memory: 04000000 @ 00000000 (usable)
Mon Feb 20 17:13:48 2017 kern.info kernel: [    0.000000] Initrd not found or empty - disabling initrd
Mon Feb 20 17:13:48 2017 kern.warn kernel: [    0.000000] No valid device tree found, continuing without
Mon Feb 20 17:13:48 2017 kern.info kernel: [    0.000000] Zone ranges:
Mon Feb 20 17:13:48 2017 kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x0000000003ffffff]
Mon Feb 20 17:13:48 2017 kern.info kernel: [    0.000000] Movable zone start for each node
Mon Feb 20 17:13:48 2017 kern.info kernel: [    0.000000] Early memory node ranges
Mon Feb 20 17:13:48 2017 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x0000000003ffffff]
Mon Feb 20 17:13:48 2017 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000003ffffff]
Mon Feb 20 17:13:48 2017 kern.debug kernel: [    0.000000] On node 0 totalpages: 16384
Mon Feb 20 17:13:48 2017 kern.debug kernel: [    0.000000] free_area_init_node: node 0, pgdat 803d04d0, node_mem_map 81000000
Mon Feb 20 17:13:48 2017 kern.debug kernel: [    0.000000]   Normal zone: 128 pages used for memmap
Mon Feb 20 17:13:48 2017 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Mon Feb 20 17:13:48 2017 kern.debug kernel: [    0.000000]   Normal zone: 16384 pages, LIFO batch:3
Mon Feb 20 17:13:48 2017 kern.warn kernel: [    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
Mon Feb 20 17:13:48 2017 kern.warn kernel: [    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
Mon Feb 20 17:13:48 2017 kern.debug kernel: [    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
Mon Feb 20 17:13:48 2017 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 
Mon Feb 20 17:13:48 2017 kern.info kernel: [    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 16256
Mon Feb 20 17:13:48 2017 kern.notice kernel: [    0.000000] Kernel command line:  board=UBNT-LOCO-XW mtdparts=spi0.0:256k(u-boot)ro,64k(u-boot-env)ro,7552k(firmware),256k(cfg)ro,64k(EEPROM)ro console=ttyS0,115200 rootfstype=squashfs,jffs2 noinitrd
Mon Feb 20 17:13:48 2017 kern.info kernel: [    0.000000] PID hash table entries: 256 (order: -2, 1024 bytes)
Mon Feb 20 17:13:48 2017 kern.info kernel: [    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
Mon Feb 20 17:13:48 2017 kern.info kernel: [    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
Mon Feb 20 17:13:48 2017 kern.info kernel: [    0.000000] Writing ErrCtl register=00000000
Mon Feb 20 17:13:48 2017 kern.info kernel: [    0.000000] Readback ErrCtl register=00000000
Mon Feb 20 17:13:48 2017 kern.info kernel: [    0.000000] Memory: 60352K/65536K available (3071K kernel code, 165K rwdata, 412K rodata, 312K init, 205K bss, 5184K reserved, 0K cma-reserved)
Mon Feb 20 17:13:48 2017 kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
Mon Feb 20 17:13:48 2017 kern.info kernel: [    0.000000] NR_IRQS:51
Mon Feb 20 17:13:48 2017 kern.info kernel: [    0.000000] Clocks: CPU:535.000MHz, DDR:400.000MHz, AHB:200.000MHz, Ref:40.000MHz
Mon Feb 20 17:13:48 2017 kern.info kernel: [    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7144898866 ns
Mon Feb 20 17:13:48 2017 kern.info kernel: [    0.000010] sched_clock: 32 bits at 267MHz, resolution 3ns, wraps every 8027976190ns
Mon Feb 20 17:13:48 2017 kern.info kernel: [    0.008312] Calibrating delay loop... 266.64 BogoMIPS (lpj=1333248)
Mon Feb 20 17:13:48 2017 kern.info kernel: [    0.091155] pid_max: default: 32768 minimum: 301
Mon Feb 20 17:13:48 2017 kern.info kernel: [    0.096214] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
Mon Feb 20 17:13:48 2017 kern.info kernel: [    0.103270] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
Mon Feb 20 17:13:48 2017 kern.info kernel: [    0.113601] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
Mon Feb 20 17:13:48 2017 kern.info kernel: [    0.125386] NET: Registered protocol family 16
Mon Feb 20 17:13:48 2017 kern.info kernel: [    0.131661] MIPS: machine is Ubiquiti Loco M XW
Mon Feb 20 17:13:48 2017 kern.info kernel: [    0.367902] clocksource: Switched to clocksource MIPS
Mon Feb 20 17:13:48 2017 kern.info kernel: [    0.374557] NET: Registered protocol family 2
Mon Feb 20 17:13:48 2017 kern.info kernel: [    0.380300] TCP established hash table entries: 1024 (order: 0, 4096 bytes)
Mon Feb 20 17:13:48 2017 kern.info kernel: [    0.387736] TCP bind hash table entries: 1024 (order: 0, 4096 bytes)
[... cut ...]
Mon Feb 20 17:14:07 2017 authpriv.info dropbear[1153]: Not backgrounding
Mon Feb 20 17:14:18 2017 authpriv.info dropbear[1156]: Child connection from fe80::c66e:1fff:fefe:d886%br-lan:54132
Mon Feb 20 17:14:24 2017 authpriv.notice dropbear[1156]: Auth succeeded with blank password for 'root' from fe80::c66e:1fff:fefe:d886%br-lan:54132
Mon Feb 20 17:14:50 2017 kern.notice kernel: [   75.327939] random: nonblocking pool is initialized
Mon Feb 20 17:14:50 2017 daemon.info urandom_seed[1009]: Seed saved (/etc/urandom.seed)
root@LEDE:~#

NeoRaider, 2017-04-24:
Weird, unlocking of Macronix flash *should* work. There must still be another issue in the flash driver...

FreifunkUFO, 2017-04-25:
i was repeating the tests, but no luck:

- tftp ubnt 5.6.15 and then via webui lede-17.01.1

- tftp ubnt 5.6.15, via webui ubnt 6.0.3, then via webui   lede 17.01.1

FreifunkUFO, 2017-04-25:
but now mtd is slightly different (rootfs-data)

root@LEDE:~# cat /proc/mtd 
dev:    size   erasesize  name
mtd0: 00040000 00010000 "u-boot"
mtd1: 00010000 00010000 "u-boot-env"
mtd2: 00760000 00010000 "firmware"
mtd3: 00140000 00010000 "kernel"
mtd4: 00620000 00010000 "rootfs"
mtd5: 003e0000 00010000 "rootfs_data"
mtd6: 00040000 00010000 "cfg"
mtd7: 00010000 00010000 "EEPROM"
root@LEDE:~#

RubenKelevra, 2017-04-26:
I'm watching you at work looks promising. Hope @neoraider can find the missing piece.

What's up with the NanoBeam M5? Is this the same hardware or might it be worth testing in parallel?

Best regards 

Ruben

FreifunkUFO, 2017-05-04:
im asking at ubnt forum to get a airos 5.5 firmware for litebeam, hopefully they are friendly 
https://community.ubnt.com/t5/airMAX-General-Discussion/5-5-AirOS-firmware-for-Litebeam-M5/m-p/1913405#U1913405

kpanic23, 2017-05-04:
like e.g. http://ftp.ff3l.net/ubiquiti-downgrade/XW.v5.5.10-u2.28005.150723.1358.bin ?

FreifunkUFO, 2017-05-05:
on ubnt they are not willing to help.
@kpanic23 did you link just a "normal" ubnt firmware (which doesnt support this device) or is it a very special built version of ubnt firmware for my device?

FreifunkUFO, 2017-05-26:
crazy recovery hints just for bookmarking it: 
https://community.ubnt.com/t5/UniFi-Wireless/HOWTO-Unbrick-your-UniFi-AP/td-p/338684/page/5

A-Kasper, 2017-05-26:
> crazy recovery hints just for bookmarking it:
> https://community.ubnt.com/t5/UniFi-Wireless/HOWTO-Unbrick-
> your-UniFi-AP/td-p/338684/page/5

as long as i know you need to downgrade to 5.5 first after that you are
ready to flash gluon

FreifunkUFO, 2017-05-29:
please read other comments: atm there is no airos 5.5 version to downgrade!

FreifunkUFO, 2017-11-26:
now there is a new try, and it was successfully for me :-)
 i had to use ubnt 5.6.4 and was able to flash via webui.

https://patchwork.ozlabs.org/patch/828095/

now i have:

```
root@211-31-litebeam:~# cat /proc/mtd
dev:    size   erasesize  name
mtd0: 00040000 00010000 "u-boot"
mtd1: 00010000 00010000 "u-boot-env"
mtd2: 00760000 00010000 "firmware"
mtd3: 00170000 00010000 "kernel"
mtd4: 005f0000 00010000 "rootfs"
mtd5: 003f0000 00010000 "rootfs_data"
mtd6: 00040000 00010000 "cfg"
mtd7: 00010000 00010000 "EEPROM"

```

rotanid, 2018-01-19:
@FreifunkUFO it doesn't look like the patch you linked made it into the OpenWrt codebase yet. do you know its status?

@NeoRaider according to the LEDE TOH, the nanobeam m5 xw is supported using the nano loco m5 xw image:
https://lede-project.org/toh/hwdata/ubiquiti/ubiquiti_nanobeam_m5
if this is correct, should we add an alias or something?

NeoRaider, 2018-01-20:
Current version of the Litebeam support patch: http://patchwork.ozlabs.org/patch/862262/

I guess we can add an alias guarded by a BROKEN check.

rotanid, 2018-01-20:
@RubenKelevra @FreifunkUFO the NanoBeam M5 is now "supported" (the loco m5 xw image probably would have worked before) but currently marked as BROKEN until there are reports that it is working

mweinelt, 2018-11-26:
Litebeam M5 is part of OpenWrt 18.06 since 2018/03/02, so if anyone has the device we can work on integrating it.

https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=d0988235dd277b9a832bbc4b2a100ac6e821f577

FreifunkUFO, 2018-12-14:
this ticket IS about the interest to integrate this device, just like other tickets about hardware.
so, why do you close this ticket?

FreifunkUFO, 2019-01-04:
@bobcanthelpyou of course yes, i own such a device. see screenshots above. so there is no need to close that ticket, its a "normal" ubnt device

bobcanthelpyou, 2019-01-04:
@FreifunkUFO could you create merge request please?

FreifunkUFO, 2019-01-04:
yes, if there is some code i could merge, test or pull-request it.

i can't see any difference to other "new" hardware, where some gluonners start issues (here on github) about supporting it. please reopen that issue-ticket (or: why do you want to see this ticket closed? its a "normal" ubnt- airmax-device)

mweinelt, 2019-01-04:
Apply this change:

```
diff --git a/targets/ar71xx-generic b/targets/ar71xx-generic
index 96b19d8f..1d0ac529 100644
--- a/targets/ar71xx-generic
+++ b/targets/ar71xx-generic
@@ -266,6 +266,8 @@ device ubiquiti-rocket-m ubnt-rocket-m
 alias ubiquiti-rocket-m2
 alias ubiquiti-rocket-m5

+device ubiquiti-litebeam-m5 ubnt-lbe-m5
+
 device ubiquiti-nanostation-m ubnt-nano-m
 alias ubiquiti-nanostation-m2
 alias ubiquiti-nanostation-m5
```

And build with:

```
make -j$(nproc) GLUON_TARGET=ar71xx-generic DEVICES=ubiquiti-litebeam-m5
```

mweinelt, 2020-03-03:
@FreifunkUFO Another year has passed. I'm closing this. Please speak up if you intend to test this still.

closed 2020-03-03 by mweinelt

FreifunkUFO, 2020-03-06:
Yes, thank you for the Openwrt-Support on Litebeam M5.
So its a now a normal/another ubnt device.

mweinelt, 2020-03-06:
I'm not sure what that means. Is the LBE-M5 part of the ath79 target? Do you íntend to create a pull request to add Gluon support?

suppenkasper0815, 2020-12-04:
Hey Guys,

what's about the integration? I found offical support for LiteBeam M5 on openwrt device page. There is an offical sysupgrade and factory image, but of course it won't load via web-ui. 

Now I used SCP to copy newest openwrt image into /tmp and tried it via ssh:

`XW.v6.1.7# fwupdate.real -m fwupdate.bin -d
Found mtd block: /dev/mtd0(u-boot)
Found mtd block: /dev/mtd1(u-boot-env)
Found mtd block: /dev/mtd2(kernel)
Found mtd block: /dev/mtd3(rootfs)
Found mtd block: /dev/mtd4(cfg)
Found mtd block: /dev/mtd5(EEPROM)
Got U-Boot variable: mtdparts = mtdparts=ath-nor0:256k(u-boot),64k(u-boot-env),1024k(kernel),6528k(rootfs),256k(cfg),64k(EEPROM)
Adding U-Boot partition: u-boot 9F000000 00040000
Adding U-Boot partition: u-boot-env 9F040000 00010000
Adding U-Boot partition: kernel 9F050000 00100000
Adding U-Boot partition: rootfs 9F150000 00660000
Adding U-Boot partition: cfg 9F7B0000 00040000
Adding U-Boot partition: EEPROM 9F7F0000 00010000
Calculating flash size:
Adding block: /dev/mtd0("u-boot") - size: 00040000
Adding block: /dev/mtd1("u-boot-env") - size: 00010000
Adding block: /dev/mtd2("kernel") - size: 00100000
Adding block: /dev/mtd3("rootfs") - size: 00660000
Adding block: /dev/mtd4("cfg") - size: 00040000
Adding block: /dev/mtd5("EEPROM") - size: 00010000
Total flash size: 00800000
Flash start: 9F000000
Flash end: 9F800000
Header MAGIC 'UBNT'
Current: XW.ar934x.v6.1.7.32555.180523.1754

New ver: XW.ar934x.v5.6.15.30572.170328.1052
Versions: New(329231) 5.6.15, Required(328971) 5.5.11
FW Part: "u-boot"(1), MAGIC: 'PART', Base: 0x9F000000, DLen: 0x00037CAC, PLen: 0x00040000
FW Part: "kernel"(2), MAGIC: 'PART', Base: 0x9F050000, DLen: 0x000E9789, PLen: 0x00100000
FW Part: "rootfs"(3), MAGIC: 'PART', Base: 0x9F150000, DLen: 0x005E0000, PLen: 0x00660000
FW Part: "script"(4), MAGIC: 'EXEC', Base: 0x00000000, DLen: 0x00006BBD, PLen: 0x00006BBD
Bad Image Structure
Signature check failed
XW.v6.1.7# fwupdate.real -m fwupdate.bin -d -f
Bad Image Structure
Signature check failed
XW.v6.1.7# fwupdate.real -m fwupdate.bin -d -force
Bad Image Structure
Signature check failed
XW.v6.1.7# 
`

I also did this with official Firmware XW.v6.1.7.32555.180523.1754.bin

Here I saw much more ongoing and I guess there was a real flashing because the device runs now 6.1.7.

Unfortunately I have no serial access. I don't know how to open the feedhorn and get it waterproof after opening again. - So as far as I understood in some other discussions we need a beta-image (I registred and accept terms for beta, but couldn't find betaimages at ubiquiti-pages), flash an unsigned old image (offically available on OEM-site) and from there flash the openwrt*.factory image. A direct flash of unsigned images don't work, it stops with error code 43.

I would love if we could get this device to work under openwrt because I wanna use wiregurd with this device and I don't know how to install this on OEM-FW. I guess no chance for it ;) 

Any experience for performance with this device under openwrt?

The device-page in openwrt:

[https://openwrt.org/toh/hwdata/ubiquiti/ubiquiti_litebeam_m5](url)