freifunk-gluon/gluon #1100: WiFi crashes on TL-WR1043NDv2

belzebub40k, 2017-04-17:
With LEDE we see crashes of the WiFi interfaces on two TL-WR1043NDv2. From one them user was able to retrieve the dmesg and logread output. But they don't look very helpful. I there anything else we could check when it happens the next time?

[logread](https://p.mufff.in/yoqugefuqo.hs)
[dmesg](https://p.mufff.in/ezizojubaz.hs)
[ip addr show](https://p.mufff.in/oguheziqof.hs)

NeoRaider, 2017-04-18:
Hmm, is this batman-adv compat 14 or 15? I suspect there is a race condition in batman-adv that breaks reference counting when interfaces are quickly removed and re-added.

Is this before or after e45c30330d2eede2cec39c3082ad5e6bdc40a6ba?

belzebub40k, 2017-04-18:
It is compat 15 and the build was before the mentioned commit. I'll rebuild with the latest master.

NeoRaider, 2017-04-18:
I assume e45c303 will hide the issue, but there is still an underlying issue in the kernel that should be fixed at some point.

closed 2018-01-19 by rotanid