freifunk-gluon/gluon #1102: optionally phone home during build process

rubo77, 2017-04-24:
Maybe we can add an option in the build-process to optionally send stats of the build process automatically to a collecting server, so you will get an email each time someone on earth builds a firmware:

> If you are building for an active community please consider setting this option to "spread the word" of your community existing

RubenKelevra, 2017-04-24:
I strongly disagree I don't want my buildserver to send any "stats" to anyone.

Everybody who wants to share information has various ways to do this exactly how they want. There's no need for additional informationtransfers from highly secured systems.

Adorfer, 2017-04-24:
@rubo77  If you consider the rules of the existing freifunk.api gatekeepers as too strict/too prohibitive, then their rules or way of their interpretation should be adjusted to the level you would accept as "make it easy for new communities to get listed".

rubo77, 2017-04-24:
Sure, I agree, there shouldn't be **in no way** be sent any information from a highly secure buildserver without question.

I thought of an **optional option** so maybe during the build proces, there could popup this question: 

> Do you want to send the community information to the freifunk-api? [y/N]

The information could be collected and presented as a link with the information in the GET vars of the URL string, so you can send it from your own Browser.

This would make it much easyer to add data to the Freifunk-Api

NeoRaider, 2017-04-24:
Such a feature is very unusual for open-source software, and I don't see any reason why you would want this.

Adorfer, 2017-04-24:
> This would make it much easyer to add data to the Freifunk-Api

let's assume this feature would exist, please specify: 

- what data should be transmitted
- how/where should the data be stored
- how should have read access to that data

and perhaps: 
- how to screen data between "some testing builds / dry runs" and "official firmwares by the maintainers of a domain". and "some people making fun of trying out some disfunctional settings, just for learning purposes". 

In other words: What would you deduct from the submitted data, exept the fact that some system has submittet some octets with no link to physical reality? 
(this is ra REAL question, not just reasoning for fun)

RubenKelevra, 2017-04-24:
Additionally, most build-environments are usually fully automated, so there's no chance for any "popup".

The question for me is: how would this improve the community communication?

If we want stats, there's a way to just publish the clone stats of this repo.

If we want to collect information about the build process and/or hatdware specs, I think nobody need those infos.

If we want to collect community infos: just guide them to a place where we collect them, like api.

rubo77, 2017-04-24:
Maybe that would be enough: echo a link at the end of the make process that leads to the Freifunk API with the community information that can be automatically extracted from the site.conf inside the link, so the API form will be pre-filled with that data already.

NeoRaider, 2017-04-24:
I don't understand what a firmware build has to do with the Freifunk API. Certainly running nodes are much more interesting than some kind of build?

Also keep in mind that Gluon is not used for Freifunk only, but also other mesh networks.

Adorfer, 2017-04-25:
> Maybe that would be enough: echo a link at the end of the make process that leads to the Freifunk API with the community information that can be automatically extracted from the site.conf inside the link, so the API form will be pre-filled with that data already.

So you are basically looking for a script, extracting site.conf (perhaps site.mk, don't know) to prefill api. 
That is nothing that is tied to the build process. not at all! That's a separate thing you could offer on nearly any website (or integrate on the api website, to read and interpret the site.conf). 

> Also keep in mind that Gluon is not used for Freifunk only, but also other mesh networks.
And there is Freifunk-Networks which are not accepted by the freifunk-api.

H4ndl3, 2017-04-27:
I vote for closing this issue, since gluon isn't used for Freifunk only. Also, if implemented, this could be seen as some kind of "advertisement" which could annoy users, even though it has good intentions.

NeoRaider, 2017-04-27:
Agreed.

closed 2017-04-27 by NeoRaider