freifunk-gluon/gluon #1110: Problems with wifi channel 1 on TP-Link TL-WR1043ND v4

H4ndl3, 2017-04-27:
A user from our community [reported a problem](https://forum.freifunk-muensterland.de/t/tp-link-1043-wifi-signal-verschwindet/2655) with his 1043ND v4. When selecting channel 1, the wifi can't be found by any devices, when selecting another channel (in this case 5) the wifi seems to work as expected. I can't help with debugging this because I don't own a 1043NDv4 but thought I'd open this issue so others are aware of it and could help.

FFPeter, 2017-04-27:
The problem depends on the place where the nodes are installed. When the node ist installed in another building, the problem is gone. If I install another node at the same place, the new node has the same problem. 

I switched to chan 1 on: Thu Apr 27 21:04:43 2017
Switched to chan 5 on: Thu Apr 27 21:05:17 2017
[logread_chan5_634-2.txt](https://github.com/freifunk-gluon/gluon/files/962659/logread_chan5_634-2.txt)

Maybe, this screencast is useful to: https://youtu.be/L_S2vjwa9iQ

FFPeter, 2017-04-27:
[dmesg_chan5_634-2.txt](https://github.com/freifunk-gluon/gluon/files/962680/dmesg_chan5_634-2.txt)

NeoRaider, 2017-04-27:
The output of iwinfo and dmesg/logread on channel 1 would be helpful.

FFPeter, 2017-04-27:
This is the output of iwinfo on channel 1:

```
client0   ESSID: "Freifunk"
          Access Point: 22:13:9F:33:3D:88
          Mode: Master  Channel: 1 (2.412 GHz)
          Tx-Power: 14 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -61 dBm
          Bit Rate: unknown
          Encryption: none
          Type: nl80211  HW Mode(s): 802.11bgn
          Hardware: unknown [Generic MAC80211]
          TX power offset: unknown
          Frequency offset: unknown
          Supports VAPs: yes  PHY name: phy0

ibss0     ESSID: "02:d1:11:37:fc:83"
          Access Point: 02:D1:11:37:FC:83
          Mode: Ad-Hoc  Channel: 1 (2.412 GHz)
          Tx-Power: 14 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -61 dBm
          Bit Rate: unknown
          Encryption: unknown
          Type: nl80211  HW Mode(s): 802.11bgn
          Hardware: unknown [Generic MAC80211]
          TX power offset: unknown
          Frequency offset: unknown
          Supports VAPs: yes  PHY name: phy0

```
Uploaded logread/dmesg contains both cases. I switched to chan 1 on: Thu Apr 27 21:04:43 2017

NeoRaider, 2017-04-27:
Whoa, -61 dBm noise... if that value is accurate, it could explain your issue (I have around -95 dBm on my nodes here). You could try using a software like HORST to check for the available airtime.

FFPeter, 2017-04-27:
I can see the problem in the basement of that building to. 
On my mobile fone I only saw one ssid on chan 5.
This is iwinfo of channel 1 on that node:

```
client0   ESSID: "Freifunk"
          Access Point: DA:82:C1:86:7A:10
          Mode: Master  Channel: 1 (2.412 GHz)
          Tx-Power: 14 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -54 dBm
          Bit Rate: unknown
          Encryption: none
          Type: nl80211  HW Mode(s): 802.11bgn
          Hardware: unknown [Generic MAC80211]
          TX power offset: unknown
          Frequency offset: unknown
          Supports VAPs: yes  PHY name: phy0

ibss0     ESSID: "02:d1:11:37:fc:83"
          Access Point: 02:D1:11:37:FC:83
          Mode: Ad-Hoc  Channel: 1 (2.412 GHz)
          Tx-Power: 14 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -54 dBm
          Bit Rate: unknown
          Encryption: unknown
          Type: nl80211  HW Mode(s): 802.11bgn
          Hardware: unknown [Generic MAC80211]
          TX power offset: unknown
          Frequency offset: unknown
          Supports VAPs: yes  PHY name: phy0
```

Maybe there´s a strong interference frequency around. But I even could see any Freifunk SSID on channel 1 being only one meter next to the node.

RubenKelevra, 2017-04-28:
This doesn't look like a software or hardware issue but just a bad RF environment.

Please recheck your findings on "greenfield" means a low RF noise environment like a bunker or a deep basement.

FFPeter, 2017-04-29:
I wonder that the wifi signal is complete lost, even if the mobile phone is next to the node. In another building the rx signal at the client is at maximum, but every few minutes the signal is completly lost for a few seconds.
Do you now if there´s a feature that shuts off the tx signal on the node if the noise level is over a specific limit?

RubenKelevra, 2017-04-29:
That on the one hand, wifi drivers got two thresholds, one for "other wifi data" and one for "noise" for their listen before talk.

A speciality of 2.4 GHz wifi is the overlapping channels, which is eliminated on 5 GHz wifi. The issue on 2.4 Ghz is, if somebody is sending wifi data on channel 2, 3 or 4, the possibility is high, that your driver detect this as noise and might try to submit its data over the data of the other network.

This causes the major issues on this frequency band, because both networks need to transmit data often twice.

If you got a Nanostation or similar for 2.4 GHz available, you might want to scan the frequency band for the noise and select a differend channel from 1, 5, 9, 13 for Europe and 1, 6, 11 for US/Canada.

And yes, it's normal that your wifi data will be unreadable after a few meters on strong RF noises.

I had that issue once, after 2 meters - no SSID. 

The RF noise in this case was two boxes with 20 desktop mics each. All turned on and transmitting while they were charging... I guess someone thought that red light on the mics must be the charging light...

rotanid, 2017-05-11:
as this apparently isn't an issue with gluon or OpenWrt/LEDE, i'll close the issue.

closed 2017-05-11 by rotanid