freifunk-gluon/gluon #1111: Additional email field for tech-c

RubenKelevra, 2017-04-28:
Currently we got one email field, which seem to be enough for the most cases, but we often got places where we have an administrative contact as well as a technical one. It would be nice to add both email addresses,

We plan to notify the technical contact with technical issues while the administrative one just get information mails about the project with less technical informations.

We already implemented this, do I want to get a small feedback from you, if we should polish it a bit more and add it as pull request or maintain it along with the project?

jplitza, 2017-04-29:
Do I have to enter my address twice then if I am administrative and technical contact?

Besides the issues of whether you are even allowed to send automated or mass mailings to the addresses entered in nodes (see #851), I think this is over-engineering for quite rare situations, and simply entering both addresses into one field should suffice.

RubenKelevra, 2017-04-29:
Well, our tech-c field is fully optional, if filled it's used if not it's not used. Mandatory is only the contact field.

RubenKelevra, 2017-04-29:
I don't think abusing a text field is the right solution in this case. We're already sending confirmation mails and the handling of mailadress informations is not part of this proposal.

If you don't see a usecase, fine, but we do have one ans it's currently used often on outside-installations and installations of large scale.

jplitza, 2017-04-29:
I don't feel like entering two addresses into a free-format field that doesn't even specifically say "e-mail" (I always enter my XMPP address as well, and implemented software to even parse Twitter handles in there) is abuse.

And you specifically asked for feedback, so I shared my thoughts. I'm not disputing you have a use case, otherwise you wouldn't have built the feature. I'm just saying I, personally, don't see the need, because that's what I thought you asked for, opinions. Sorry if I misunderstood.

RubenKelevra, 2017-04-29:
Oh well, sorry for the misunderstanding and the harsh wording. :)

I think the most users use the contact information for mail, we never had a user which used it different, but I guess we had changed the description a bit. The Pico-Peering agreement states that users need to supply an email address, that's why we require the field to be filled with a valid email address.

We send a confirmation mail to the given mail-address and after a click on a link we enable the vpn-access. This is also required as we are ISP in Germany, and if we have "not-specified" number of "customers", different rules apply, like if we had more than 10,000 "customers".

Since we're unsure if members or vpn-users are counted here, we simply add this "registration" to have always a valid number of "customers" below 10k.

I think, a node is part of the network similar to a ip network is part of the internet and it's common to supply a tech-c and a administration contact on a ip network. 

Since the second field is fully optional, I think it's not breaking your usecase either, as well as this will be supplied in a additional package, which you can simply not compile. :)

rotanid, 2018-01-19:
as this seems to be a rather special use case, i suggest you provide a PR where this feature is not activated by default (it has to be set in site.conf).

closed 2018-01-19 by rotanid