freifunk-gluon/gluon #1112: Additional field for postal address

RubenKelevra, 2017-04-28:
Currently we have two fields: lat and lon.

I think the most users get first in contact with the need for exact geolocations in our project, because of these fields.

Additionally it's difficult to flash a node and still get a working internet connection on the computer, often troubles users with a workstation with no wifi.

So we thought about the issues and most of the time the user would be able to insert a postal address for the node.

So the easiest process for a configuration mode would be an additional field where we ask for a full address, ans a simple query to simple web service which then returns the lat/lon to the node.

The postal address can be removed after the lat/lon is stored because it's not longer needed.

We added such a field a while ago and it's used regularly by the users. So we was wondering if we should send a pull-request for the change or if we should maintain our patch along.

rubo77, 2017-04-28:
This would be cool to have it as an optional package.

RubenKelevra, 2017-04-29:
We like to provide the field itself as optional package, but the functionality of requesting the position from our servers and removing the postal address would be too specific and depend on the local server infrastructure.

We don't want to offer a public service nor do we want to ask a public API because this would leak the address plus the IP of the node to the provider of the service. 

That's why we plan to use small local query API as proxy.

jplitza, 2017-04-29:
This sounds paradoxical: You are worried about leaking a postal address to the lookup service, but are yelling the coordinates returned into the public?

If I were to implement this, I'd probably ask Nominatim, maybe using a Reverse Proxy provided by the community if available.

RubenKelevra, 2017-04-29:
> This sounds paradoxical: You are worried about leaking a postal address to the lookup service, but are yelling the coordinates returned into the public?
> 
Well, we do offer coords for the node, because the user want them public, but the ip address of the node should not be exposed together with the address to a third party.

I don't think that sounds paradoxical.

> If I were to implement this, I'd probably ask Nominatim, maybe using a Reverse Proxy provided by the community if available.
> 
Yeah, that was our intention. :)

jplitza, 2017-04-29:
Again: The node will be *publicly* announcing the coordinates together with its IP address. How is this not revealing that data to third parties? Or do I misunderstand and you're arguing that the postal address is more confidential than the coordinates?

RubenKelevra, 2017-04-29:
Our map does not publish the IP of the WAN-connections, but if we would ask a public service, this would expose it.

We do not offer our nodes a internet connection via mesh-access at the moment, I think that's the difference here. :)

There might be also some opinions that a postal address is more specific than a lat/lon point, because we could do some fuzzing to the coordinates, but this wasn't my point. :)

Well, since there were a emotional discussion about the freifunk padserver about the need of an robot.txt to forbid any indexing of the content, while the pads are still browseable, I think many people see a difference between indexable information and non indexable.

Our map-data can be accessed but is not indexed. If we ask a service like google maps for the lat/lon we would actively help to create a profile. :)

jplitza, 2017-04-29:
Indeed, the misunderstanding was that I assumed the lookup will happen with via mesh. As of now, only the VPN tunnel itself is supposed to use the WAN connection in Gluon, everything else should use mesh.

RubenKelevra, 2017-04-30:
This doesn't work for us because the mail address has to transferred prior to a VPN connection cause the registration is done by validating the email address. :)

jplitza, 2017-04-30:
What does the email address have to do with this feature? o_O

RubenKelevra, 2017-04-30:
Nothing, just explained why it's needed to submit data outside the mesh-network in our setup.

rotanid, 2018-01-19:
as this seems to be a rather special use case, i suggest you provide a PR where this feature is not activated by default (it has to be set in site.conf).

closed 2018-01-19 by rotanid