freifunk-gluon/gluon #1114: Support for Unifi AP AC LR

RubenKelevra, 2017-04-30:
The Unifi AP AC Lite as well as the Unifi AP AC Pro got support somewhat a year ago, but the Unifi AP AC LR is still not listed.

Tha hardware looks pretty similar, is support possible? :)

NeoRaider, 2017-05-02:
The OpenWrt Wiki is very unclear on this. The hardware seems to have specs similar to the UAP AC Lite (1 Ethernet port, no USB), but the text later mentions that the Pro image should be used... The link to the image for the UAP AC LR is also incorrect.

If you can find out which LEDE image to use on the UAP AC LR, or if none of the existing images work, we can work on the Gluon support.

RubenKelevra, 2017-05-02:
Alright, thanks I'll get some hardware for testing. :)

PR #1332 (2018-02-20): ar71xx-generic: set unifi-ac-lr alias on ac-lite
Add alias for Ubiquiti unifi AP AC LR

close #1114
closed 2018-02-20 by rotanid