freifunk-gluon/gluon #1115: Can't build actual master

mokkin, 2017-05-01:
I can't build the actual master due to an make error. This happens on two servers I successfully used before. (Debian 8 / Ubunt 16.04)
```user@server0:~$ git clone https://github.com/freifunk-gluon/gluon.git 1.0-20170501 -b master
Klone nach '1.0-20170501'...
remote: Counting objects: 19879, done.
remote: Compressing objects: 100% (77/77), done.
remote: Total 19879 (delta 41), reused 0 (delta 0), pack-reused 19799
Empfange Objekte: 100% (19879/19879), 6.66 MiB | 3.44 MiB/s, Fertig.
Löse Unterschiede auf: 100% (7767/7767), Fertig.
Prüfe Konnektivität... Fertig.
user@server0:~$ cd 1.0-20170501/
user@server0:~/1.0-20170501$ git clone https://github.com/freifunk-leipzig/freifunk-gluon-leipzig.git site -b 2017.1
Klone nach 'site'...
remote: Counting objects: 544, done.
remote: Compressing objects: 100% (8/8), done.
remote: Total 544 (delta 0), reused 0 (delta 0), pack-reused 536
Empfange Objekte: 100% (544/544), 98.27 KiB | 0 bytes/s, Fertig.
Löse Unterschiede auf: 100% (259/259), Fertig.
Prüfe Konnektivität... Fertig.
user@server0:~/1.0-20170501$ make update
--- Updating module 'lede' ---
Initialized empty Git repository in /home/user/1.0-20170501/lede/.git/
fatal: read error: Connection reset by peer
Makefile:31: recipe for target 'update' failed
make: *** [update] Error 128
```

jplitza, 2017-05-01:
I had the same problem yesterday. The git repository server of the LEDE project seems to be broken at the moment. You can work around the problem by replacing `git://git.lede-project.org` by `https://git.lede-project.org` in the `modules` file or manually fetching the commit mentioned in there from the https URL.

Since this is not really a bug in Gluon, I'm closing the issue here.

closed 2017-05-01 by jplitza

mokkin, 2017-05-01:
Thanks @jplitza this has solved me issue but now I'm getting a new one:
```
walter@freifunkvm:/srv2/gluon/1.0-20170501$ make -j6 GLUON_TARGET=ar71xx-generic|
make[1]: Entering directory '/srv2/gluon/1.0-20170501/lede'                     |
make[2]: Entering directory '/srv2/gluon/1.0-20170501/lede'                     |
make[2]: Leaving directory '/srv2/gluon/1.0-20170501/lede'                      |
Collecting package info: package/feeds/packages_luci/luci-app-freifunk-diagnosti|
Collecting package info: package/feeds/packages_luci/luci-app-freifunk-policyrou|
Collecting package info: package/feeds/packages_openwrt/perl-file-sharedir-insta|
Collecting package info: done                                                   |
#                                                                               |
# configuration written to .config                                              |
#                                                                               |
make[1]: Leaving directory '/srv2/gluon/1.0-20170501/lede'                      |
Configuration failed: unable to enable package 'gluon-alfred'                   |
Makefile:90: recipe for target 'config' failed                                  |
make: *** [config] Error 1                                                      |
```
Do you have any idea why he cannot enable alfred?

NeoRaider, 2017-05-02:
I've updated our modules to use https.

Please provide the *lede/.config* and *lede/tmp/.config-package.in* of your build tree.

FreifunkUFO, 2017-05-02:
@NeoRaider here it is: http://gadow.freifunk.net:8004/srv2/gluon/1.0-20170501/

NeoRaider, 2017-05-02:
Your site.mk mentions the non-existing package *gluon-mesh-batman-adv*, which is the name of a source package, but not a binary package. The correct names are still *gluon-mesh-batman-adv-14* and *gluon-mesh-batman-adv-15*.

FreifunkUFO, 2017-05-02:
ok, thanks.