freifunk-gluon/gluon #1116: stop haveged if not needed

rubo77, 2017-05-02:
The haveged uses 3% of Ram on 32MB devices although it is only needed during the key generation in config mode and in normal mode only right after startup for the session keys.

There is a patch that stops haveged in normal mode on devices with only 32MB RAM: https://github.com/Freifunk-Rhein-Neckar/ffrn-packages/tree/master/ffrn-lowmem-patches

This could be enhanced, so it only stops haveged after the generation of the SSH-Session keys and only after fastd was started and running.

A-Kasper, 2017-05-02:
I wonder if we really need to have an local entropy source.. maybe we
could use external entropy sources:

https://bazaar.launchpad.net/~pollinate/pollinate/trunk/view/head:/pollinate

https://entropy.ubuntu.com

belzebub40k, 2017-05-02:
If we use an external entropy source it would require a internet connection during the initial setup of the node.

NeoRaider, 2017-05-02:
Right, a Gluon node must be expected to be without internet connection.

I don't see a point in this. haveged is already stopped by the autoupdater, and we're not particularly tight in RAM in any other situation. All reports of OOM situtations I've seen are caused either by kernel bugs, or by memory fragmentation; none of these can be avoided by stopping a single service that uses a static piece of memory.

rubo77, 2017-05-02:
I thought there are OOM problems on small 841n Models, if a node is in a so called "mesh-hell", an area with lots of mesh connections, many of them really weak.

rotanid, 2017-05-02:
and because of that @rubo77 , you don't fix the problem by disabling some unrelated service, you would just postpone the issue and a few weeks or months later you run into the OOM again.

NeoRaider, 2017-05-02:
I doubt disabling haveged will have any noticable effect; as mentioned, it uses just a static piece of memory.

Kernel data structures, on the other hand, can cause a significant amount of memory fragmentation. Recent changes to batman-adv (merged to compat 14 in https://github.com/freifunk-gluon/batman-adv-legacy/pull/8, in compat 15 since 2016.4) should significantly reduce the space needed by batman-adv's data structures, and also prevent memory fragmentation (but there are likely other places in the kernel that can cause memory fragmentation under certain workloads).

rubo77, 2017-05-02:
We have the problem in Burg with lots of mesh links: https://mesh.nord.freifunk.net/#!v:m;l:60e327e7356e-f4f26d3fef3a where routers stop working correctly, so our eulenfunk-patch restarts them avery half an hour or less.

The admin there sais, it helps already a bit to disable ssh on the nodes, so it seems the RAM is the problem.

If haveged really is not needed during normal run, it would be useful to stop it then. It would not hurt and you have some MB more tolerance before you get OOM problems.

So my question: is it still needed during normal run?

NeoRaider, 2017-05-02:
I wouldn't say it's strictly required, but it's a good idea to have it running for a bit after boot to get the entropy pool initialized properly.

But still, you're working on symptoms rather than the actual cause of the OOM issue. Of course randomly stopping services will slightly improve the situation, but it's just a bad workaround, and will only delay the OOM. Please rather test if the new batman-adv in the Gluon master improves your situation; if you still get OOMs, we need to investigate what actually causes the issues.

A-Kasper, 2017-05-02:
I can't ggree with this "the OOM just happens later" we are talking about
the Network size the node is able to handle without going unstable. Yes...
you are right, that you can just make this stable netqork size bigger and
that the problems will appear again in bigger Setups... but that's the
clue: in bigger setups. speaking: you can use cheaper devices longer....

It's nothing to be discussed against each other. we are working with
limited ressources. we should use all optimizations we've are able to

rotanid, 2017-05-02:
but we don't want optimizations that have negative side-effects.
so either propose some, that don't have any (neoraider already suggested a possible way) or understand that there won't be changes that introduce new problems!

closed 2017-05-11 by rotanid