freifunk-gluon/gluon #1118: Activate PoE on CPE510 via uci fails

mokkin, 2017-05-03:
Following the command written at https://github.com/freifunk-gluon/gluon/wiki/Commandline-administration#activate-gluon-20162x fails for LEDE based gluon:
```
root@liliensteinstr-cpe510-n:~# uci set system.gpio_switch_poe_passthrough.value
=1
uci: Invalid argument
```

mokkin, 2017-05-03:
The syntax of the wiki entry was wrong. It has to be:
```
uci set system.poe_passthrough.value=1
```
Ticket solved.

closed 2017-05-03 by mokkin

FreifunkUFO, 2017-05-04:
unter lede (ohne gluon) auf einer M5 sehe ich aber auch weiterhin den hier zuerst gesendeten Befehl :-O

SimJoSt, 2017-05-04:
Huh. Seemed to have been my mistake.
Weird, I only added it right after testing and using it.

NeoRaider, 2017-05-04:
Ugh, the GPIO configs were renamed between OpenWrt CC and LEDE...

NeoRaider, 2017-05-04:
Hmm, it's possible that the LEDE setup scripts don't update the configuration, so an OpenWrt CC system upgraded to LEDE will still use the old names. If this is the case, it should be fixed in LEDE.