freifunk-gluon/gluon #1120: build fails when using very long paths

t3r, 2017-05-05:
This is probably not a gluon issue but should be adressed to openwrt or probably quilt. I'll start here, anyway.

I try to build using Jenkins´ multibranch pipeline. This nice tool has the habit of automatically creating _very_ long directory names for the workspace like /home/jenkins/job_branch-LONG_HASH and in my case it's
`/home/jenkins/workspace/funk_experimental_lauenburg-XCXMZWWG24Y2ZJAFNYCNOHRMYYWMVD5GBJJRU75H46UN5SAKC44A`

as a result, within gluon the tool quilt gets built with too long shebang lines like this:
`#! /home/jenkins/workspace/funk_experimental_lauenburg-XCXMZWWG24Y2ZJAFNYCNOHRMYYWMVD5GBJJRU75H46UN5SAKC44A/gluon/build/ar71xx-generic/openwrt/staging_dir/host/bin/bash`

This finally results in an error when quilt gets called:
`/home/jenkins/workspace/funk_experimental_lauenburg-XCXMZWWG24Y2ZJAFNYCNOHRMYYWMVD5GBJJRU75H46UN5SAKC44A/gluon/openwrt/build_dir/host/quilt-0.63/bin/quilt: /home/jenkins/workspace/funk_experimental_lauenburg-XCXMZWWG24Y2ZJAFNYCNOHRM: bad interpreter: No such file or directory`

as the interpreter gets truncated due to a linux kernel limited maximum length of the shebang line.

Building gluon on the same machine under some shorter path works without a problem.

t3r, 2017-05-09:
After some testing I came to the conclusion that openwrt simply does not build using Jenkins multibranch Pipelines or in environments with very long path names.

For the record: The quilt issue could be fixed by replacing
`QUILT_CMD:=quilt --quiltrc=-`
in openwrt/include/quilt.mk with
`QUILT_CMD:=$(BASH) quilt --quiltrc=-`
however the build fails later with error messages indicating a too big environment.

I suggest to close this issue.

NeoRaider, 2017-05-09:
This issue is fixed in the LEDE-based master, I'll backport the fix to v2016.2.x.

NeoRaider, 2017-05-09:
Backported in 14c2aff7c0e7c578bca770c2b6c900a1dec90181

closed 2017-05-09 by NeoRaider