freifunk-gluon/gluon #1122: Archer C60 support

Adorfer, 2017-05-11:
Der Archer C60 hat - wenn ich es richtig sehe - LEDE-Support. 
https://lede-project.org/toh/hwdata/tp-link/tp-link_archer_c60_v1
Angesichts des günstigen  Preises wäre es ein sinnvoller Nachfolger für den WDR3600.

rotanid, 2017-05-11:
first glance: this should be supportable, yes. also the Archer C59, which looks even more interesting.
but it's not a replacement for the WDR3600 as it doesn't have gigabit interfaces.
i also don't see why i should pay 15€ more than for the Archer C25, which will be supported soon and has almost the same specifications

belzebub40k, 2017-05-11:
@Adorfer On the page you linked to they say 5GHz WiFi is not supported.

rotanid, 2017-05-15:
regarding C59 & C60:
it seems upstream support is partly there, working factory flash via vendors webinterface will be there soon
regarding similar C58:
similar hardware, pull request is there
regarding 5GHz:
not yet supported for these three, there seem to be problems with QCA9886.

so i think we should only add support after the factory flash is working and then also only with BROKEN until 5GHz support is there (if it can be done at all)

RubenKelevra, 2017-07-08:
Archer C60 stability with Gluon (IBSS & 2.4 GHz) does have now and then outages in the interconnection between nodes. Dmesg shows the following messages (maybe related maybe unrelated):

```
ath: phy1: timeout (100000 us) on reg 0xa240: 0x00000000 & 0x00000001 != 0x00000001
ath: phy1: Could not kill baseband RX
ath: phy1: timeout (100000 us) on reg 0xa240: 0x00000000 & 0x00000001 != 0x00000001
ath: phy1: Could not kill baseband RX
```

LEDs are working fine, status (the I/O-LED) as well as 2.4 GHz LED.

5 GHz WIFI is not detected at all, I guess the hardware is not turned on/switched to the right mode.

A bit confusing is, 2.4 GHz is shown as phy1 but the configuration for this interface shows it as radio0 - which I though is directly related to eachother - seems to be not...

```
ls -l /sys/devices/platform/*/net/client0/phy80211
lrwxrwxrwx    1 root     root             0 Jul  8 02:53 /sys/devices/platform/qca956x_wmac/net/client0/phy80211 -> ../../ieee80211/phy1

uci get wireless.client_radio0.device
radio0

```

rotanid, 2017-07-08:
as always (see Archer C25), someone has to backport the hardware support to lede-17.01 before gluon will pick it up.

RubenKelevra, 2017-07-08:
Yeah I just wanted to give some feedback on this hardware. :)

mokkin, 2017-08-28:
I also tested the C59 with LEDE master on 2.4GHz which works fine. It should be supported even when 5 GHz is still not working because it is already a good alternative to WR841 series. (RAM/Flash size)

hargut, 2017-09-10:
FYI: I just stumbled across:
https://forum.lede-project.org/t/archer-c59-v1-missing-ieee802-11-a-n-ac-5ghz/2226/18 and
https://bugs.lede-project.org/index.php?do=details&task_id=864
but did not test or verify anything of these fixes.

herostrat, 2017-09-24:
Hey guys,
I foung this issue while searching for openwrt/lede support for the C59 and found the following which might help you guys (and other people searching...)

Initial support for C59 (and C60):
https://github.com/lede-project/source/commit/e39dc8d823c86559eedbbdcee5f5c14b827fed0f
This commit is in master and therefore should be working (with the exception of LAN1 and 5Ghz as heinzek states)

Commit https://github.com/lede-project/source/commit/d6baeb5c480f547f13f92b8994cf467d25e2ef7e fixes the LAN1 Port problem and got also merged to master (https://github.com/lede-project/source/pull/846)

This pull request https://github.com/lede-project/source/pull/1198 should fix the 5Ghz problem but is as far as I can tell not merged to master.

I am not really familiar with the processes in lede and tbh new with big git projects and don't really know if I got everything right but as far as I understand it the next version of LEDE should most likely have this fixes implemented.
Please correct me if I am wrong and if you can find out or know when/if this fixes get pushed to master and the precompiled packages.

rotanid, 2017-09-25:
you're completely right, but that's nothing new and has almost nothing to do with Gluon.
that a device is supported in LEDE doesn't make it work automatically in Gluon.
also, the Gluon master branch doesn't yet use the LEDE master branch, therefore a backport would be necessary.

mokkin, 2017-11-10:
The Archer C59 is supported by LEDE trunk now including the 5GHz bugfix.
https://bugs.lede-project.org/index.php?do=details&task_id=864
I successfully testet it yet on both 2.4 and 5 GHz and it works fine. It would be great to backport this device at least to gluon master to have a new dualband beginners device.

rotanid, 2017-11-15:
feel free to step up and do the backport, the Gluon project needs more volunteers!

FreifunkUFO, 2017-11-23:
someone tried it for gluon:
https://github.com/blocktrron/gluon/commit/d10cd89c8b8b3f4fcc32305e26ef90e1c406b6db

i dont know why she/he is soldering:
https://twitter.com/rootsateifel/status/933486697098211329

see non-gluon discussions also at 
https://forum.freifunk.net/t/archer-c59-v3/15986/5
photos at https://twitter.com/rootsateifel/status/933486697098211329

rotanid, 2017-11-23:
yep, @blocktrron is working on it, be patient or contact im directly if he needs help.

blocktrron, 2017-11-24:
I'm currently working with @ArwedSchmidt on it. The C59 looks very promising. It works flawlessly so far with 5GHz, although he has no 5GHz Mesh neighbours to test with.

For my part, i have a C58 running, with the following things not working in no particular order:

- LAN Port 1 detects Link state but no data is transferred. This also happens to be the case with Lede Snapshot, Stock firmware not yet tested. So it could still be the case that my device is faulty.
- Frequent crashes with 5 GHz enabled (Client and/or Mesh). This one seems to be related  to the fact that the C58 has half the RAM (and Flash) of the C59. The C59 goes up to 46% RAM Usage with 5GHz enabled, so maybe the C58 is possibly going OOM. I can't verify this, i broke my derial connetion while attempting to solder it :confused: 

I guess the last one could be related to the C25 and i guess the C60 could have the same problems. With 5 GHz disabled (Mesh+Client) it runs like a charm.

Possibly we have the 841s of 5GHz routers now :smiley:

mokkin, 2017-11-24:
> Possibly we have the 841s of 5GHz routers now :smiley:
I hope soo too, but unfortunately it looks like they become rare. My bunch of five C59 was ordered two weeks ago and they send me periodically mails that they are in queue... Also on those typical price comparison pages there much less vendors now than a couple of month before and the price is increasing.

mokkin, 2017-11-24:
If you @blocktrron or @ArwedSchmidt need another one of C59 for testing purpos, especially mesh, I could borrow you one after I receive my new ones.

rotanid, 2017-11-24:
if you look at the lifetime (until EOL) of similar devices, "we" are simply too slow to keep up.
C25 is also almost not available anymore. :(

ArwedSchmidt, 2017-11-24:
Just came back from work and will now test @blocktrron 's backport for C59 in depth. 

- For now I can confirm that the C59 seems to not mesh on LAN Port 1 though I remember having had no problems with ssh/pinging from that port with a LEDE master build. 
- C59 seems to be stable in terms of memory usage.
- no luck with serial either: really so easy to destroy the smd resistors?

Warning: better don't try to unplug u.fl antenna connectors, destroyed one when trying to pull the cable gently off. C59 is not marked as EOL, btw.

ArwedSchmidt, 2017-11-25:
For me, everything works as of https://github.com/blocktrron/gluon/commit/cf5db62652f9f1ee3357a0a542b8f969009cc816 @blocktrron still got issues with LAN1.

@mokkin Maybe you can build, flash and test your C59?

blocktrron, 2017-11-26:
So i have been testing regarding the LAN Port 1 issue: LAN Port 1 is not transmitting nor receiving any data if no other LAN Port (2-4) is up, although the connection state is correctly negotiated ans also logged. 

As soon as another LAN Port comes up, the port starts transmitting and receiving data.

At least what i can tell, everything else is working correctly for the C59, despite IBSS mesh support is currently not backported.

NeoRaider, 2017-11-26:
@blocktrron Does LAN1 work correctly in the LEDE master?

blocktrron, 2017-11-26:
@NeoRaider LAN Port 1 is affected with the same behaviour in LEDE snapshot.

NeoRaider, 2017-11-26:
Sounds like QCA956X_ETH_CFG_SW_PHY_SWAP and/or QCA956X_ETH_CFG_SW_PHY_ADDR_SWAP are wrongly set in `target/linux/ar71xx/files/arch/mips/ath79/mach-archer-c59-v1.c`. Removing one or both of these flags can change the internal port order, so if this fixes the issue, `target/linux/ar71xx/base-files/etc/board.d/02_network` might also need to be updated.

NeoRaider, 2017-11-26:
Hmm, looking at your description of the issue again, QCA956X_ETH_CFG_SW_PHY_SWAP and QCA956X_ETH_CFG_SW_PHY_ADDR_SWAP might also be correct, and phy_poll_mask might be the culprit. Try setting that to BIT(0) instead of BIT(4).

blocktrron, 2017-11-28:
@NeoRaider Thanks for pointing out phy_poll_mask, LAN Port 1 is working correctly now.

mokkin, 2017-12-02:
@blocktrron I'm building your firmware right now and I'm gonna test it on monday.
@ArwedSchmidt In the LEDE Wiki they say it is discontinued in 2017. My shipment from Germany (Reichelt) has been cancelled, but my second one from Spain is on the way :-)

ArwedSchmidt, 2017-12-02:
@mokkin I wasn't sure. Lately we tried ordering a few C59 and got told that it got discontinued but the C60 is still available. Which model did you order? We should soon have a C60 for testing purposes and can then tell whether @blocktrron's branch works there fine too. Is anybody willing to help by creating the pull request based on blocktrrons work? @NeoRaider gave a good tutorial on what's required to do so in https://github.com/freifunk-gluon/gluon/pull/1162 

My time is kinda limited the next week so others would prolly be faster and Blocktrron already put a lot of effort into this. Any help welcome :-)

rotanid, 2017-12-02:
@ArwedSchmidt i'm sure @blocktrron will do a pull request when he thinks the work finished, you don't have to interfere here ;-)

blocktrron, 2017-12-02:
Personally, i would like someone to test a C60 before creating a PR. I won't buy one, because i suspect it to be crashing just like the C58 due to insufficient RAM.

Another point would be IBSS support. There are still communities out there using it and (although my community doesn't use it anymore) it would be great to have it working.

Thanks to @ArwedSchmidt for contriuting to this backport by the way!

FreifunkUFO, 2017-12-03:
@blocktrron the 2.4 ghz wifi is supporting ad-hoc (and 11s) just like all other tplink/ath9k.
according to the 5ghz ac-hardware you cant normally use ad-hoc there, but 11s is running, and of course the access-point-mode for the vAPs

mokkin, 2017-12-04:
Unfortunately my build fails. Any idea why?
```
Configuring gluon-config-mode-mesh-vpn.
Setting flags for package libc to hold.
Setting flags for package libgcc to hold.
Setting flags for package libpthread to hold.
Setting flags for package librt to hold.
make[4]: Entering directory '/home/gluon/gluon-c59-test/lede'
make[4]: Nothing to be done for 'package/preconfig'.
make[4]: Leaving directory '/home/gluon/gluon-c59-test/lede'
cp -fpR /home/gluon/gluon-c59-test/lede/build_dir/target-mips_24kc_musl-1.1.16/root-ar71xx /home/gluon/gluon-c59-test/lede/build_dir/target-mips_24kc_musl-1.1.16/root.orig-ar71xx
/home/gluon/gluon-c59-test/lede/staging_dir/hostpkg/bin/lua: stdin:37: site.conf error: expected `site_seed' to be a string
stack traceback:
	[C]: in function 'assert'
	stdin:37: in function 'assert_type'
	stdin:54: in function 'need_string'
	stdin:59: in function 'need_string_match'
	stdin:159: in main chunk
	[C]: in function 'dofile'
	(command line):1: in main chunk
	[C]: ?
postinst script ./usr/lib/opkg/info/gluon-core.postinst has failed with exit code 1
package/Makefile:60: recipe for target 'package/install' failed
make[3]: *** [package/install] Error 1
make[3]: Leaving directory '/home/gluon/gluon-c59-test/lede'
package/Makefile:102: recipe for target '/home/gluon/gluon-c59-test/lede/staging_dir/target-mips_24kc_musl-1.1.16/stamp/.package_install' failed
make[2]: *** [/home/gluon/gluon-c59-test/lede/staging_dir/target-mips_24kc_musl-1.1.16/stamp/.package_install] Error 2
make[2]: Leaving directory '/home/gluon/gluon-c59-test/lede'
Build failed - please re-run with -j1 to see the real error message
/home/gluon/gluon-c59-test/lede/include/toplevel.mk:198: recipe for target 'world' failed
make[1]: *** [world] Error 1
make[1]: Leaving directory '/home/gluon/gluon-c59-test/lede'
Makefile:126: recipe for target 'all' failed
make: *** [all] Error 2
```

blocktrron, 2017-12-04:
@mokkin With which site are you builing with?

mokkin, 2017-12-04:
This one: https://github.com/freifunk-leipzig/freifunk-gluon-leipzig/tree/11s

blocktrron, 2017-12-04:
@mokkin The site seed is probably missing. Can you add it to your site.conf and try again?

mokkin, 2017-12-07:
Building and flashing has worked, but now I'm in boot loop...

ArwedSchmidt, 2017-12-07:
I got a C60 v2 (!) and it turns out it seems to be quite risky to open the plastic case. Haven't opened it yet, so I cannot tell whether HW is identical.

rotanid, 2017-12-07:
@blocktrron the C60 is marked as BROKEN anyway, so i would say feel free to open a PR for these devices

blocktrron, 2017-12-07:
@mokkin What device are you using?
@rotanid Will do. If anyone wants to test IBSS support, hopefully all required patches are in my archer-c5x-ibss branch. Personally, i have no time to verify IBSS working.

mokkin, 2017-12-08:
I'm using a C59v1.
Failsafe isn't working too.

blocktrron, 2017-12-08:
@mokkin Could you maybe provide a serial log? I've build my branch from scratch and flashed it onto a C59 without issues. I assume you tried tftp recovery - Have you tried to use a switch between your computer and the Archer?

mokkin, 2018-01-07:
@blocktrron I tried hard to get some signals. I even soldered some wires on the very tricky TP4 and TP5 contacts on the backside, but I couldn't receive anything.
Now it's back on stock and I'm going to build and flash again with your updated trunk.

ArwedSchmidt, 2018-01-08:
Had same issues here. No signal on TP4 / TP5. We weren't sure whether I
destroyed some of the SMD resistors close to TP4/TP5 while soldering.

mokkin, 2018-01-08:
Finally it's working. My C59 is runnig fine on both bands now!
The only issue I have is the status page. For unknown reasons I cannot access it through the browser.

rotanid, 2018-01-09:
@mokkin did you test with #1281 ?
if so, please report over there, not in this issue, thanks!

PR #1281 (2018-01-16): ar71xx: add support for Archer C58/C59/C60
Reopened due to accidental branch-deletion, see #1275 

This adds support for Archer C58/C59/C60.

Archer C58/C60 marked as broken, due to frequent crashing on activated 5GHz related to insufficient RAM. Working fine with disabled 5GHz.

Archer C59 not affected (128MB instead of 64MB RAM)

closes #1122

Currently not working:
- IBSS meshing on 5GHz (possibly fixed in my ibss branch, needs testing)
closed 2018-01-16 by rotanid