freifunk-gluon/gluon #1123: support for TP-Link Archer C25

rotanid, 2017-05-11:
the new Archer C25 from TP-Link has dual-radio, 775MHz Qualcomm Atheros CPU, 64MB RAM, 8MB Flash, 5* 10/100 Mbps Ethernet.

@Brother-Lal did already work on LEDE support:
https://github.com/lede-project/source/pull/1069

jannic, 2017-05-11:
I made a build of gluon based on the LEDE patches:
https://github.com/jannic/gluon/tree/c25
This gives a bootable firmware, but is not ready for merging because of a few issues:

- There is some major memory leak, probably related to ath10k. The router reboots after ~1 minute with an oom error. If I disable the 5GHz wifi, the leak seems to go away.
- LAN/WAN ports are swapped
- LED support seems to be broken (all LEDs are always on)
- not much testing done, yet - just 'it boots and gets mesh connectivity'

This is a purely mechanical backport of the LEDE C25 patches  to LEDE 17.1, adding a bare minimum of gluon integration. Take it as a proof of concept and a starting point for further development, it's not yet finished.

NeoRaider, 2017-05-11:
The swapped LAN/WAN ports are weird, does this work on plain LEDE?

jannic, 2017-05-11:
Didn't try plain LEDE, so I don't know. The C25 support of LEDE is still in development, so it's quite possible that there is still a bug.
On the other hand, the LEDE patch lists eth0 as the WAN port, which seems to be correct (I'm currently connected to the WAN port and eth0 is UP.)
But gluon assigns eth0 to the br-client bridge.

Brother-Lal, 2017-05-11:
Please try my patches with a bare lede and see if you can reproduce the issues!
I had no issues in that regard, leds, wan/lan wifi should all be working, so it would be nice to have feedback if it's broken for you.

jannic, 2017-05-11:
@Brother-Lal sure, I'll do if I find the time.
There are actually two versions of LEDE to test: Your tree, based on LEDE master, and the 'backport' (just fixing merge conflicts, nothing fancy) to 17.01. Perhaps I missed some patch in LEDE master which should be backported, as well.

jannic, 2017-05-11:
Ok, LEDE (both 17.1 and master) properly assigns eth1 to br-lan, so it looks like it's a gluon-only issue.
The LEDs don't work with 17.1, but they do with master. Perhaps I broke something while backporting, or they just need some feature not yet available in 17.1.
Can't see the 5GHz related memory leak. Maybe gluon-specific, maybe because there is no load with pure LEDE, while gluon joins the mesh and sees quite a lot of traffic. (But only on 2.4GHz, as this device doesn't do 5GHz ibss)
In summary, it looks like of the issues I listed above, none is relevant to LEDE master.

jannic, 2017-05-11:
One thing is missing in both LEDE and gluon: sysupgrade support doesn't work, yet. I had to flash firmware using the tftp recovery method of the bootloader.

Brother-Lal, 2017-05-11:
Thanks! I tested factory, but not sysupgrade...
Will report back in the lede pr

jannic, 2017-05-11:
The swapped LAN/WAN interfaces are fixed with this patch:
https://github.com/jannic/gluon/commit/b8de3f50ff1827dd5a78a55e5b25bf8d4039cbc3

And the memory leak seems to be gone when using the 11s firmware instead of the ibss one.

jannic, 2017-05-11:
And the LEDs work with this patch:
https://github.com/jannic/gluon/commit/3d65fa9f16e13d684af2301d5f697e13ae17e8e7
(Taken from https://github.com/lede-project/source/commit/47f3690d3b8b2830d8646b87aa09be61615129ae#diff-d85399e9e12e4e705aee1f3236dd9e9c)

jannic, 2017-05-11:
Got sysupgrade working, as well:
https://github.com/jannic/gluon/commit/f60b3c7a5e1520f52745075b70225caf2e1a4382

Brother-Lal, 2017-06-01:
I tried my own dirty backport, and i had issues too with Memory leaks/OOM reboots.

Did the -ct firmware work for you, or the default one?

i have ath10k-ct installed, but the normal firmware. (which looks fishy on its own...)

jannic, 2017-06-01:
Those OOM reboots happen with `GLUON_ATH10K_MESH ?= ibss` but not with `GLUON_ATH10K_MESH ?= 11s` (at least for me).
I hope this can be solved by using the correct firmware, but didn't have time to investigate further.

RubenKelevra, 2017-06-04:
The LEDE-support seem to have improved lately. Maybe try the latest version again?

https://github.com/lede-project/source/pull/1069#issuecomment-305975254

rotanid, 2017-06-04:
@RubenKelevra if you'd have read the whole link you provided you would have noticed, that @jannic is also involved over there - so please don't post unless you have something relevant to contribute.

rubo77, 2017-06-09:
@jannic: How do you build the firmware with your changes?

I created a branch and rebased it to current master at https://github.com/rubo77/gluon/tree/c25

But when I build this I get the error

```
Applying: ar71xx: add support for TP-Link Archer C25 v1
error: patch failed: target/linux/ar71xx/base-files/etc/board.d/02_network:122
error: target/linux/ar71xx/base-files/etc/board.d/02_network: patch does not apply
Patch failed at 0001 ar71xx: add support for TP-Link Archer C25 v1
```

Don't I have to add a certain lede-branch as well? If so, How?

rotanid, 2017-06-09:
@jannic are you planning to create a PR for Archer C25 on lede-17.01 ?

jannic, 2017-06-09:
@rotanid My goal is having support for Archer C25 on the Freifunk Aachen firmware. As Aachen is still using Gluon 2016.2.x, this is a long-term goal.
I will continue working on getting support into Gluon, and this of course means backporting the patches to LEDE 17.01, but I don't have specific plans to do a PR.

rubo77, 2017-06-09:
@jannic I know how to build the firmware usually. I successful built hundredth of versions already, but I have problems with this branch.

can you give a list of exact commands to clone gluon, add the remote, change branch and what else to do before starting with the standard make commands as usual?

rubo77, 2017-06-19:
I tried a bit with this firmware, but it didn't really seem stable, maybe I can try again tonight....

What are your experiences?

jannic, 2017-06-19:
It seemed to be rather stable, with some caveats:
- ibss mesh on 5GHz caused OOM within a few seconds
- chosing `GLUON_ATH10K_MESH ?= 11s` in site.mk fixed that issue, but as our comunity doesn't use 11s yet, this may just be because that way, the 5GHz mesh was not used at all.
- didn't do real long-term or load tests, but didn't notice any other issues within a few days at low load.

rubo77, 2017-06-19:
Can you disable the 5GHz functionality then if 11s is not supported in the site.mk?

Then the C25 would be a good low-cost alternative to the 841N.

rotanid, 2017-06-19:
maybe @Brother-Lal can do some longer 5ghz 11s mesh testing to check if the issue is ibss-only

rubo77, 2017-06-20:
Kann hier mal bitte jemand, der sich auskennt weitere Infos ergänzen?

https://wiki.openwrt.org/inbox/tp-link/tp-link_tp-link_archer_c25_v1

Das wäre super ;)

Brother-Lal, 2017-06-20:
I think the lede wiki would be a better start.. https://lede-project.org/toh/start
And the gluon link is completely wrong in the wiki too, because the support for c25 for gluon is here, but the lede support was in this pr: https://github.com/lede-project/source/pull/1069

Anyway on topic:, i have disabled 5ghz completely , because it makes wifi unusable...
So either my backport/flashing has damaged the 5ghz wifi, or it is just not stable on my device.
Did anybody try to mesh with other 5ghz gluon nodes and then test the connection?
As soon is it had 5Ghz on, Clients would get no data (although they could connect!).

With 5g disabled everything runs smooth

rubo77, 2017-06-27:
Maybe some more solutions in https://github.com/freifunk-gluon/gluon/pull/1162

jannic, 2017-06-27:
@rubo77 is there anything in particular in that PR that looks useful for C25 support? I didn't find anything obvious.
btw, @Brother-Lal, I can confirm that 5GHz support is unstable even with ibss disabled. So no, you probably didn't damage your device.

rubo77, 2017-07-06:
@jannic in which way is 5GHz unstable?

I created a firmware image with the branch of @RubenKelevra which seems to run stable:  
https://github.com/rubo77/gluon/tree/rubo-c25

How can you force the unstableness, you mention?

rotanid, 2017-08-07:
ich habe mich auch mal daran versucht und unter anderem noch die aktuelleste Firmware für den QCA9887 backported auf lede-17.01.
Hier findet ihr meine bisherige Arbeit daran:
https://github.com/rotanid/source/tree/lede-17.01-archer-c25  (6 commits)
https://github.com/rotanid/gluon/commits/v2017.1.x-tecff-c25 (4 commits)
es scheint aber immer noch kein 5GHz zu funktionieren - immerhin ist das aber nun mal schön gemacht und könnte zumindest als BROKEN in gluon eingehen für 2,4GHz-only

rotanid, 2017-08-07:
ich habe mich auch mal daran versucht und unter anderem noch die aktuelleste Firmware für den QCA9887 backported auf lede-17.01.
Hier findet ihr meine bisherige Arbeit daran:
https://github.com/rotanid/source/tree/lede-17.01-archer-c25  (6 commits)
https://github.com/rotanid/gluon/commits/v2017.1.x-tecff-c25 (4 commits)
es scheint aber immer noch kein 5GHz zu funktionieren - immerhin ist das aber nun mal schön gemacht und könnte zumindest als BROKEN in gluon eingehen für 2,4GHz-only

Es wäre gut, wenn mehr Leute außer mir das reine LEDE-17.01 build testen um sicher zu gehen, dass 5 GHz dort noch funktioniert - ich glaube das tut es, würde aber gerne weitere Bestätigungen haben!

RubenKelevra, 2017-08-07:
@rotanid thanks for your work. Please keep in mind that the main language of this project is English to remove the language barrier for non-german speakers.

My hacky patches enables the 5 GHz capabilities and it seems to work - commecting over 802.11s, batman-adv interconnection as well as IPv6-Ping over this Batman-adv-network works.

The main issue we‘ve discovered is a serve memory leak which eat up nearly one MB per 2 seconds.

Didn’t had time so far to track it down, so maybe this only occours with the CT-Firmware which Lede doesn’t use per default.

rotanid, 2017-08-08:
@RubenKelevra maybe you can point me to patches i should add to my fork that solved problems?
i have no memory leak at least.

jannic, 2017-08-14:
Hi @rotanid,
I am currently running https://github.com/rotanid/source/tree/lede-17.01-archer-c25 and didn't see any issues yet. The router is running for 17h now without signs of memory leaks or other instabilitites.
I'm simply using the device as a freifunk client on 5GHz and provide a private WLAN on 2.4GHz for my mobile phone as a client, so both bands are used, but no heavy load.)
Good work!
I'll let it run for a few more hours, and then try gluon.

RubenKelevra, 2017-08-14:
I tried Gluon with my patchset yesterday again on this device and was unable to reproduce the memoryleak. I think there’s only a small chance for the hardware to get in this state but a simple reboot does not cure it, you need to powercycle it to stabili the hardware again.

Talking about dual-channel 802.11s mesh+ap with CT-firmware. I’m going to setup a long term stability test this week. :)

rotanid, 2017-08-14:
my 17.01 backport branch with kmod-ath10k insted of kmod-ath10-ct is running for 2 days now without problems.
i also ran a Download for 20 hours with wget, which did a total of 120GB traffic.
no problems. maybe someone else can test my branch too, if someone else confirms i will do pull requests at LEDE and gluon.

rotanid, 2017-08-14:
my 17.01 backport branch with kmod-ath10k insted of kmod-ath10-ct is running for 2 days now without problems.
i also ran a Download for 20 hours with wget, which did a total of 120GB traffic over 5GHz 11s mesh.
no problems. maybe someone else can test my branch too, if someone else confirms i will do pull requests at LEDE and gluon.

RubenKelevra, 2017-08-15:
I would vote for a pullrequest but with broken-flag because of primitive testing, so it maybe still got into gluon 2017.2 and the broken-flag could be removed on a minor version-bump.

Your parchset isn’t that much different to mine, just clean, I would think our results can be compared.

I hope I can reproduce my observed bug in the next days and can report which circumstances actually cause it.

rotanid, 2017-08-16:
@mweinelt reported a memory leak with a build based on the patches and gluon master branch.
we have to investigate this first... couldn't reproduce it yet.

rotanid, 2017-08-16:
@mweinelt reported a memory leak with a build based on the patches and gluon master branch.
we have to investigate this first... couldn't reproduce it yet.

also, i have updated my branches master-c25 and v2017.1.x-c25 to also work for ibss builds - though i haven't tested ibss and i won't do it as we have long moved on to 802.11s

rotanid, 2017-08-30:
although the router ran almost fine at home with one 5GHz 11s mesh partner, the router started rebooting all the time in an environment with multiple clients but NO 5GHz mesh partner.
someone has to dig deeper here...

RubenKelevra, 2017-08-31:
@rotanid did you attached a serial console to look at the Kernel-output?

rotanid, 2017-08-31:
no, i won't break another one open, money doesn't lay on the streets and my time is limited, too.
but @mweinelt did look at it i think and @Brother-Lal has one that is already open.

TobleMiner, 2017-08-31:
@rotanid are you talking about multiple 5 GHz clients or does the wireless band not matter?

Also I have seen ```opkg list``` giving me strange out of memory errors on the C25, too.

rotanid, 2017-08-31:
@TobleMiner multiple 5 GHz clients. haven't tried though disabling 5GHz AP only and leaving 5 GHz 11s mesh running 
@kpanic23 reported on IRC, that he can do `opkg list` just fine now.  ( actually, most of the discussion takes place on IRC in #gluon - way more convenient than commenting here all the time )

J0WI, 2017-10-14:
I can confirm that 2.4 GHz and 5 GHz works fine well (over 2.4 GHz mesh, 11s). I forwarded some gigabytes without any issues.

rotanid, 2017-10-22:
the pull request has been merged, the device is supported with the BROKEN marker in the master branch.
BROKEN because enabling 5GHz results in frequent reboots (possibly OOM related) in some airtime-noisy environments (e.g. in many urban areas)

closed 2017-10-22 by rotanid

speedyb0y, 2017-10-22:
Hi! Sorry for my english...

Does anyone know about the pinouts for serial access on the Archer C25?
I can't find any image on Google, I tried OpenWRT and LEDE wiki's, but it seems this router is a brand new one... :(

Brother-Lal, 2017-10-22:
The first one close to the network sockets is 3.3v, the next gnd, then rx, then tx.

hackedbellini, 2017-11-15:
Hey guys. I just read the discussion here and, since the [knowledge base](https://lede-project.org/toh/hwdata/tp-link/tp-link_archer_c25_v1) doesn't have anything about that I thought to ask here.

1) Is the firmware usable right now? Even with some issues (e.g. "don't enable 5hz wifi")

2) If I found a major issue that doesn't let me use the wifi, is there a way to go back to the stock rom?

edeso, 2017-11-15:
On 11/15/2017 18:54, Thiago Bellini wrote:
> Hey guys. I just read the discussion here and, since the [knowledge base](https://lede-project.org/toh/hwdata/tp-link/tp-link_archer_c25_v1) doesn't have anything about that I thought to ask here.
> 
> 1) Is the firmware usable right now? Even with some issues (e.g. "don't enable 5hz wifi")

exactly.. runs stable here. enabling 5GHz leads to crashes/reboots. a timely login via ssh (needs to be enabled), 'wifi down', disabling 5GHz wifi, 'reboot', stabilizes the box again. 
 
> 2) If I found a major issue that doesn't let me use the wifi, is there a way to go back to the stock rom?
> 

yes.. tftp recovery. similarily as described for the wr841
  https://wiki.openwrt.org/toh/tp-link/tl-wr841nd#tftp_recovery_via_bootloader_for_v8_v9_v10_v11_v12
get a stock firmware and just rename the file to 'ArcherC25v1_recovery.bin'. works the w/ a factory lede/gluon image as well.

..ede

hackedbellini, 2017-11-16:
@edeso thanks for the explanation!

Is the 5GHz issue being worked on already? If not, is there an issue that I can use to follow-up?

Also, what do you think of everything comparing to the stock rom? I was used to custom firmwares on my last router (used ddwrt on it) and this was my first time in a while using the original firmware. One issue that I'm having is that the router will reboot itself when the connection drops. Sometimes it drops for 10 seconds, but the reboot takes more than a minute and that is very bad for gaming =P

edeso, 2017-11-16:
On 16.11.2017 16:51, Thiago Bellini wrote:
> @edeso <https://github.com/edeso> thanks for the explanation!

nP

> 
> Is the 5GHz issue being worked on already? If not, is there an issue that I can use to follow-up?

can't tell. even the lede forum linked to the freifunk ticket. they seemed to be the driving force behind the port.

didn't really look for a lede ticket wrt. to the 5GHz issue though.

> Also, what do you think of everything comparing to the stock rom? I was used to custom firmwares on my last router (used ddwrt on it) and this was my first time in a while using the original [...]

sorry. immediately flashed the box w/ gluon when it came. so i can't tell you anything about the stock firmware ;)

..ede

rotanid, 2017-11-16:
@hackedbellini as the problem with 5GHz most likely is in the binary firmware, the only one who can fix this is the vendor - and they most certainly won't do that.
interestingly enough, at my place even 5GHz works with this device, but everyone else reports problems.
further questions would be better on the LEDE mailing list or Forum, as your question don't have to do anything with Gluon ;-)

hackedbellini, 2017-11-17:
> further questions would be better on the LEDE mailing list or Forum, as your question don't have to do anything with Gluon ;-)

I'll do that! Thanks

Adorfer, 2019-06-27:
any method of reproducing the alleged 5GHz-issue? (i have one on my desk and it's meshing 5GHz to an R6120 and serving AP clients )