freifunk-gluon/gluon #1127: Feature Request: Add Option to hide client stats

J0WI, 2017-05-17:


NeoRaider, 2017-05-17:
Hiding these statistics won't do any good, the number is still announced through the mesh by the routing protocol (including each client's MAC address; unfortunately, this is a technical necessity, as it enables roaming).

rubo77, 2017-05-18:
Maybe create apr in eher Mesh Viewer or hop Glass to hide certain data from public

rubo77, 2017-05-18:
Maybe showing false data would be an even better solution, cause just hiding the data can make observers curious so they start creating their own analysis

J0WI, 2017-05-18:
For capturing these stats from the mesh you need to be in the range of the AP. But over the map/status site you can access these stats remotely.

NeoRaider, 2017-05-18:
@J0WI Well, in range of any node. Or in the mesh VPN... Just type `batctl tg` on any node, and you'll see a list of all client MAC addresses in the whole mesh, neatly listed together with the MAC address of the node they're connected to.

If anything, I'd propose making it possible to run a node anonymously, as the client information can't really be hidden without sacrificing roaming (and certainly not as long as we're using batman-adv). Obviously, this is in conflict with the Pico-Peering Agreement; this policy decision is for each community to make for itself (personally, I don't have clear opinion on this; I can see good arguments for both sides).

rubo77, 2017-05-18:
I think it is not needed to make a node completely anonymously.

just the amount of data transmitted through that node and the number of clients should not be shown on the public status page (and on the meshviewer)

This is data, that surely "is there", if you know where to search, but it should not be shown to everyone publicly if the node ower doesn't want it. (I guess the reason is, that the node owner just doesn't want **everyone** to see when he/she uses the internet at home and when not)

rotanid, 2017-05-19:
to me this looks like another case where someone should rather buy his own VPN account instead of mis-using Freifunk, adjusting it for his/her personal benefit

rubo77, 2017-05-19:
That could be one Person's desire 

But i can also think of people who don't want to be observed in their privacy like that that everyone can see at which times they are online and which not.

J0WI, 2017-05-20:
In my opinion Freifunk should also be free from tracking. But in fact you are able to easily track clients/APs. So how can we solve or at least work around this?

Is there a possibility to map MACs to a hash value?
Is there a possibility to hide/randomize the  amount of connected clients just for the map?

rubo77, 2017-05-20:
Sure, the map is another project, and you can add whatever you like in those projects, for example: [hopglass](https://github.com/hopglass/hopglass/) or [meshviewer](https://github.com/freifunk-bielefeld/meshviewer).

I added an issue in hopglass: https://github.com/hopglass/hopglass/issues/94

rubo77, 2017-05-26:
> The actual stats are reported through respondd on the nodes whose only purpose is to provide stats. If you want to remove certain data from stats, you should do so by modifying the data this daemon reports

rotanid, 2017-06-07:
closing according to neoraiders comment.
feel free to create separate issues or better, pull requests for similar things that were discussed here.

closed 2017-06-07 by rotanid

Adorfer, 2017-06-08:
number of endpoints behind a node, including himself (localmac) in the count, formal of current batctl: 
batctl tg |grep $(batctl t $MAC)|cut -d" " -f 3|sort|uniq|wc -l