freifunk-gluon/gluon #1128: TP-Link TL-WA850RE v2

RubenKelevra, 2017-05-20:
This is a feature request for hardware support for the WA850RE v2.

We got some devices of this kind and like to ask what the blocker on supporting these is?

If only testing is required we can do this.

rotanid, 2017-05-20:
i guess the blocker is motivation because of:
* old device, probably discontinued
* crappy hardware-specifications (4mb/32mb)
but if you test & add support, nobody will stop you.
(my source: https://lede-project.org/toh/hwdata/tp-link/tp-link_tl-wa850re_v2 )

RubenKelevra, 2017-05-21:
I don't see how this would be a blocker, the version 1 is also supported as well as all 841N versions which most community-networks (of freifunk) consist of.

We got some devices and would like to help with testing a added support. :)

rotanid, 2017-05-22:
this is no blocker in terms of "won't be merged/supported" but most developers certainly aren't very happy to work on hardware with tiny flash/ram.
try adding the support yourself, it's already supported in LEDE so it will be rather easy.
ask in irc #gluon for help if the readthedocs information and looking at the code doesn't help, often there is someone happy to help.

FreifunkUFO, 2017-07-03:
the 850v2 is already supported by openwrt/lede. but -due to the tplink safeloader -  device has less flash then 850v1 (even when having same 4mb).
unfortunatly many firmware developers hate these devices, so they told you to do it by yourself :-(

CodeFetch, 2018-01-12:
I hate these devices, too, but maybe you can flash the 850v1 with the raw dump of 850v2 if the hardware is the same. Most TP-Link-routers are programmable in-system.

mweinelt, 2018-11-26:
OpenWrt support added on 2017/02/10 in 56561b9f5198d39604809e5cc10dead3e12c1fdf, since 2018/01/12 part of the ar71xx-tiny target.

If anybody is interested to integrate this device please open a pull request with a test log.

closed 2018-11-26 by mweinelt

adrianschmutzler, 2018-11-26:
Just a comment: I tried to include that with Freifunk-Franken firmware early 2018, but stopped it again due to
- even lower available flash space than on other 4 MB devices (mentioned above)
- no TFTP via LAN; and in my opinion not desirable to open 230 V device for serial access
- Amazon already shipped v4 (!) at that point

FreifunkUFO, 2018-12-14:
@mweinelt  this ticket IS about the interest to integrate this device, just like other tickets about hardware.
so, why do you close this ticket?

Byggvir, 2018-12-14:
Well I own a device. But is one device worth the trouble?

mweinelt, 2018-12-14:
If device support exists users owning a device might find some interest in participating. Not all people can be expected to participate in the development process and discussion.

Byggvir, 2018-12-14:
I would be able to create an image, when I find out what I have to include for the device.

CodeFetch, 2018-12-17:
@Byggvir The problem with the device seems to be the flash layout. Currently the DIR-615 C1 is the device with the lowest available flash supported by Gluon. Actually I might be the only owner of one of these devices in Germany as they were only sold in U.S. and China. We only keep this device to have a little tolerance for keeping images small enough. If the flash layout of the WA850RE only allows an even smaller size, it would make no sense to support it. I don't have checked the flash layout of the WA850RE... Is it smaller?

blocktrron, 2018-12-18:
@CodeFetch As the device is using the safeloader image format: Yes, it's pretty small

https://git.openwrt.org/?p=openwrt/openwrt.git;a=blob;f=tools/firmware-utils/src/tplink-safeloader.c;h=358247e6267858119dbba4965592fdf4d7bb2d3d;hb=HEAD#l840