freifunk-gluon/gluon #1129: Feature Request: Disable IPv6 RA for Clients

A-Kasper, 2017-05-22:
Many services are still broken or slow with IPv6. Android and iPhones are using IPv6 as default if available. So the user experience is not very nice.
Because we don't got enough IPv4 addresses we need to NAT IPv4. For maintenance of the nodes it's not possible to deactivate the whole IPv6 connectivity at the Backbone part of the network. 
I would like to have a toggle in config mode presettable in site.conf to disable IPv6 RA on Client Network. 

The Idea is, that if activated RAs are filtered out in lan and client wlan. who need IPv6 can stil enable and use it by setting a gateway manual. the nodes would still be reachable via internet with IPv6. If a nodehost want to offer IPv6 RA he could enable it in config mode or via lucy. 
This could also reduce IPv6 multicast traffic.

kpanic23, 2017-05-22:
Just use ULAs for IPv6. IPv4 is preferred over ULAs. We're using NAT for IPv4 and NPT with ULAs for IPv6. Works great.

A-Kasper, 2017-05-22:
this would be nice if we could split it in client network and node network.
I want to have real ipv6 adresses for the nodes... i don't know if it's
possible to block ras for non ula networks...
the idea is to make ipv6 possible but not as default.

belzebub40k, 2017-05-22:
I guess with "real IPv6 addresses" you mean public IPv6 addresses. This could be achieved with a custom package that set some ebtables rules like [radv-filterd](https://github.com/freifunk-gluon/gluon/tree/radv-filterd/package/gluon-radv-filterd) does to block forwarding of external RAs on br-client.

A-Kasper, 2017-05-22:
yes.. this was my idea.... but implemented in config mode/site.conf/luci

rubo77, 2017-05-22:
You say

>Many services are still broken or slow with IPv6

Can you give some examples?

Maybe we better try to fix these services instead.

A-Kasper, 2017-05-22:
Das ist kein Freifunk Problem. IPv6 wird halt noch sehr halbherzig
betrieben. Meine persönliche Erfahrung ist, dass es häufig langsamer ist.
Hintergrund ist Gerüchteweise, dass dann in größeren Systemen einzelne
Kisten auch IPv6 Anbieten, aber die Kapazitäten im Vergleich zu den IPv4
Kapazitäten halt nicht annähernd so groß sind. Sprich ein Teil der Server
spricht zusätzlich zu IPv4 auch IPv6.
Deaktiviere einfach mal am Client IPv6 und nutze den Freifunk und sag
selbst ob Du das Gefühl hast, dass es schneller fluppt...

Die Düsseldorfer haben glaube ich auch IPv6 komplett abgeschaltet. Batman
ist auch nicht gerade ein IPv6 Freund... Es gibt leider noch Gründe als
Default lieber IPv4 zu haben.

Mir gefällt eine Mischung aus ULA UPv6 für Clients und public IPv6 für
Nodes sehr spannend. Evtl. kann man das auch anders regeln, indem man in
der site.conf eine Routerliste angibt, aber irgendwie klingt das nach
Zahnschmerzen und single point of failure...

kpanic23, 2017-05-22:
What's the problem using just ULAs? With NPT, access to the nodes from public Internet is possible by manually changeing the prefix of the node's IP to the gateway's prefix. Provided the gateway's firewall is allowing it.

A-Kasper, 2017-05-22:
ah.. you are right... i didn't know NPT and thought it was like NAT... this
sounds like a very nice workaround
what do we need in gluon to report the public node ip to hopglass?
I need to read a little bit further how to use your way, but it sounds very
interesting to me...
is your setup (config files) somehow public, to have a look how it's done?
In my mind this "IPv4 as default" setup should be gluon default as long as
IPv6 isn't as reliable as IPv4....

kpanic23, 2017-05-22:
"IPv4 as default" has nothing to do with gluon. It's up to the respective client's OS' implementation. And all OS's I can think of choose public IPv6 over IPv4 and IPv4 over ULA.
Since our nodes are deliberately not reachable from public Internet, our configuration would not help. Also, our network is segmented in quite a few subnets, and most of our gateways only have a /64, we do not have real 1:1 mapping.
But key to NPT is:
`ip6tables -t nat -A POSTROUTING -o eth0 -j NETMAP --to 2001:db8:19a1:27fc::/64`
So for example address fdc7:3c9d:ff31:1:fa1a:67ff:fe5a:63cd would get mapped to 2001:db8:19a1:27fc:fa1a:67ff:fe5a:63cd when leaving eth0, and vice versa.
You could map to a different prefix depending on the source interface, in order to create real 1:1 mappings for multiple internal subnets.
But this has nothing to do with gluon, so might be the wrong place to discuss this...

A-Kasper, 2017-05-22:
that was why I asked for what is needed in gluon.. but it seems like what I
need could be implemented without changes in gluon. this issue can be
closed. thanks for your suggestion...
I understood that it's an OS or application choice what is prefered.. but I
want the best user experience ;)

rotanid, 2017-05-22:
closing this issue as requested.
also, this "feature" doesn't need any change in gluon.

personal opinion:
please don't do it. IPv4 is going to be much more problematic the longer you wait.

closed 2017-05-22 by rotanid