freifunk-gluon/gluon #1130: fastd Knoten werden in Statuspage nicht mehr angezeigt

A-Kasper, 2017-06-01:
siehe z.B. 
[image]

http://[2a03:2260:300b::32b5:c2ff:fe0e:8b6a]/

Gammelknup, 2017-06-01:
Kann es sein, dass es sich um einen L2TP Knoten handelt? siehe rechts ->Mesh-VPN

A-Kasper, 2017-06-01:
Nein. 
Weitere Info:
http://[2a03:2260:300b:0:62e3:27ff:fec6:d990]/
Hat die gleiche Firmware aber direkt per sysupgrade aufgespielt. Da geht alles wie es soll.

Vermutung:
- Umstellung der Syntax in der site.conf (es gibt ja jetzt die unterebene fastd) führt beim update irgendwie zu Irritationen. 
- Bug im neuen Format

MTRNord, 2017-06-01:
@A-Kasper PS: die ipv6 bringt den meisten hier nix da sie in einem anderem netz sind ;)

NeoRaider, 2017-06-01:
@MTRNord Ist ne öffentliche IP-Adresse :)

MTRNord, 2017-06-01:
@NeoRaider ok dann ist mein vodafone anschluss mal wieder gegen mich :)

NeoRaider, 2017-06-01:
The fact that it works on your other node is quite confusing...

Could you also provide your */lib/site.json*? The issue is somewhere in the gluon-mesh-vpn-fastd respondd provider.

A-Kasper, 2017-06-01:
I've used jq . to make it better readable
https://pastebin.com/PbT366w4

commit 9dae2ebb (2017-06-01): gluon-mesh-vpn-fastd: respondd: adjust for new site.conf mesh VPN structure

Fixes #1130
closed 2017-06-01 by NeoRaider