freifunk-gluon/gluon #1131: x86 upgrade scheitert wegen Platzmangel

A-Kasper, 2017-06-01:
siehe: https://pastebin.com/JDeqcM2e

update x86-generic 2016.2.5 auf master.

kpanic23, 2017-06-01:
Duplicate of #1010?

A-Kasper, 2017-06-01:
Nein. #1010 ist gefixed. eigentlich.  Aber verm. Fehlerhaft
Ich weiss nicht wie das Upgrade gemacht wird, aber könnte es sein, dass 1 GB RAM und 1 GB CF sich schneiden?

NeoRaider, 2017-06-01:
The issue has nothing to do with RAM or flash size.

sysupgrade will tar up a small list of config files (see `sysupgrade -l`) and add store this tgz file in the kernel partition. It is possible that your unusual setup with additional packages greatly increased the number or size of configuration files. Check the size of the generated file using `sysupgrade -b /tmp/config.tgz`).

If your configuration is indeed too big, there is no way to fix this at the moment, as the old sysupgrade script of Gluon v2016.2.x can't deal with growing kernel partitions. The sysupgrade script has been improved in the Gluon master, so Gluon v2017.2 will be the first Gluon version with more space for configuration during sysupgrade...

A-Kasper, 2017-06-01:
12 -rw-r--r--    1 root     root       10.9K Jun  1 15:11 config.tgz

A-Kasper, 2017-06-01:
I'll try the following:
1. check if factory image works
2. check upgrade on another maschine...

2017-06-01 15:13 GMT+02:00 Andre Kasper <ein.kasper@gmx.de>:

NeoRaider, 2017-06-01:
Please also check the size of the backup tgz *before* upgrading on the other machine.

A-Kasper, 2017-06-01:
other sizes from other futros: 
12,7k 
12,5k

A-Kasper, 2017-06-01:
we tried it with other machines.. always the same: 
https://pastebin.com/Cdhxjuj7

NeoRaider, 2017-06-01:
Please provide your upgrade image, so I can have a look at it.

A-Kasper, 2017-06-01:
you can find it here:
http://build.freifunk-bochum.de/ffbobeta/sysupgrade/broken

NeoRaider, 2017-06-01:
Reverted fcce9e529501e5952d8c62dae0fe009f36ad2878; closing, as we're now back at the original issue #1010.

closed 2017-06-01 by NeoRaider