freifunk-gluon/gluon #1132: 1043NDv2 hört nicht auf Client IPv6

A-Kasper, 2017-06-01:
Der 1043v2 hat aufgehört seine Dienste auf der Client IPv6 anzubieten. SSH und Statuspage sind darüber nicht zu erreichen. 
aber:
- IPv6 im Client Netz tut
- Traceroute vom Router über IPv6 geht normal übers Freifunk Netz raus
- Er erreicht selbst alle IPv6 Ziele normal
- Die Dienste sind über die lokale IPv4 Adresse am Uplink erreichbar

Möglicherweise nützliche Logs/Ausgaben hier:
https://pastebin.com/kyCiKQDr

Vermutete Ursache: 
Thu Jun  1 07:07:02 2017 daemon.notice netifd: client (1469): cat: write error: Broken pipe

NeoRaider, 2017-06-01:
Does "client IPv6" refer to the IPv6 address of *br-client* in this context, `2a03:2260:300b::32b5:c2ff:fe0e:8b6a`?

The log message you cited is very common and can be ignored.

A-Kasper, 2017-06-01:
yes

NeoRaider, 2017-06-01:
I tried accessing `2a03:2260:300b::32b5:c2ff:fe0e:8b6a` from here, and it is working without problems (I can view the status page, and get an SSH login prompt). I assume it was a temporary and/or local problem.

A-Kasper, 2017-06-01:
I can confirm this. Friends of mine could reach it. It's somehow spooky, because i can ping it and i can connect to all other nodes but however it seems to be a local problem 
-->closed

NeoRaider, 2017-06-01:
A possible explanation: your computer's address is in the WAN network connected to the node you're trying to access. This is a known issue we probably won't be able to solve until we improve separation of WAN and mesh by using network namespaces (which might be impossible if the network namespace support makes the kernel grow too much).

A-Kasper, 2017-06-01:
this sounds like the problem i've got. but can't remember that i had this problem before... thanks for figuring out

A-Kasper, 2017-06-09:
The issue is still present... in irc we found out, that this behaviour could be reproduced with osx and windows. with linux it worked for tioan... 
--> need further investigations where this problem comes from and how we could solve it. 

At a supernode i would try to fix it with sysctl net.ipv6.fwmark_reflect=1 and fwmark for all packages from fastd, but i don't know if something like this is possible with gluon. I you are right, we could try to delete the route for the whole ipv6 network of the uplink and replace it with a route for the single ip of the gateway.

rotanid, 2017-06-09:
@Brother-Lal and me also couldn't reproduce it on Linux, we checked it today

rotanid, 2017-06-17:
i have to correct myself, on another network i saw the same behaviour.
maybe it is not only related with the operating system you use but also the behaviour of the locale router advertising the IPv6 prefix and routes?

feph, 2017-06-18:
I think I am running into the same problem. I have a he.net tunnel (6in4) that provides my lan clients and my FFHH node (on the br-wan interface) with a global v6 address. But I can't reach the uhttpd served status page, nor ping the br-client v6 address from my own network.

I think the problem might be the route selection or firewall (iptables) of the FF node. Weirdly, I can do SYN, SYNACK, ACK  with the node (port 80), but I never receive content. I don't have much intuition into the ip6tables config on the openwrt side, but I've seen some extra rules for SYN/ACK there regarding the wan interface.

Is there a simple way to add a rule that forces outbound v6 traffic (originated from sshd/uhttpd) to take the same route (br-client v6 -> vpn/mesh) as the incoming, requesting traffic? I guess you would have to conntrack/mark some things - I'd really love to find a solution for this.

A-Kasper, 2018-04-08:
Is this fixed in master?

A-Kasper, 2018-08-29:
can't test, but i think this is fixed now. can someone confirm and close?

NeoRaider, 2018-08-29:
We have not changed anything regarding IPv6 routing tables, so I don't think this is fixed.

The only approach to a proper fix that I can think of is to move the WAN interface to a separate network namespace, but that plan still has a large number of missing pieces (that might require us to replace or significantly extend netifd...).

rotanid, 2023-01-30:
we did not observe this problem for years, but with our switch to multiple domains (on a new gateway) the issue is "back". weird.

rotanid, 2023-02-22:
additional info obtained by debugging:
- does not happen when using MacOS or Android, but does happen on "standard" Linux
- it seems on Linux the SSH TCP connection does try to use mtu / package sizes bigger than possible, while MacOS/Android do respect that they have to use a lower mtu / package size