freifunk-gluon/gluon #1133: CPE210(EU) v1.1 Firmware: 1.3.3 - How to Flash Freifunk?

kas515, 2017-06-02:
Hello,

I am not able to Flash a Custom Firmware on the CPE210.
I have read that i can add the needed Checksums with the Safeloader, but i dont Understand How to Use the Safeloader. I have Successfull compiled it on Windows. But i keep getting the Error "partition too big".

Could anyone help me with my Problem?

Regards,

KAS

NeoRaider, 2017-06-03:
What kind of firmware are you trying to flash? The CPE210 factory firmwares built by OpenWrt/LEDE are generated using tplink-safeloader; a usage outside of the OpenWrt/LEDE build system is possible, but not really intended.

kas515, 2017-06-07:
https://vfn-nrw.de/gluon/lindlar/stable/4.2.0/factory/gluon-lindlar-4.2.0-tp-link-cpe210-v1.1.bin
This is the Firmware I was trying to Flash. I want to Rebuild that Image with Safeloader so I could flash it.
The device is a CPE210 EU Version.

NeoRaider, 2017-06-07:
Safeloader is simply the format of the CPE210 factory images (or, to be more precise, the name of the bootloader handling these images). The image you have linked *is* built using the tplink-safeloader tool (CPE210 factory images are by definition), but its version is too old to support the CPE210 EU version.

Please ask your community for updated images; CPE210 EU support was added in Gluon v2016.2.2, released in December 2016.

closed 2017-06-07 by NeoRaider