freifunk-gluon/gluon #1135: regression in reported wifi client count

mweinelt, 2017-06-06:
I was irritated that the client count for my DIR860L was 0 on hopglass, but 8 on meshviewer, so I started looking into the respondd data that the node delivers.

And sure enough `wifi` doesn't show any clients at all.

```
"clients": {
  "total": 0,
  "wifi": 0,
  "wifi24": 8,
  "wifi5": 0
},
```

Adorfer, 2017-06-06:
even if wifi would the sum of wifi24+wifi5 (=8 in this example), the total should be "8" too. 
BTW: i do not get the math here, where there is redundant information about "Wifi" and no figure about  "wired clients". 
In my oppinion: wired, wifi and wifi5  would give everything: 
total= wired+wifi
wifi24= wifi - wifi5 (if present)

mweinelt, 2017-06-06:
@adorfer As far as I can tell `wifi` is collected from the wifi flag in the batman-adv transtable (local), while `wifi24` and `wifi5` are collected from something closer to the wifi stack, like `iw` or `iwinfo`, that has the knowledge to differentiate between the radios (and know their frequencies).

NeoRaider, 2017-06-06:
Interesting. Does 0 clients on wifi5 look correct?

Also what's the output of `iw dev ... station dump` for both radios, and `batctl tl`?

mweinelt, 2017-06-06:
```
root@64289-rocket-cloud:~# iw dev client0 station dump
Station 64:bc:0c:7f:23:2a (on client0)
	inactive time:	20 ms
	rx bytes:	93993
	rx packets:	918
	tx bytes:	206671
	tx packets:	565
	tx retries:	33
	tx failed:	9
	rx drop misc:	0
	signal:  	-57 [-67, -57] dBm
	signal avg:	-57 [-68, -57] dBm
	tx bitrate:	117.0 MBit/s MCS 14
	rx bitrate:	6.0 MBit/s
	expected throughput:	73.699Mbps
	authorized:	yes
	authenticated:	yes
	associated:	yes
	preamble:	short
	WMM/WME:	yes
	MFP:		no
	TDLS peer:	no
	DTIM period:	2
	beacon interval:100
	short preamble:	yes
	short slot time:yes
	connected time:	36 seconds
root@64289-rocket-cloud:~# iw dev client1 station dump
Station a8:7c:01:5a:1c:35 (on client1)
	inactive time:	12906460 ms
	rx bytes:	465
	rx packets:	12
	tx bytes:	2110
	tx packets:	49
	tx retries:	736
	tx failed:	48
	rx drop misc:	0
	signal:  	-85 [-88, -85] dBm
	signal avg:	-85 [-87, -86] dBm
	tx bitrate:	6.5 MBit/s MCS 0
	authorized:	yes
	authenticated:	yes
	associated:	yes
	preamble:	short
	WMM/WME:	yes
	MFP:		no
	TDLS peer:	no
	DTIM period:	2
	beacon interval:100
	short preamble:	yes
	short slot time:yes
	connected time:	12907 seconds
Station 50:f0:d3:fa:7c:e6 (on client1)
	inactive time:	12076250 ms
	rx bytes:	1887
	rx packets:	50
	tx bytes:	2981
	tx packets:	55
	tx retries:	722
	tx failed:	52
	rx drop misc:	0
	signal:  	-89 [-90, -89] dBm
	signal avg:	-88 [-89, -88] dBm
	tx bitrate:	6.5 MBit/s MCS 0
	rx bitrate:	1.0 MBit/s
	authorized:	yes
	authenticated:	yes
	associated:	yes
	preamble:	short
	WMM/WME:	yes
	MFP:		no
	TDLS peer:	no
	DTIM period:	2
	beacon interval:100
	short preamble:	yes
	short slot time:yes
	connected time:	12105 seconds
Station 84:cf:bf:8a:4e:83 (on client1)
	inactive time:	11379050 ms
	rx bytes:	579
	rx packets:	16
	tx bytes:	1868
	tx packets:	45
	tx retries:	594
	tx failed:	41
	rx drop misc:	0
	signal:  	-87 [-87, -88] dBm
	signal avg:	-87 [-87, -87] dBm
	tx bitrate:	6.5 MBit/s MCS 0
	authorized:	yes
	authenticated:	yes
	associated:	yes
	preamble:	short
	WMM/WME:	yes
	MFP:		no
	TDLS peer:	no
	DTIM period:	2
	beacon interval:100
	short preamble:	yes
	short slot time:yes
	connected time:	11392 seconds
Station 84:98:66:8b:97:39 (on client1)
	inactive time:	1812910 ms
	rx bytes:	2237
	rx packets:	23
	tx bytes:	4536
	tx packets:	44
	tx retries:	594
	tx failed:	38
	rx drop misc:	0
	signal:  	-89 [-91, -89] dBm
	signal avg:	-88 [-89, -88] dBm
	tx bitrate:	6.5 MBit/s MCS 0
	rx bitrate:	6.0 MBit/s
	expected throughput:	4.394Mbps
	authorized:	yes
	authenticated:	yes
	associated:	yes
	preamble:	short
	WMM/WME:	yes
	MFP:		no
	TDLS peer:	no
	DTIM period:	2
	beacon interval:100
	short preamble:	yes
	short slot time:yes
	connected time:	1818 seconds
Station f0:db:f8:a4:a5:44 (on client1)
	inactive time:	71650 ms
	rx bytes:	8584
	rx packets:	136
	tx bytes:	6033
	tx packets:	46
	tx retries:	74
	tx failed:	3
	rx drop misc:	11
	signal:  	-85 [-89, -85] dBm
	signal avg:	-86 [-88, -86] dBm
	tx bitrate:	21.7 MBit/s MCS 2 short GI
	rx bitrate:	9.0 MBit/s
	expected throughput:	13.366Mbps
	authorized:	yes
	authenticated:	yes
	associated:	yes
	preamble:	short
	WMM/WME:	yes
	MFP:		no
	TDLS peer:	no
	DTIM period:	2
	beacon interval:100
	short preamble:	yes
	short slot time:yes
	connected time:	114 seconds
```
```
root@64289-rocket-cloud:~# batctl tl
[B.A.T.M.A.N. adv 2017.1, MainIF/MAC: primary0/66:3a:05:3b:00:33 (bat0/90:8d:78:23:af:1c BATMAN_IV), TTVN: 118]
Client             VID Flags    Last seen (CRC       )
60:d9:a0:96:8f:51   -1 [....W.] 368.370   (0x50a92447)
33:33:ff:17:19:9d   -1 [.P....]   0.000   (0x50a92447)
64:bc:0c:7f:23:2a   -1 [....W.]   3.170   (0x50a92447)
33:33:00:00:00:02   -1 [.P....]   0.000   (0x50a92447)
90:8d:78:23:af:1c    0 [.P....]   0.000   (0x5f1834bb)
33:33:ff:7f:23:2a   -1 [.P....]   0.000   (0x50a92447)
33:33:ff:43:54:81   -1 [.P....]   0.000   (0x50a92447)
33:33:00:02:10:01   -1 [.P....]   0.000   (0x50a92447)
01:00:5e:00:00:01   -1 [.P....]   0.000   (0x50a92447)
33:33:ff:00:00:00   -1 [.P....]   0.000   (0x50a92447)
33:33:ff:23:af:1c   -1 [.P....]   0.000   (0x50a92447)
90:8d:78:23:af:1c   -1 [.P....]   0.000   (0x50a92447)
f0:db:f8:a4:a5:44   -1 [....W.] 135.450   (0x50a92447)
33:33:ff:01:00:01   -1 [.P....]   0.000   (0x50a92447)
33:33:00:00:00:01   -1 [.P....]   0.000   (0x50a92447)
```

NeoRaider, 2017-06-06:
Hmm... I assume that clients that have roamed away won't be in the local TT anymore, but they might still be in the WLAN station list. This would also match the very long "inactive time" values seen for some clients.

I guess this behaviour could be improved by filtering clients that have been inactive for too long, but it's probably not a regression.

NeoRaider, 2018-09-06:
@lrnzo Any dot may be replaced with a specific letter (denoting some flag); you'll need to look at the batadv source code for a list of all possible flags. Wireless clients have the W flag; the P flag is for "permanent" addresses, which are either local addresses of the node itself, or multicast addresses. So addresses not having the P or W flags are likely wired clients.

mweinelt, 2018-10-26:
Stations with high inactive time (>300s) are now correctly purged since https://github.com/openwrt/mt76/commit/9d59d2a53fc64facbe0ca5dd8f82c29b8db70739.

rotanid, 2018-10-26:
so your initial reason to open this issue is fixed, but i guess this doesn't mean the overall difference between the client counts is fixed? if it's fixable afterall...

mweinelt, 2018-10-26:
The huge inactive times listed in https://github.com/freifunk-gluon/gluon/issues/1135#issuecomment-306636365 are removed from the equation, but yes the original issue persists.

NeoRaider, 2018-11-17:
We won't be able to get the numbers to match completely - they count different things after all - but we can get them closer in common situtations by further limiting maximum inactivity time for clients we count (in both the batadv and the nl80211 respondd providers).

We'll need to decide on a limit for that. Is one minute too short? I'd like to apply the same limit for wired and wireless batadv clients.

mweinelt, 2018-11-17:
batman-adv timeout for TT entries is 10 minutes (https://git.open-mesh.org/batman-adv.git/blob/HEAD:/net/batman-adv/main.h#l47), timeout for stations is roughly 5 minutes.

I think a minute is just fine.

PR #1573 (2018-11-18): Do not count clients inactive for more than 60s 
Only compile-tested so far, I won't be able to run-test it before tomorrow.

Fixes #1135.
closed 2018-11-18 by NeoRaider