freifunk-gluon/gluon #1140: can't have emoji in node name

MoritzFago, 2017-06-12:
Various things break if you use emoji in the  node name in config mode

NeoRaider, 2017-06-12:
Can you explain what breaks? In theory, any UTF8 string should work.

rubo77, 2017-06-12:
If you use only an emojy as nodename, then the hostname will be empty. Which is untested 👾

NeoRaider, 2017-06-12:
That's not true. When no ASCII characters remain after "pretty name" -> hostname conversion, the hostname will be set to "localhost".

MoritzFago, 2017-06-12:
In  2016.2.5, which is the current version in FF-Kiel, you can't set `👾spaceinvader` or `👾`as a node name, the wizard complains: `One or more fields contain invalid values!`

MoritzFago, 2017-06-12:
Sorry i misattributed some problems as being caused by the unusual hostname, a test device works, except for some unrelated other bug, so i'll close that and i am sorry for all inconvenience i may caused.

closed 2017-06-12 by MoritzFago

Adorfer, 2017-06-13:
in my oppinion a hostname should always be a valid dns localpart. So maybe any punnycode-craze, but valid for a hosts->dns script.

NeoRaider, 2017-06-13:
@Adorfer "Node name" != "hostname". The node name is converted to a hostname by removing all invalid characters; this is done in the same way most Linux distributions do this.

Adorfer, 2017-06-13:
does the respondd provide nodename + hostname? 
just to make it working for users to access their nodes via dns-name. (off course we have to check that both names get displayed in the network visualization ui.

NeoRaider, 2017-06-13:
The hostname is only used internally, and never transmitted over the network. The existence of an internal hostname can be ignored for most purposes (we only set it so it is shown in the shell prompt).

DNS integration of names announced via respondd could work by converting the name to Punycode (if possible), or simply by expecting users who want DNS to work not to use invalid characters.

Note that nothing about the node name handling has changed since Gluon v2016.2.