freifunk-gluon/gluon #1141: Internt of Things Paket (IoT)

rubo77, 2017-06-14:
Ich versuche ein  IoT Paket zu erstellen, anhand des private-wlan paketes. 

Dies wäre der Text:

> Dein Knoten kann ein zusätzliches separates WLAN aufbauen, das den daran angeschlossenen Geräten keinen Internetzugang ermöglicht und auch keinen Zugriff auf dein Privates WLAN. Diese Funktionalität ist völlig unabhängig von den Mesh-Funktionen des Knotens. Beachte, dass du nicht gleichzeitig das Meshen über den WAN-Port aktiviert haben solltest.

Soweit bin ich schon: https://github.com/rubo77/gluon/commit/81b4f10ddbb328d3676ff74ece087874251fc6fa

Was noch fehlt:

- Ich bin mir nicht sicher ob das mit dem local primary_iface = 'wan_radio2' einfach geht?
- und es fehlt noch die funktion, dass dieses wlan dann weder ins private noch ins freifunk netz darf, wie kann man das umsetzen?

Wär cool, wenn mir hier ein Profi helfen könnte, ich nehme PRs in meinem gluon-fork an

rubo77, 2017-06-14:
Das ist besonders dann sinnvoll, wenn man Internet of Things Geräte, wie Glühbirnen oder Microwellen mit extrem veralteter Software ein wlan zur Kommunikation bereitstelen will und diese aber gleichzeitig abschirmen will von seinem eigenen WLAN

NeoRaider, 2017-06-14:
Der Issue-Tracker ist nicht der richtige Ort für solche Fragen, dafür haben wir eine Mailing-Liste.

closed 2017-06-14 by NeoRaider