freifunk-gluon/gluon #1142: respondd does not listen on br-client

jplitza, 2017-06-17:
After the update to 2016.2.6 (but probably not really related to that), the respondd on one of my nodes was started with these arguments:
```
 1825 root      2260 S    /usr/bin/respondd -g ff02::2:1001 -p 1001 -d /lib/gluon/respondd -i ibss0 -i br-wan
```
While by itself this is okay, it should have been restarted when the client interface came up. But it wasn't. Executing `/etc/init.d/gluon-respondd reload` correctly included the interface `-i br-client` in the argument list, meaning that the argument building in the init script seems to work.

I suspect a race condition on startup: The init script fetches the list of interfaces while the client interface is still down, but before it finishes, the interface comes up and procd doesn't execute the reload because the first run of the init script hasn't exited yet.

mweinelt, 2018-11-26:
Is this issue reproducible and still relevant?

NeoRaider, 2018-11-30:
The code has changed significantly since 2016.2.x. Closing until further notice.

closed 2018-11-30 by NeoRaider