freifunk-gluon/gluon #1143: UAP Outdoor+ update

woerde, 2017-06-17:
Since at least v2016.2.1 using freifunk-muensterland.de builds, an erratic behavior is observed. All observations were done in situations where no LAN cable is attached to the UAP Outdoor+ but connection is via WLAN-mesh.

1. Flashing a new firmware or updating without keeping the settings (via luci or uci) works correct and stably

2. Updating with keeping the settings (via autoupdate, luci, or uci) leads always (?) to an state of erratic functionnality:

- the SSID "Freifunk" (not "Freifunk-Offline") is distributed, indicating the AP believes it is connected to the FF backbone via mesh. However,  no connection of clients can be established successfully (android: "waiting for IP-configuration")

- the device cannot be stably accessed by its ipv6 address (ssh, http).

- the device is occasionally visible at the status page of other devices in the local Freifunk cloud. But it provides only few information (mac address, correct reading of the firmware version) but no information about its connection status to other devices, in contrast to the normal function.

In short: the update brings the UAP Outdoor+ into a state where it has only part of the usual functionallity.

The solution in all observed cases (5 different devices, fw v2.1, 2.4, 2.5, 2.6) was an update without keeping the settings to any FW version.

If interesting I can provide logs, hardware, more information, etc. for diagnosis..

NeoRaider, 2017-06-19:
I don't know how the SSID-changer package works, as it is not an official Gluon packages, so I can't comment on that issue.

For the other issues, please provide the output of the `ip a` and `logread` commands, and the contents on /etc/config/network and /etc/config/wireless.

woerde, 2017-06-24:
Thank you for the fast reply and sorry for the delayed delivery.

Due to my limited skills I had to connect the problematic device directly to my home lan and hence to the internet (rather than connecting to the device via WLAN) in order to get ssh access. This means the following information describes a different state of the hardware configuration.

root@freifunkOstbevernTest2:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel master br-wan qlen 1000
    link/ether 80:2a:a8:a0:84:60 brd ff:ff:ff:ff:ff:ff
3: eth1: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc fq_codel master br-mesh_lan qlen 1000
    link/ether 82:2a:a8:a0:84:60 brd ff:ff:ff:ff:ff:ff
4: dummy0: <BROADCAST,NOARP> mtu 1500 qdisc noop
    link/ether 52:2e:9e:4f:7e:e8 brd ff:ff:ff:ff:ff:ff
5: teql0: <NOARP> mtu 1500 qdisc noop qlen 100
    link/void
7: br-mesh_lan: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue
    link/ether 6e:c7:33:a0:31:e4 brd ff:ff:ff:ff:ff:ff
    inet6 fe80::6cc7:33ff:fea0:31e4/64 scope link
       valid_lft forever preferred_lft forever
8: br-wan: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue
    link/ether 6e:c7:33:a0:31:e0 brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.43/24 brd 192.168.1.255 scope global br-wan
       valid_lft forever preferred_lft forever
    inet6 2002:5b3e:49b5:0:6cc7:33ff:fea0:31e0/64 scope global dynamic
       valid_lft 6914sec preferred_lft 3314sec
    inet6 fe80::6cc7:33ff:fea0:31e0/64 scope link
       valid_lft forever preferred_lft forever
13: mesh0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1532 qdisc mq master bat0 qlen 1000
    link/ether 6e:c7:33:a0:31:e1 brd ff:ff:ff:ff:ff:ff
    inet6 fe80::6cc7:33ff:fea0:31e1/64 scope link
       valid_lft forever preferred_lft forever
14: client0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq master br-client qlen 1000
    link/ether 80:2a:a8:a1:84:60 brd ff:ff:ff:ff:ff:ff
    inet6 fe80::822a:a8ff:fea1:8460/64 scope link
       valid_lft forever preferred_lft forever
15: br-client: <BROADCAST,MULTICAST> mtu 1500 qdisc noop
    link/ether ea:e8:3b:fb:c5:fc brd ff:ff:ff:ff:ff:ff
16: primary0: <BROADCAST,NOARP,UP,LOWER_UP> mtu 1532 qdisc noqueue master bat0
    link/ether 6e:c7:33:a0:31:e3 brd ff:ff:ff:ff:ff:ff
17: bat0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue
    link/ether 80:2a:a8:a0:84:60 brd ff:ff:ff:ff:ff:ff

root@freifunkOstbevernTest2:~# logread
Tue Jun 13 00:19:54 2017 kern.debug kernel: [   14.210000] ath: EEPROM indicates default country code should be used
Tue Jun 13 00:19:54 2017 kern.debug kernel: [   14.210000] ath: doing EEPROM country->regdmn map search
Tue Jun 13 00:19:54 2017 kern.debug kernel: [   14.210000] ath: country maps to regdmn code: 0x3a
Tue Jun 13 00:19:54 2017 kern.debug kernel: [   14.210000] ath: Country alpha2 being used: US
Tue Jun 13 00:19:54 2017 kern.debug kernel: [   14.210000] ath: Regpair used: 0x3a
Tue Jun 13 00:19:54 2017 kern.debug kernel: [   14.220000] ieee80211 phy0: Selected rate control algorithm 'minstrel_ht'
Tue Jun 13 00:19:54 2017 kern.info kernel: [   14.230000] ieee80211 phy0: Atheros AR9280 Rev:2 mem=0xb0000000, irq=40
Tue Jun 13 00:19:54 2017 daemon.notice haveged: haveged starting up
Tue Jun 13 00:19:58 2017 daemon.info haveged: haveged: ver: 1.9.1; arch: generic; vend: ; build: (gcc 4.8.3 CV); collect: 128K
Tue Jun 13 00:19:58 2017 daemon.info haveged: haveged: cpu: (); data: 32K (P); inst: 32K (P); idx: 19/40; sz: 32699/67115
Tue Jun 13 00:19:58 2017 daemon.info haveged: haveged: fills: 0, generated: 0
Tue Jun 13 00:20:02 2017 authpriv.info dropbear[1051]: Not backgrounding
Tue Jun 13 00:20:05 2017 daemon.warn netifd: You have delegated IPv6-prefixes but haven't assigned them to any interface. Did you forget to set option ip6assign on your lan-interfaces?
Tue Jun 13 00:20:05 2017 kern.info kernel: [   29.120000] device eth1 entered promiscuous mode
Tue Jun 13 00:20:05 2017 kern.info kernel: [   29.140000] br-mesh_lan: port 1(eth1) entered forwarding state
Tue Jun 13 00:20:05 2017 kern.info kernel: [   29.140000] br-mesh_lan: port 1(eth1) entered forwarding state
Tue Jun 13 00:20:05 2017 daemon.notice netifd: Interface 'mesh_lan' is enabled
Tue Jun 13 00:20:05 2017 kern.info kernel: [   29.180000] device eth0 entered promiscuous mode
Tue Jun 13 00:20:05 2017 daemon.notice netifd: Interface 'wan6' is enabled
Tue Jun 13 00:20:05 2017 daemon.notice netifd: Interface 'wan' is enabled
Tue Jun 13 00:20:05 2017 kern.info kernel: [   29.230000] IPv6: ADDRCONF(NETDEV_UP): br-wan: link is not ready
Tue Jun 13 00:20:05 2017 daemon.notice netifd: Interface 'mesh_wan' is enabled
Tue Jun 13 00:20:05 2017 daemon.notice netifd: Interface 'loopback' is enabled
[... 286 lines cut ...]
        list basic_rate '6000'
        list basic_rate '9000'
        list basic_rate '18000'
        list basic_rate '36000'
        list basic_rate '54000'

config wifi-iface 'mesh_radio0'
        option ifname 'mesh0'
        option network 'mesh_radio0'
        option device 'radio0'
        option mesh_fwding '0'
        option mesh_id '02:d1:11:37:fc:77'
        option mcast_rate '12000'
        option mode 'mesh'
        option disabled '0'
        option macaddr '6e:c7:33:a0:31:e1'

config wifi-iface 'client_radio0'
        option ifname 'client0'
        option network 'client'
        option disabled '0'
        option device 'radio0'
        option mode 'ap'
        option macaddr '80:2a:a8:a1:84:60'
        option ssid 'Freifunk'

NeoRaider, 2017-06-24:
This is a duplicate of #1079, which is fixed in Gluon v2017.1. Unfortunately, the solution was deemed to complex for a backport to v2016.2.x, so an upgrade to a newer version is advisable (at least after the upcoming release of Gluon v2017.1.1, as v2017.1 has a few problematic bugs...)

closed 2017-06-24 by NeoRaider