freifunk-gluon/gluon #1147: Configmode via Reset-BTN not possible on TL-WA7510N

mweinelt, 2017-06-21:
I don't have a device currently to verify this, but there is sufficient evidence, that entering the configmode via the reset button on the TL-WA7510N is not possible.

The reset gpio pin is possibly wrongly configured (via @neoraider)

https://forum.freifunk.net/t/tl-wa7510n-v1-3-kein-reset-auf-configmode-bei-gluon-2016-moeglich/12260/4?u=hexa
https://forum.darmstadt.freifunk.net/t/konfigurationsmodus-tp-link-tl-wa7510n-v1/194/3?u=hexa

In our case this is on 2016.2.6.

rotanid, 2017-06-22:
relatively easy to fix, if someone gets hands on a device

rotanid, 2018-01-19:
@mweinelt looks like no one cares enough to provide his/her device or fix it him/herself.
the device is from 2012, 4mb flash, 32mb ram... 
do we want to keep such issues open forever? (that's what will happen to this one i guess)

mweinelt, 2018-04-16:
No.

closed 2018-04-16 by mweinelt