freifunk-gluon/gluon #1148: 802.11s mesh down for several hours

bobcanthelpyou, 2017-06-22:
Freifunk Ilmenau (802.11s [site-ffil](https://github.com/freifunk-ilmenau/site-ffil)) has only a small network with around 20 nodes, we are a subsite of the batman adv v14 based Freifunk Erfurt (ibss [site-ffef](https://github.com/FreifunkErfurt/site-ffef)) with around 200 nodes. 

Most of our nodes are mesh-only nodes without a fastd uplink. At the moment nodes with Gluon v2016.2, v2016.2.4 and v2017.1 are online (of our ffil subsite).

We see the following issue from time to time with different TP Link models within our network:
- the client wifi interface of a node is up but all mesh connections are down for several hours
- on our map nodes with fastd uplink are still online and mesh-only nodes are shown as offline
- the status page of the affected node does not show any mesh0 nodes around

@NeoRaider suggested to do a ```iw dev mesh0 station dump``` on an affected node. Are there any other information which could be useful to bring more light up to that issue?

rotanid, 2017-06-22:
are you sure the described issue affects every one of your three firmware versions?
i wouldn't mind if v2016.2 is affected, that may very well be.

bobcanthelpyou, 2017-06-22:
I am not totally sure about v2016.2 (there is only one node online) but v2016.2.4 is affected by that issue for sure.

Sunz3r, 2017-06-22:
I can confirm this with v2016.2.5 on several WR841. It happens often on high traffic or airtime. There is nothing in dmesg or logread if this problem occur.
After "iw dev client0 scan" all Wifi-Mesh-Links are back.
https://pastebin.com/uW0NC7J8

On another Node i found this in dmesg. I dont know that this has a coherence:
https://pastebin.com/n8zgqB7E

NeoRaider, 2017-06-22:
Gluon v2016.2.x and earlier are known for poor ath9k stability. It would be interesting if you can also see the issue between nodes running v2017.1.

rubo77, 2017-06-23:
We use this quickfix to solve this as a workaround:

https://github.com/Freifunk-Nord/eulenfunk-packages/tree/lede/gluon-quickfix

rotanid, 2017-06-25:
please don't always propagate hard workarounds, the bugs will become more and more serious if we don't dig into them but just use some workaround. @rubo77 (but others do the same thing)
let's see what they have to say about v2017.1 nodes beneath eachother.

bobcanthelpyou, 2017-06-27:
It happened again to a TP-Link TL-WDR4300 v1 with v2017.1. Neighbor nodes running Gluon v2017.1 and v2016.2.4

https://ilmenau.freifunk.net/meshviewer/#!v:m;n:30b5c23839e3

The mac address (no node name) of the affected node is visible via the status page (mesh0) of the neighbor nodes. Sadly, I don't have physical access to the node at the moment and can't gather any additional information.

Could some information of the neighbor nodes could help here, too?

valcryst, 2017-07-29:
This is also happening on all nodes we have recently upgradet from 2015 to 2016/2017.
Its a total mess, because we already lost around 400 routers that are not coming back without a powercycle.
If we had known this before, we would still be on 2015.1.2.

So why was there no warning about critical instabillity ?
"poor ath9k stability" i would call this "unuseable ath9k stability"

What can we do now to improve stability, flashing the next beta release where noone knows
what happens? 
Use quickfix from Fichtenfunk and this script from FFFM?
Sorry for being so rude, but i would NEVER call this a stable release.

A-Kasper, 2017-07-29:
I agree with your critic on calling it stable. I would had also recalled
the release after finding this critical bugs... but:

What would have happened if it would still be master... correct: nobody
would still give it a try in bigger environments and the bug would still be
hidden.

Gluon has a very hard problem with it's community. We would need much more
people writing bugreports and testing patches. The developers are making a
very good job. donated to the users. free. but they really need feedback. I
could just ask everyone to compile and test the master and write bugreports
if something doesn't work.

For Releases I would like to have something like feature complete release
candidates where everybody is clear that there shouldn't be any bugs, but
we should test if there is something undetected..... but as already
mentioned: the main problem is that the community needs to support the
developers with more testing. I don't really think that someone who don't
want to test the master wants to test a release candidate....

rotanid, 2017-07-29:
please stay on topic! such discussions should use the mailing list.
the problem @valcryst is describing probably has nothing to do with 802.11s and meshing, but may be #1160 - there's a fix already for master and v2017.1.x

bobcanthelpyou, 2017-08-05:
We still hit that meshing problem (mesh0 of affected node is up, the node is seen by neighbour nodes, it is listed on the status pages, but no mesh connection) mostly for one or several hours. Running v2017.1.1 on e.g. WR841N-v10/v11

@NeoRaider mentioned an upstream issue in the LEDE bug tracker via IRC, unfortunately i can't find it any more.

FreifunkUFO, 2017-10-16:
see https://bugs.lede-project.org/index.php?do=details&task_id=863

atm its a well-known problem, "wifi doenst work until a wifi-scan is done", and many communities with many different bugfixes in the wild..

rotanid, 2018-07-12:
please check if this bug is still there with Gluon v2017.1.8 or Gluon v2018.1, if yes, try Gluon master branch (based on OpenWrt 18.06)

bobcanthelpyou, 2018-07-14:
That behavior didn't hit our routers lately. (with the latest v2017.1.x)

I could not reproduce it. It is fixed with the latest updates or our setup changed where it can't be reproduced.

I close the issue. Everyone with a similar issue and a resent Gluon version should open a new issue.

bobcanthelpyou, 2019-03-19:
We hit that issue or a very similar issue again with several TP-Link TL-WR841N/ND v11 devices in our network. Checking statistic data the nodes with neighbors and clients around "all the time" and a Vpn uplink connection seem to be affected as the clients and mesh nodes drops to zero at the same time in "error stage" for several hours, reporting neighbors and client (both zero) via respondd. Mesh only nodes are unavailable for meshing for several hours, after "error state" up time wasn't reset. But it could be an interference of other wifi stations (etc.), too. For other nodes (without neighbors, and rarely clients) it is hard to tell from statistics if these are effected, too. It seems to happen randomly.

ilmenau-WePlaFu-1
Gluon: v2018.2.1 (was v2018.1.4)
Vpn: yes
Map: https://test.c-wellie.de/#/de/graph/ec086b591ede
Issue:
* [2019-03-17 10:00pm - 2019-03-17 11:10pm (node was manuelly rebooted)](https://test.c-wellie.de/grafana/d/W-3XhpImk/statistiken-by-nodeid?orgId=1&var-nodeid=ec086b591ede&from=1552855257098&to=1552863880826)

BVClub13-Freifunk2
Gluon: v2018.1.4 (was v2017.1.8)
Vpn: yes
Map: https://test.c-wellie.de/#/de/graph/f4f26df2a80a
Issue:
* [2019-03-18 03:15pm - 2019-03-19 08:40am](https://test.c-wellie.de/grafana/d/W-3XhpImk/statistiken-by-nodeid?orgId=1&var-nodeid=f4f26df2a80a&from=1552914620968&to=1552985332954)
* [2019-03-15 10:45pm - 2019-03-15 11:50am](https://test.c-wellie.de/grafana/d/W-3XhpImk/statistiken-by-nodeid?orgId=1&var-nodeid=f4f26df2a80a&from=1552636964811&to=1552652747862)

ilmenau-ec086b3529aa
Gluon: v2018.1.4 (was v2017.1.8)
Vpn: mesh only
Map: https://test.c-wellie.de/#/de/graph/ec086b3529aa
Issue:
* [2019-03-17 07:30am - 2019-03-17 11:20am](https://test.c-wellie.de/grafana/d/W-3XhpImk/statistiken-by-nodeid?orgId=1&var-nodeid=ec086b3529aa&from=1552800140245&to=1552820371651)

ilmenau-bv17-besenwesen
Gluon: v2018.1.4 (was v2017.1.8)
Vpn: mesh only
Map: https://test.c-wellie.de/#/de/graph/18a6f7e5cbd4
Issue:
* [2019-03-11 02:15pm - 2019-03-11 03:45pm](https://test.c-wellie.de/grafana/d/W-3XhpImk/statistiken-by-nodeid?orgId=1&var-nodeid=18a6f7e5cbd4&from=1552307393275&to=1552318235135)

bobcanthelpyou, 2019-03-23:
`iw mesh0 station dump` is empty on an affected node and `iw dev mesh0 scan` brings the wifi interface still up, again.

After that a station (cpe210) shows up with `tx bitrate:	1.0 MBit/s` (via `iw mesh0 station dump`) running `iw dev mesh0 station del ...` reset it to `tx bitrate:	65.0 MBit/s MCS 7`

But it is untypical that those stations mesh with each other cause of bad signal (the cpe210 is visible for the wr841n but the wr841n is not visible for the cpe210).

bobcanthelpyou, 2019-03-26:
In other related issues some people mentioned ANI and promote to disable it.

ANI on affected node:
```
# cat /sys/kernel/debug/ieee80211/phy0/ath9k/ani
            ANI: ENABLED
      ANI RESET: 3874
     OFDM LEVEL: 5
      CCK LEVEL: 4
        SPUR UP: 7466
      SPUR DOWN: 7466
 OFDM WS-DET ON: 0
OFDM WS-DET OFF: 0
     MRC-CCK ON: 4890
    MRC-CCK OFF: 7206
    FIR-STEP UP: 13106
  FIR-STEP DOWN: 6972
 INV LISTENTIME: 0
    OFDM ERRORS: 621648521
     CCK ERRORS: 104652563
```

ANI after `iw .. scan`
```
# iw dev mesh0 scan
[...]
# cat /sys/kernel/debug/ieee80211/phy0/ath9k/ani
            ANI: ENABLED
      ANI RESET: 3888
     OFDM LEVEL: 2
      CCK LEVEL: 4
        SPUR UP: 7466
      SPUR DOWN: 7466
 OFDM WS-DET ON: 0
OFDM WS-DET OFF: 0
     MRC-CCK ON: 4890
    MRC-CCK OFF: 7207
    FIR-STEP UP: 13108
  FIR-STEP DOWN: 6972
 INV LISTENTIME: 0
    OFDM ERRORS: 622148794
     CCK ERRORS: 104795311
```

Node is back online.

FreifunkUFO, 2019-03-29:
the hint about ANI is to disable and re-enable it! 
(seems to be almost the same as a wifi scan, because there are some routines inside ANI with exceptions when doing wifi-scanning)
p.s. meanwhile there are even many more mesh-communities with even more different bugfixes in the wild..

mweinelt, 2020-03-03:
@bobcanthelpyou @Sunz3r @valcryst Is this still reproducible on v2019.1 and newer? If not I'm voting to close this issue.

bobcanthelpyou, 2020-03-22:
Hi @mweinelt, i put it on my todo list for this week, but i am not sure if i have the suitable hardware at hand, lets see.

rotanid, 2020-09-02:
closing due to lack of answers or new reports.

closed 2020-09-02 by rotanid

Djfe, 2024-05-06:
we just spotted this on a TP-Link CPE510 v1 running Gluon v2023.2.2
Maybe it makes sense to reopen this issue?
```ath: phy0: unsupported hw bitrate detected 0x1b using 1 Mbit```

https://map.aachen.freifunk.net/#!/de/map/c4e984b0b992