freifunk-gluon/gluon #1154: Raspberry Pi 2 is not connecting to Mesh-VPN

Maverick3008, 2017-06-25:
Gluon 2017.1

Although I installed the new gluon version (factory) on my raspberry pi2, it is not connecting to Mesh-VPN.

rotanid, 2017-06-25:
what do the logs say?
`logread`

Maverick3008, 2017-06-25:
That was new for me. Thx. Here is the requested file:

[Logread.txt](https://github.com/freifunk-gluon/gluon/files/1100327/Logread.txt)

rotanid, 2017-06-25:
DNS lookup of the gateway hostnames fails. can't see why...

NeoRaider, 2017-06-26:
Please provide the output of `ip a` and `ip r`, and the contents of `/etc/config/network`.

Maverick3008, 2017-06-26:
Here are the requested outputs ip a and ip r again a second logread and the contents of /etc/config/network.. 
[ip a.docx](https://github.com/freifunk-gluon/gluon/files/1103192/ip.a.docx)
[ip r.docx](https://github.com/freifunk-gluon/gluon/files/1103195/ip.r.docx)
[logread2.docx](https://github.com/freifunk-gluon/gluon/files/1103196/logread2.docx)
[:etc:config:network.docx](https://github.com/freifunk-gluon/gluon/files/1103204/etc.config.network.docx)

mweinelt, 2017-06-26:
@Maverick3008 Plaintext would be much appreciated over any proprietary format.

NeoRaider, 2017-06-26:
@Maverick3008 What's the content of `/tmp/gluon/wan-dnsmasq/resolv.conf`?

Maverick3008, 2017-06-26:
Content of /tmp/gluon/wan-dnsmasq/resolv.conf

nameserver fd00::ca0e:14ff:feab:2cde
nameserver 192.168.188.1

NeoRaider, 2017-06-26:
Hmm, the IPv6 nameserver address is a bit unusual (as it is not in your prefix), but it might be correct (does this address work as a nameserver from other devices in your network?). Anyways, it should fall back to the v4 nameserver in case the v6 one doesn't work. Can you `nslookup node12.freifunk-bochum.de 192.168.188.1`? (the v6 one will not work on the command line because of Gluon's routing rules)

Maverick3008, 2017-06-26:
Server:		192.168.188.1
Address:	192.168.188.1#53
Name:      node12.freifunk-bochum.de
Address 1: 212.23.154.21
*** Can't find node12.freifunk-bochum.de: No answer

NeoRaider, 2017-06-26:
Hmm, I'll have to investigate if your IPv6 nameserver being unreachable could be a Gluon issue; the fallback to IPv4 not working is probably a dnsmasq bug.

As a workaround could try disabling IPv6 on the WAN port in the "advanced settings".

A-Kasper, 2017-08-17:
I've had some time to investigate ths a little bit further. It seems like we were on the wrong way:

The node couldn't reach the gateway. So it can't reach the DNS and can't connect to the supernode.
After discussing this with Maverick3008 and looking at the fritz.box we found out, that the IP in the box was different from the nodes one. After rebooting everthing works fine.

What probably happened: 
The gluon node is connected via a port in a guestnet of the fritzbox. To be able to config it, the port was configured in the main net (different ip block). After that the node was rebootet got an IP in the main-network and then the fritzbox was configured as guest network. The gluon node had the wrong ip configuration from the main net and couldn't reach gateway, dns etc. because the node was in the main net a short time, the node knew the dns entry from cache.... 

Maverick3008 is on hollidays, but we discussed the behaviour, it works now and we have an idea how this could have happened. I think there is nothing in gluon to fix. 

The only thing:
I don't know what happens at fritz.boxes, if they are reconfigured this way. It could be possible, that gluon didn't recognized the missing Link to get a new adress. br-wan is a brdige. maybe there is a way to detect such events and reconfigure.

rotanid, 2017-08-17:
afaik there wont be a new address if the gateway stays the same.
and as the Fritzbox has the same mac address in both subnets, the dhcp client has no reason to ask for a new address.
"I think there is nothing in gluon to fix." -> closing. can be used for discussion nevertheless.

closed 2017-08-17 by rotanid

A-Kasper, 2017-08-17:
the gateway changes. the new gateway is in another subnet. That's why a new
IP would be needed.
I think that gluon possibly don't recognize the changing link state when
reconfiguring the fritzbox, because br-wan is a bridge and just the eth0
interface has the link status trigger. but this is really wild speculation

NeoRaider, 2017-10-25:
It is likely that this issue is the same as #1245 after all (but we were unable to see the issue in the logs here because of the kernel log spam...). A fix should be available soon.