freifunk-gluon/gluon #1155: wrong txpower on WR841 with 2017.1

Sunz3r, 2017-06-25:
It seeems the antenna offset is missing.

```
# iwinfo client0 info
client0   ESSID: "Freifunk"
          Access Point: 9A:40:22:85:8E:68
          Mode: Master  Channel: 6 (2.437 GHz)
          Tx-Power: 20 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -95 dBm
          Bit Rate: unknown
          Encryption: none
          Type: nl80211  HW Mode(s): 802.11bgn
          Hardware: unknown [Generic MAC80211]
          TX power offset: unknown
          Frequency offset: unknown
          Supports VAPs: yes  PHY name: phy0
# cat  /sys/kernel/debug/ieee80211/phy0/power
20
```

I compared the Signal-Strength with two WR841 (2017.1 and 2016.2.6) towards a third Node and between.

Adorfer, 2017-06-25:
is this just an "cosmetic" issue in the iwinfo or does it affect the real electrial antenna output?

Sunz3r, 2017-06-25:
I think this is real ... CPE210 with 2017.1 has the same bug.

Adorfer, 2017-06-25:
what bug? no display of antenna-offset in iwinfo?

Sunz3r, 2017-06-25:
CPE210 with 2017.1:
```
# iwinfo client0 info
client0   ESSID: "Freifunk"
          Access Point: 72:C6:50:C7:ED:30
          Mode: Master  Channel: 6 (2.437 GHz)
          Tx-Power: 20 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -92 dBm
          Bit Rate: unknown
          Encryption: none
          Type: nl80211  HW Mode(s): 802.11bgn
          Hardware: unknown [Generic MAC80211]
          TX power offset: unknown
          Frequency offset: unknown
          Supports VAPs: yes  PHY name: phy0
# cat  /sys/kernel/debug/ieee80211/phy0/power
20
```

rotanid, 2018-02-15:
please report and discuss this upstream

2tata, 2018-02-21:
https://bugs.lede-project.org/index.php?do=details&task_id=1289

Seems reported.

CodeFetch, 2019-02-24:
Ist gefixt in https://github.com/openwrt/openwrt/commit/2389ebbd48d2073609b6acff9a3507b1daa36e36 und https://github.com/lede-project/source/commit/2389ebbd48d2073609b6acff9a3507b1daa36e36

rotanid, 2019-02-24:
@CodeFetch how do you know these are fixing the issue of this bug report here? the upstream bugreport is still open, too. and neither the issue author upstream nor here has reported back.
have you tested and compared yourself?

CodeFetch, 2019-02-24:
@rotanid As far as I understand it the original issue was that devices if you use another regdomain than for what the values in the EEPROM were specified the router cannot send with a higher power even if it is allowed in the other regdomain. The patch just ignored the EEPROM settings which led to a too high power output as vendors used this ATH_REGDOM setting to specify the antenna offset. The original patch circumvented vendor specifications. This should not have been merged at all.

This https://github.com/openwrt/openwrt/pull/885 is a similar pull-request that was declined and there you can find comments why you should not do this.

rotanid, 2019-02-25:
thanks for elaborating.
can you comment in upstream bugtracker with your reasoning to close it, too? would be great.

closed 2019-02-25 by rotanid