freifunk-gluon/gluon #1156: worse range and bandwidth with CPE210 and WBS210

Sunz3r, 2017-06-25:
I test CPE210 and WBS210 on several location and compare it with an WR841. 
The Signal-Strength is good but TQ and RX-MCS-Values are very bad.

Can anyone test CPE210 with 2017.1?

---

Ein WR841 war im Vergleich mit CPE210/WBS210 um längen besser. Die Signalstärke ist super aber die Verbindung (TQ und RX-MCS-Werte) nicht.
Getestet an verschiedenen Standorten, Geräte und Antennen.

A-Kasper, 2017-06-25:
we trsted it already after some talk with neoraider and t_x we found out
the the routers are sending full table requests. this leads to a peak of
mgmt traffic and low airtime. are you sure that you are talking about the
txpower. i think this sounds like the problem we discussed one week ago in
irc

rotanid, 2017-06-25:
@A-Kasper they didn't test gluon v2017.1 yet, so that isn't the same issue.

horstfffl, 2017-06-25:
i can confirm what seems to be an cpe210 lna issue
last time tested with a build 6c053d9 from 20. april on two cpe210 devices
but problem was there all the time

a rocketm2+15db sector at same location gets about 3db worse rx from an 1,2km distant router(841nd)
(rx from that distant router at rocketm2+15db sector antenna+lna is 3db worse then cpe210+9db antenna+lna)

that means the cpe210 lna seems to have up to 9db more amplification then the rocketm2 lna? 

the rocketm2 gets stabil connection at that distance with TQ of about  60-90
the cpe210 gets no TQ at all because every few seconds the links goes blank(at least what the live status page shows) like i said with a 3db better rx.

because of the in the comparison to the rocketm2 higher rx db and the frequent link failures i guess the cpe210 lna is amplifiing to much.

Adorfer, 2017-07-12:
Please be aware, that the difference betwenn DE and US for txpower is 12 (US: 23dBm, DE: 11dBm ) on CPE210. 
(In other TP-Link models like WR841 it's about 2-3 dBm difference, depending on the selected channel)

rotanid, 2018-02-15:
please report and discuss this upstream

rotanid, 2018-11-25:
more information on this issue can be found in the closed issues #1023 (CPE510) and #1063 (WBS210)

closed 2020-03-03 by mweinelt