freifunk-gluon/gluon #1160: autoupdate 2017.1.0 2017.1.1 failed. powercycle needed

A-Kasper, 2017-06-26:
I've tried to autoupdate from 2017.1.0 to 2017.1.1 (candidate, actual 2017.1.x build). 
I needed to powercycle both devices: a Futro and a CPE220 to bring them up again.

A-Kasper, 2017-06-26:
I became some more feedback... thes problem seems to happen on every node we've tried to update

Sunz3r, 2017-06-26:
same with WR841 (2016.2.6 -> 2017.1) and WDR3600 (master -> 2017.1)

NeoRaider, 2017-06-26:
As I've never seen this issue, logs of a failed upgrade would be very helpful.

A-Kasper, 2017-06-27:
how can these logs be generated? afaik is the logread log empty after fw update and powercycle

A-Kasper, 2017-06-27:
https://pastebin.com/P7MSi23s

this is the ssh output of the futro yesterday... i saved it before i knew, that the powercycle would be needed, because of these timeouts...

Adorfer, 2017-06-28:
if you have a console when this is happending: please check if the unit is coming up with br-client fully set up (with ipv6-dresses aothers than fe80-ones)

NeoRaider, 2017-06-28:
The only way to get a proper log is a serial console.

A-Kasper, 2017-06-30:
There is also an error if you sysupgrade via console. Maybe this is related?
[image]

A-Kasper, 2017-06-30:
sorry.. seems like this is the "normal" errormessage of the chromeos sshclient when disconnected... ignore the last input to this issue

A-Kasper, 2017-07-04:
most router seem to be affected. i hope i can provide more details soon, but i'm not sure

A-Kasper, 2017-07-07:
I've tried to get log files, but after updating most nodes in the network the failure seems to be disappered. I think the autoupdater is affected by the fixed ram-flooding bug. maybe it could be checked if enough RAM is available before starting the update....

rotanid, 2017-07-08:
question is, what should be done if there isn't enough ram, if you include such a check?

kpanic23, 2017-07-11:
We have just updated >70 nodes by hand via SSH. We rebooted every single node via SSH and applied the update thereafter. Not a single one crapped out.

What about patching the autoupdater:
After downloading the manifest and noticing that a newer version is available, the autoupdater could check if uptime >X reboot the node first. After the reboot, the next time the autoupdater runs uptime will be <X, therefore the autoupdater continues to flashing the firmware.
This might reduce the possibility of a crash during autoupdate.
ToDo: sensible value for X (perhaps 1d? 4h?)

A-Kasper, 2017-07-11:
I like your Idea.. but I would make it a little bit easier:
If newfirmware download it, check md5 and reboot
If firmwarefile available: flash it and reboot, else start services as usual.

A-Kasper, 2017-07-11:
hmm.. I'm unsure if that was correct
the image in RAM could be saved through softreboot, but i'm not sure if it could cause problems because also other content may be saved... maybe my concept was to short...

NeoRaider, 2017-07-11:
We can't really save anything big in RAM during a reboot, the crashlog is more something like "well, we'll write the logs to some specific place in RAM and hope they're still fine after reboot".

Please note that there are already lots of fallbacks in place that will reboot the node when something goes wrong during upgrade. For some of these fallbacks it shouldn't even be possible to fail. So we have a very weird situation in which the hardware doesn't work at all anymore, but the hardware watchdog doesn't trigger.

New autoupdater timing schemes might be possible, but we 1) don't want to reboot more than necessary to keep the outage on upgrades as short as possible, and 2) we need to keep in mind that for "regular" upgrade mode, we check only for upgrades every 24h at the moment.

Please also have a look at https://youtu.be/IC-etQwlYAE - I don't like all ideas in the talk, but it raises some interesting points.

In any case, I don't want to change things without a sensible reason to do so. Without logs of the actual issue, we have no idea what causes this "impossible" hang.

kpanic23, 2017-07-11:
I don't know if that's a sensible enough reason, but autoupdate from 2017.1 to 2017.1.1 yesterday killed about ⅔ of the 400 nodes in our community...
[image]

NeoRaider, 2017-07-11:
With "sensible", I mean we need to have the slightest idea *what* is going wrong, not only that something is going wrong.

A-Kasper, 2017-07-11:
Let's try to collect what we know:

   - the failure occurs after flashing and restoring the configuration
   - the failure occurs with sysupgrade via ssh, so maybe gluon is not the
   first code to look at
   - the failure just occurs in unstable network
   - because there were some more bugs we can't really say what it is
         - High Load
         - High Network Traffic because of Table Requests
         - RAM shortage because of this Bug I can't really explain filling
         (it with logs?)
      - The failure could be avoided by restarting the node before update
   --> it depends on router conditions
   - The failure also appeared at futro offloader with stronger hardware
   then Plastikrouter

How can we hunt:

   - Try to simulate the three named Problems and starting sysupgrade to
   identify the problem
   - enable multicast in a testdomain and make high traffic to get this
   network conditition
   - have a serial attached node while running sysupgrade

What else?

kpanic23, 2017-07-12:
Actually, on most of our nodes the crash must have occured **_before_** flashing!
After the powercycle, the nodes came back with 2017.1, and, a little bit later, updated to 2017.1.1
So, for me, it seems like the autoupdater starts with downloading the firmware, stopping all services, cutting all network connections and, instead of flashing, crashes, because reasons...
I believe, it might be because of #1157 filling up memory because the only nodes surviving the update were those with a quite low uptime.

A-Kasper, 2017-07-12:
Hmmm I can't say that our nodes definitly started updated. it would be
possible that they were updating directly after powercycle and the map just
didn't show the old version. I maybe I had concluded too fast.... please
take it possible that it was here as you describe it... I'm not sure sorry.

mweinelt, 2017-07-12:
We noticed routers needing a hard restart after they tried upgrading themselves. They apparently hang before the update when the network is shutting down.

This started happening around the time the new autoupdater was added, so either in 2016.2.6 or 2017.1, I'm not sure.

```
[694578.387455] sysctl (17375): drop_caches: 3
[694584.992598] device client0 left promiscuous mode
[694584.997857] br-client: port 4(client0) entered disabled state
[694585.282546] batman_adv: bat0: Interface deactivated: mesh0
[694585.455630] batman_adv: bat0: Removing interface: mesh0
[694586.252530] br-client: port 3(local-port) entered disabled state
[694586.258887] br-client: port 2(eth1) entered disabled state
[694586.264600] br-client: port 1(bat0) entered disabled state
[694586.338687] device bat0 left promiscuous mode
[694586.343721] br-client: port 1(bat0) entered disabled state
[694586.387604] device eth1 left promiscuous mode
[694586.392271] br-client: port 2(eth1) entered disabled state
[694586.449501] eth1: link down
[694586.466320] batman_adv: bat0: Interface deactivated: br-wan
[694586.472153] batman_adv: bat0: Removing interface: br-wan
[694587.251952] batman_adv: bat0: Interface deactivated: primary0
[694587.257881] batman_adv: bat0: Removing interface: primary0
[694588.109484] IPv6: ADDRCONF(NETDEV_UP): eth1: link is not ready
[694588.115579] device local-port left promiscuous mode
[694588.120755] br-client: port 3(local-port) entered disabled state
[694588.724214] br-wan: port 1(eth0) entered disabled state
[694588.742443] device eth0 left promiscuous mode
[694588.747019] br-wan: port 1(eth0) entered disabled state
[694588.778638] eth0: link down
[694588.783182] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[694589.472655] sysctl (17700): drop_caches: 3
Watchdog handover: fd=3
- watchdog -
killall: telnetd: no process killed
Sending TERM to remaining processes ... logd netifd sh sysupgrade mac80211.sh iw gluon_bat0.sh ip dnsmasq dhcpv6.script dhcpv6.sh dhcp.sh ubus ubus ubus ubus ubus dnsmasq dnsmasq ubusd 
Sending KILL to remaining processes ... netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip [694595.828094] unregister_netdevice: waiting for mesh0 to become free. Usage count = 1
netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip [694598.168106] unregister_netdevice: waiting for primary0 to become free. Usage count = 1
netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip [694599.058077] unregister_netdevice: waiting for br-wan to become free. Usage count = 1
netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip [694605.968079] unregister_netdevice: waiting for mesh0 to become free. Usage count = 1
netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd i [... cut ...] iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip [695135.966217] unregister_netdevice: waiting for primary0 to become free. Usage count = 1
netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip netifd iw ip [695136.846233] unregister_netdevice: waiting for br-wan to become free. Usage count = 1
[and so on]
```

NeoRaider, 2017-07-12:
While the underlying issue reported by @mweinelt must be fixed in the kernel, it also showed two bugs in upgraded and the sysupgrade scripts. Fixes for these bugs have been added in 74f43ab80f01badc5db2b3143ca294cd51be25d4 and also backported to v2017.1.x. This should ensure that failures during sysupgrade will always trigger a reboot instead of hanging forever.

tobox, 2017-07-13:
Is there a workaround to prevent the hanging when upgrading an affected version? Maybe some command line stuff to force a reboot a few minutes after starting the update.

A-Kasper, 2017-07-13:
There were reports, that manual rebooting and launching the update after
fresh reboot helps to update successful.

rotanid, 2017-07-29:
has anyone successfully tested updates after the fixes?
@NeoRaider will this also be backported to v2016.2.x ? as i understand it, v2016.2.6 uses the same sysupgrade-changes and therefore might also be affected by the bug

mweinelt, 2017-07-30:
The autoupdater changes make our experimental branch much more reliable. No hanging routers anymore.

NeoRaider, 2017-07-30:
The changes are backported to v2016.2.x now.

rubo77, 2017-07-31:
Will you release a 2016.2.3?

Adorfer, 2017-07-31:
> Will you release a 2016.2.3?
https://gluon.readthedocs.io/en/v2017.1.x/releases/v2016.2.3.html

You mean v2016.2.7? As long as you build from a the "2016.2.x" you get this backport.
(in other words: No urgent need, it would be just a repetition of "autoupdater improved" which was in previous RNs.)

rotanid, 2017-07-31:
technically i agree with @Adorfer .
but as we know, unfortunately many people don't use or even test non-released versions...

Adorfer, 2017-07-31:
In order to give some feedback: 4 nodes are just running the first 24h, all updates without problem so far, i will "re-update" to same version tomorrow. 14 other nodes will join in then (i did not backdate the manifest though)

closed 2017-08-07 by NeoRaider