freifunk-gluon/gluon #1165: gluon-status-page doesn't use gluon-web-theme as source

Nurtic-Vibe, 2017-06-27:
gluon-status-page uses its own CSS file without relying on the existing gluon-web-theme,
so changes to e.g. colors have to be done in two packages.

I'd suggest to include the theme in all other web packages.

Example Patch to change colors now:
`diff --git a/package/gluon-status-page/src/css/main.css b/package/gluon-status-page/src/css/main.css
index 0604802..3d129bd 100644
--- a/package/gluon-status-page/src/css/main.css
+++ b/package/gluon-status-page/src/css/main.css
@@ -23,7 +23,7 @@ a:hover {
 header {
   display: flex;
   padding: 0 14px;
-  background: #dc0067;
+  background: #52995d;
   color: rgba(255, 255, 255, 0.98);
   position: absolute;
   top: 0;
diff --git a/package/gluon-web-theme/sass/cascade.scss b/package/gluon-web-theme/sass/cascade.scss
index ab22c06..3cc1a1d 100644
--- a/package/gluon-web-theme/sass/cascade.scss
+++ b/package/gluon-web-theme/sass/cascade.scss
@@ -13,7 +13,7 @@
 @charset "utf-8";

 $ffyellow:  #ffb400;
-$ffmagenta: #dc0067;
+$ffmagenta: #52995d;
 $ffzusatz:  #009ee0;
 $red:       #ee3300;
`

rubo77, 2017-06-27:
can you add a screenshot?

rubo77, 2017-06-28:
Why do you want to change magenta to green?
[image]

The code looks strange:

    $ffmagenta: #52995d;

I think the var name would have to be renamed to `$green` then

NeoRaider, 2017-06-28:
@rubo77, the patch was just an example, not a proposed change.

rotanid, 2018-07-12:
as the status page has been rewritten completely based on gluon-web, i'm closing this issue.

NeoRaider, 2018-07-12:
Both pages still use completely separate CSS files, so this is still open.