freifunk-gluon/gluon #1167: RFC: provide a default value for the 11s mesh ID

NeoRaider, 2017-06-28:
As we now have the site seed to generate site-specific random values, and we already recommend to use a cryptic random string as a mesh ID, we might as well provide a generated default value to reduce the number of required site.conf settings (it will still be possible to override the default to accomodate special usecases and keep compatibility with exisiting deployments). Even though this question will likely incur some bikeshedding, I'd like to get some input on the format of the mesh ID.

Options I have in mind:
* 16 hex digits
* 8..16 alphanumeric characters
* a base64 string
* UUID? (likely too long to be displayed completely on smartphone screens, leading to ugly marquee effects?)

The characters of a hexadecimal or alphanumeric string could also be grouped using dots/dashes/...; such a specific formatting would allow to take note of mesh SSIDs that are likely generated by Gluon without hinting at a relation with Freifunk/... for end users.

bobcanthelpyou, 2017-06-28:
Would that mean a different mesh id with every build? Wouldn't that make mesh-only nodes harder to survive automatic updates? Or could it be a hash or something similar based on the site name or client SSID which is always the same?

Where and why is it recommended to use a cryptic random string? Maybe i missed it in the documentation somewhere?

NeoRaider, 2017-06-28:
@bobcanthelpyou The documentation was recently changed: http://gluon.readthedocs.io/en/latest/user/site.html#site-conf

The site seed is a new site.conf option that is also used to generate the site-specific VXLAN ID (documented on the same page). The site seed must never change for a single mesh, so it can be used to generate a stable random mesh ID.

A-Kasper, 2017-06-29:
> en/latest/user/site.html#site-conf

I don't like random strings. it should be transparent whats behind the ssid
I can't see any benefit from random strings. If you want to get rid of
config values use "$sitestring.mesh" as default for the mesh network.

In general we should discuss how to manage the development goals "easy
usability" and "modular framework" together. I don't really like feeatures
not be implemented because gluon config gets to complicated. maybe there is
a way with an optional expert section in site.conf or something like this...

NeoRaider, 2017-06-29:
@A-Kasper You'll always be able to define a mesh ID in site.conf - it's just that there's no reason to force sites to define one when we recommend a random string anyways.

FreifunkUFO, 2017-07-03:
by default the 11s mesh ID should be the same for ALL freifunk communities. 
its needed to make Layer3-communication possible. (even when many gluons only have L2 atm)

NeoRaider, 2017-07-03:
@FreifunkUFO Mesh domains are separated by mesh IDs, so each community must be assigned a separate ID; this property of Gluon will *not* change when switching to L3. In any case, this has nothing to do with this ticket.

CodeFetch, 2017-11-20:
Random cryptic networks are so ugly (if you see them e.g. on your mobile). At least there should be some prefix option.
I wish http://devel.open80211s.narkive.com/ZAzHDzy6/patch-nl-cfg-mac-80211-implement-hidden-mesh-id was a standard 802.11s feature.

rotanid, 2018-01-19:
"At least there should be some prefix option."
let me quote a comment above:
"You'll always be able to define a mesh ID in site.conf"