freifunk-gluon/gluon #1170: Default settings for local wireless acces

jerix777, 2017-07-02:
Hi ! I've installed gluon on my TP-Link WA801ND V3 through WIFI cause the single ethernet port on the device is bricked in my case.
So now, i can't access the device. But, leds blinks normally (power first, ethernet after... no wifi).
 Is there a way to access the device ?

jerix777, 2017-07-02:
Normally (i think...) wireless must be ON and available on after boot ! non ?

NeoRaider, 2017-07-02:
After flashing initially, a Gluon node will be in "Config Mode", the first-setup wizard. For security reasons, only wired access is available in Config Mode.

If your Ethernet is broken, the only way to access the device is to open it and connect to the serial console.

closed 2017-07-02 by NeoRaider

jerix777, 2017-07-02:
Wow ! serial connector ! ok, i'm going to buy it and hope that it will solve the problem. 
But, i think that wireless must be available on "Config Mode" too. That will make life easier for us.
Thanks for your response **NeoRaider** !

NeoRaider, 2017-07-02:
@jerix777 This has been discussed before, and there is simply no way to provide wireless access in a secure way without previous configuration of credentials. Therefore, we'll always require Ethernet access for the initial configuration.

jerix777, 2017-07-02:
Okay. But I regret my imprudence anyway. Thanks

jerix777, 2017-07-02:
Please may i have the related link of that thread ?

jplitza, 2017-07-10:
@jerix777 See for example #683 and the links in there.

jerix777, 2017-07-10:
@jplitza  Ok. If I was able to turn back hte time...
Anyway, if A=B and B=C then B **could be** equal C.
Thanks a lot ;)

Adorfer, 2017-07-18:
> This has been discussed before, and there is simply no way to provide wireless access in a secure way without previous configuration of credentials. 

There is one, which hardly anybody will like to use: setup a private wifi with random password. 
Put out the password (12 digits. numbers and letters) as morse code, there are LEDs on most devices...<g>