freifunk-gluon/gluon #1176: Feature Request: Watchdog

A-Kasper, 2017-07-07:
There are some "quick-fix" scripts in the wild. They are monitoring some essential services and reboot the nodes if something seems to be wrong. I would like gluon to improve it's self-recovery features to prevent nodes from getting unreachable as much as possible.

FreifunkUFO, 2017-07-09:
watchdog capabilities are existing, via cronjobs, see /usr/lib/micron.d
but, some monitoring scripts has to be optimized (or collected from different communities).

note: there is no need to reboot a router just on every mistake. often there are repair-commands with no need to reboot

jplitza, 2017-07-10:
Correct me if I'm wrong, but I think this is a duplicate of #476, which was rejected (like all similar requests) for masking the real errors and hence preventing noticing and fixing the underlying errors.

closed 2017-07-10 by jplitza

A-Kasper, 2017-07-10:
you are wrong. #476 is a reboot without any checks. This is the feature Request of a watchdog. There are Bugs in some packages and they are real. Yes.. the watchdog is not a replacement for fixing bugs, but you need a stable network. I've several hard reachable nodes. It is necessary that the nodes are recovering in case of failures. this is also necessary to reach them to get their logs to find the issues of the failure. 

I can't agree in discussing self recovery and staying reachable against errorhunting. Both is necessary. A goal that a node have to be dead and someone has to get keys and support to reach it physically is not very useful. For example:  I've some Nodes in Laternen where I need the support of Stadt Bochum and TMR to reach them. This is normally just possible during working hours... where I also work. It's real pain to recover dead nodes in the wild if the get unreachable. after two weeks 30% of all bochum nodes are still missing because nobody had powercycled them yet after the missing reboot after the autoupdater.

Can you please explain what is the benefit of dead unreachable nodes which could be reachable if service crashes would be dedected and the services eould be restarted? I think this would also helps zo get the bugs. look at the autoupdater bug. nobody knows why. it's very hard damage. the affected nodes are unreachable... would they be reachable we maybe could read their logs....

A-Kasper, 2017-07-10:
to make it clearer: logerading reboots are stupid. i can agree with that. but restarting batman, fastd or whole networking is helpful. the quick gix scrpts in the wild are very good and they are used very wide. By using communities are calling for such capabilities....

FreifunkUFO, 2017-07-12:
@A-Kasper 
so please link some "quick gix scrpts in the wild". which communities are have some?

so, what are you missing? the cronjob (for doing tests) is still existing..

or do you want some official gluon packages to make proper tests (instead of quick-hack script, where each community tries to find a crazier way to implement some checks)? that would be nice

jplitza, 2017-07-13:
You specifically mentioned rebooting the node, thus my assumption that this was a duplicate.

Still, this is the kind of error-shadowing behavior that is generally not accepted in Gluon. If reaching and debugging a node is that unacceptable in some cases, I think it is reasonable to expect a non-standard firmware with a regular reboot or watchdog pinging some gateway to be used in these cases. However, an official package will be included in almost all default firmwares, because debugging is hard and nobody wants to do it, and thus over time degrading the quality of Gluon.