freifunk-gluon/gluon #1177: Feature Request: Show Full Node-Path to Gateway for Mesh-Only

A-Kasper, 2017-07-07:
We've got a setup in a Kleingartenanlage, where 4 nodes with internetaccess provide the uplink for about 15 mesh-only nodes in the terrain. It would be nice, if the Statuspage would be able to show the full link path with nodenames until the batman gw is reached. This could be provided by batctl tr to chosen gateway...

If this is not possible via Statuspage (Name Resolution of the Nodes) maybe the path could be provided via respondd?!

rubo77, 2017-07-07:
hopglass already shows this path. example: https://mesh.freifunk.in-kiel.de/#!v:m;n:c46e1fe8f164

A-Kasper, 2017-07-07:
That is exactly what I'm looking for. We are also using Hopglass, but it doesn't show the path. Is there any hint? 

Also the feature could be provided via hopglass i would like to habe it on the statuspage if possible

NeoRaider, 2017-07-07:
The status page is only supposed to show local data. An external mapping solution like hopglass is much better fit to display global information like paths.

`batctl tr` can only show MAC addresses of mesh interfaces on the way, there is no easy way to map these to names or primary MAC addresses without all the information the map has.

closed 2017-07-07 by NeoRaider

rotanid, 2017-07-08:
@A-Kasper make sure your hopglass-server and hopglass are up-to-date.
the path-feature was introduced in late 2016 as far as i remember

A-Kasper, 2017-07-08:
my build is latest master.... maybe it doesn't work with gluon 2017.1.1