freifunk-gluon/gluon #1180: ipv6-address from br-wan as its source address is used when source-specific routing is active

christf, 2017-07-07:
When there is ipv6 on the wan-port, it was observed that queries from the local dnsmasq installation that is used as dns-cache will be sent out on the wan-interface.

The main routing table does not contain a default route, table 1 contains the default-ipv6-route for the wan traffic. There is an ip rule that causes packets having the src-ip from br-wan to be routed according to table 1.
For  dnsmasq queries, this very IPv6-Address is used as source-address, leaking dns queries into wan that should be routed via the mesh.

There is a tested workaround for this that causes dnsmasq to set its source-ipv6-address: https://github.com/freifunk-gluon/gluon/pull/934/commits/1c6117f788e6d98c9298901b5efa698a0bb5e969

unfortunately this only works for dnsmasq. A solution that works for other programs as well is preferred.
We discussed on irc that network namespaces would resolve the issue. But having a dependency of the babel network on network namespaces is not desirable.
Turning off source-specific routing may also be able to mitigate the effects, however that would prevent prefixd to operate. 

On batman-based networks this does not seem to be an issue due to source-specific routing being turned off.

christf, 2018-08-18:
This is resolved by the PREFSRC patchset for babeld. 
Also the workaround here broke in the last set of updates for openwrt.

Still I am closing this one as babeld now supports a proper solution.

closed 2018-08-18 by christf

rotanid, 2018-08-19:
in current gluon master branch it is not fixed, is it?