freifunk-gluon/gluon #1183: batman-adv 2017.1 multicast optimization DDOS

A-Kasper, 2017-07-09:
batman-adv has a known bug in multicast optimization. one malformed package is able to DDOS the whole mesh.

As quickfix gluon2017.1.1 is compiled with multicast optimization deactivated. This helps a lot because the number of wrongly distributed packages is lower, but if just one node with mo activated is in the network the failure could still ddos the whole network.

This Ticket is to track this issue.

batman-adv developers have an untested fix ready:
https://www.open-mesh.org/issues/336#note-2

belzebub40k, 2017-07-10:
Do I get this right, the problem also happens with MO disabled but doesn't cause a DDoS on the mesh?
Are nodes with batman-adv 2016.x also effected by this problem (except that they do not cause a DDoS)?
Is there a know way to identify the faulty nodes which send the malformed packages?

A-Kasper, 2017-07-10:
I don't understand your question. Have your read the description in the patch?:

> This patch fixes an issue in the translation table code potentially
leading to a TT Request + Response storm. The issue may occur for nodes
involving BLA and an inconsistent configuration of the batman-adv AP
isolation feature. However, since the new multicast optimizations, a
single, malformed packet may lead to a mesh-wide, persistent
Denial-of-Service, too.
>
>The issue occurs because nodes are currently OR-ing the TT sync flags of
all originators announcing a specific MAC address via the
translation table. When an intermediate node now receives a TT Request
and wants to answer this on behave of the destination node then this
intermediate node now responds with an altered flag field and broken
CRC. The next OGM of the real destination will lead to a CRC mismatch
and triggering a TT Request and Response again.
>
>Furthermore, the OR-ing is currently never undone as long as at least
one originator announcing the according MAC address remains, leading to
the potential persistency of this issue.
>
>This patch fixes this issue by storing the flags used in the CRC
calculation on a a per TT orig entry basis to be able to respond with
the correct, original flags in an intermediate TT Response for one
thing. And to be able to correctly unset sync flags once all nodes
announcing a sync flag vanish for another.

The MO enabled leads to CRC errors, so the nodes request a full TT. This means a lot of load and traffic. A Node with MO disabled doesn't confuse the flags, but if it get's a "broken" package it also sends the tablerequest and is part of the CRC Hazzard.

belzebub40k, 2017-07-10:
@A-Kasper Your answered my questions with "A Node with MO disabled doesn't confuse the flags". I just wanted to know if these malformed packages may cause other less critical problems like increased CPU load on nodes without MO. And if so how to isolate them.

A-Kasper, 2017-07-10:
yes. if you've got a package with confused flags also nodes without MO will
detect the CRC Error and react with a full table request. this leads to
high net and cpu load. the error propagates through the network.

I don't know a way to isolate such nodes. I hope the linked patch will fix
it soon.

A-Kasper, 2017-07-10:
to be clear: in bochum the network went nearly unusaable with rates under
500 kb/s und load between 2.0 and 5.2
since neoraider implemented the workaround (disabled MO) the network went
stable again. we had hard issues with autoupdate....

i tried to activate MO on supernodes only. the network started to crash
immidiatly. reboots, load, tq breakdown etc.... also with the deactivated
MO in the Plastikrouter. after deactivating it again -> stable.

A-Kasper, 2017-07-10:
> @A-Kasper <https://github.com/a-kasper> Interesting, in our network (MWU)
> only nodes with Gluon 2017.1 were unusable because of the high CPU load but
> we didn't had any problems as you described it. But maybe it is because we
> only had about 10 experimental nodes with 2017.1. Oddly it disappeared
> without updating all of them to a version without MO so it looks like as
> the faulty node stopped broadcasting.

We had just 15 testnodes before rolling out. you could see a that the mngmt
traffic is rising but if if you haven't many nodes with high traffic the
count of malformed packages seemed to be handlebel for the nodes. the
problems came with wide deployment or high traffic nodes like supernodes.
this seems to be logical to me, because more crc errors could appear if you
have more malformed packages.

the problem grows with count and traffic of nodes which confuse the flags.
that was our problem. we thought it was stable and seemed to be the girdt
community which deployed in a bigger network

kpanic23, 2017-07-10:
That explains what happened here... We also had 2017.1 running on a few nodes running on experimental branch. After everything seemed running flawlessly, we rolled it out as stable on our biggest sub-community (~400 nodes). Then hell broke loose. It was close to unusable.
A few hours ago I have signed 2017.1.1 as stable. Now, sadly, 2/3 of the nodes are offline... But I seem to be getting usual speeds again.

A-Kasper, 2017-07-10:
there are two types of offline nodes:
1. nodes which are in such highcpu network Situation that they can't handle
their tasks. Many of them will selfrecover...
2. there is a bug in autoupdater which doesn't reboot the node after
flashing and configuring the nodeupdate. after the powercycle the node
works like charm but about 30-50%  didn't come back wirhout reboot. because
logs are lost while upgrading this bug is not identified yet. if you could
provide logs via an affected node via serial: there is a ticket. we need
more information what happens here

T-X, 2017-07-10:
@belzebub40k 
> Is there a know way to identify the faulty nodes which send the malformed packages?

You can spot whether your network is affected by this bug by looking at the output of "batctl tg": In Gluon, we do not use the batman-adv client isolation feature, therefore any node announcing a TT entry with a wireless (W) or isolation (I) flag likely sends malformed packets.

For unicast MAC entries, you should see the according node immediately. For multicast MACs, it is a little more tricky as other nodes announcing the same multicast MAC will inherit the flags (like for 33:33:00:00:00:01). However you could try to identify nodes sending malformed packets by looking at multicast MACs with a W or I flag which are only announced by a single node. That node is doing something seriously wrong then, as there is no code in batman-adv to add any flags to multicast TT announcements.

So far I had heard from one report from Freifunk Voigtland, where the W/I flags were introduced to multicast TT entries by a broken virtual machine. It was sending all kinds of packets with bits flipped and other erratic behaviour, afaik. I'd be very interested to hear what kind of devices are sending malformed packets in other communities.

T-X, 2017-07-20:
Antonio, the TT developer/maintainer has had a look at the patch and it seems to make sense to him. However, he expressed that he'd be grateful if there'd be a little more testing in real scenarios, as translation table is a quite essential part.

Is there a community with a certain amount of nodes on an autoupdater configured to an "experimental" branch and who'd like to give the patch, which @A-Kasper mentioned already, a try? (https://www.open-mesh.org/issues/336#note-2)

First people testing it thoroughly and successfully may get some fame in the eternal Linux git commit history with a "Tested-by: ..." tag :-).

NeoRaider, 2017-07-20:
@T-X How about providing a Gluon branch containing the patch to facilitate testing?

A-Kasper, 2017-07-21:
the failure deploys if the bug is enabled in high traffic nodes. if someone
tells me how to get a version with this fix, i would try it on increasing
supernodes. this is much easier to fix is something goes wrong

rotanid, 2017-07-29:
there are gluon PRs with a possible fix by @T-X , please test them!
master: #1199 
v2017.1.x: #1200

commit c66b18ca (2017-08-07): modules: update openwrt-routing packages

Updates batman-adv and alfred to 2017.2 and adds the batman-adv TT-sync
fix.

Fixes #1183
commit b0a6cb72 (2017-08-07): modules: update openwrt-routing packages

Updates batman-adv and alfred to 2017.2 and adds the batman-adv TT-sync
fix.

Fixes #1183
closed 2017-08-07 by NeoRaider