freifunk-gluon/gluon #1184: How to build a local mesh base on Gluon without VPN access ?

jerix777, 2017-07-10:
Hi !
I have five AP with freifunk-gluon installed. I just want to mesh-link them on local network.
-AP1 and AP2 physically attached to le ADSL modem on their WAN port;
-Others APs will mesh access to AP1 and AP2.

I don't want to make VPN access to freifunk servers. Just use mesh fonctionalities on local network...
Is this possible ? If yes, how ?

Thanks.

belzebub40k, 2017-07-10:
If you what to use mesh for you local network Gluon might not be the right solution for you as the links between nodes are not encrypted? Or is the local network network a public network without access restriction?

If you don't care about encrypted WiFi then you can simply build Gluon without any gluon-mesh-vpn-* packages and only use Mesh-on-WAN/LAN/WLAN. But as Gluon does have a DHCP server and limited support for RAs these things need to be covered by your normal Router.

If you need encrypted mesh-links is is probably the best solution to set up your own mesh with plain LEDE.

NeoRaider, 2017-07-10:
The basic idea is: all LAN ports of all Gluon nodes of the same mesh topologically form one big switch. Creating local mesh using a minimal site.conf is possible (and actually something I use occasionally); you'll need to:
* Use batman-adv-15 (as it has Bridge Loop Avoidance (BLA) enabled by default, which will be necessary if you want multiple "uplink" nodes; personally, I'd just use a single uplink node instead of relying on BLA
* Connect the uplink Gluon nodes' to your uplink router using their LAN ports (the WAN port will be unused, as you don't have VPN)
* Set the gateway mode in /etc/config/batman-adv to 'server' for uplink nodes

As @belzebub40k noted, you'll need a DHCP server outside the Gluon mesh (for example a simple home router between the DSL modem and the Gluon nodes), and Gluon does not support encrypted meshes.

closed 2017-07-10 by NeoRaider

jerix777, 2017-07-10:
> If you what to use mesh for you local network Gluon might not be the right solution for you as the links between nodes are not encrypted. Or is the local network network a public network without access restriction?

Yes it's a public network without restiction.

> If you don't care about encrypted WiFi then you can simply build Gluon without any gluon-mesh-vpn-* packages and only use Mesh-on-WAN/LAN/WLAN. But as Gluon does have a DHCP server and limited support for RAs these things need to be covered by your normal Router.

Ok. I'm going to make the build.

jerix777, 2017-07-10:
Thank NeoRaider,
>Connect the uplink Gluon nodes' to your uplink router using their LAN ports (the WAN port will be unused, as you don't have VPN)
This APs don't have LAN ports ! Theyy have a single WAN port.

NeoRaider, 2017-07-10:
@jerix777 You can use the (undocumented) option
```lua
single_as_lan = true
```
on the top level of site.conf to make the single port a LAN port.

jerix777, 2017-07-10:
As I heard @NeoRaider , this is the architecture of the site you said:
ADSL-MODEM<-->Firewall+DHCP<---->AP1<--->Mesh Network
- For AP1 : make modification to /etc/config/batman-adv and set mode to server;
so, is it the single modification I should make on AP1 in config-mode to work ?

- What about the others AP on the mesh network ?

jerix777, 2017-07-10:
@NeoRaider 
What is the path of site.conf file ?

NeoRaider, 2017-07-10:
@jerix777 There are example files in the `docs` directory; you should probably read http://gluon.readthedocs.io/en/v2017.1.x/user/getting_started.html if you want to create a new site definition.

You can leave out a lot of the packages in the example config for such a local mesh setup (all the gluon-setup-mode, gluon-config-mode and gluon-web stuff), instead gluon-lock-password and gluon-authorized-keys might be useful.

You might want to take this discussion to the Gluon mailing list, as this is an issue tracker.

jerix777, 2017-07-10:
@NeoRaider 
Ok, I'm going on to have a long working day. But, please how to contact you for some guidances ?
Thank a lot for your help !

jerix777, 2017-07-11:
@NeoRaider 
Hi, I was build last night multiple version of meshing my APs on same network. But infortunatly doesn't work :( 
Could you please give me your site.conf and site.mk files ?