freifunk-gluon/gluon #1187: fastd -> tunneldigger upgrade-problem

Sunz3r, 2017-07-11:
Hi :)

If an old version of tunneldigger was installed and fastd enabled the upgrade wont enable tunneldigger with removed fastd.

The Problem is that "400-mesh-vpn" comes before "400-mesh-vpn-tunneldigger" in upgrade-process.
I solve this by rename "400-mesh-vpn-tunneldigger" to "399-mesh-vpn-tunneldigger".

---

https://www.freifunk-gera-greiz.de/forum/-/message_boards/message/115277#_19_message_115050
https://www.freifunk-gera-greiz.de/forum/-/message_boards/message/115277#_19_message_115210

NeoRaider, 2017-07-11:
Hmm yes, the scripts are designed to work for migrations of
* Gluon tunneldigger -> Gluon fastd
* Gluon fastd -> Gluon tunneldigger
* Inofficial tunneldigger -> Gluon tunneldigger

I assume you're talking about an upgrade of the third case, where the old inofficial tunneldigger package is upgraded to the Gluon package, but there are also remains of an (even earlier) fastd configuration? Could you send examples for the states of /etc/config/fastd and /etc/config/tunnerldigger *before* the upgrade?

Sunz3r, 2017-07-11:
Yes, the tunneldigger was pre-gluon.

This only happend when pre-gluon-config exists and fastd enabled

https://pastebin.com/nVkaJhDh

lcb01a, 2017-07-14:
I noticed this bug as well but was unsure if it had anything to do with the current way of handling the VPN migration.
I checked the scripts and they seem to get executed in the wrong order, the migration should be done after all initial configs have been written. 

400-mesh-vpn runs and creates the interface configuration for gluon, since the migration code is part of this script it will try to read an non-existing config in the case you migrate from fastd.
This leads to the issue descriped as the tunneldigger config isn't in place yet so it will try to enable a uci section which doesn't exist.
This is actually not connected to the package used in the past as they shared the same configuration structure. It only applies if a community switches to tunneldigger while also updating to gluon 2017.X so communities who did this in the past are on the safe side.

I will prepare a fix for this and attach a pull request.

commit 95243988 (2017-08-07): gluon-mesh-vpn: fix fastd <-> tunneldigger migration

The generic upgrade script is moved to run after the more specific scripts.
In addition, the script will now remove the configuration sections of
uninstalled VPN packages, so both positive and negative changes of the
default enable state can be migrated correctly.

Based-on-patch-by: Cyrus Fox <cyrus@lambdacore.de>
Fixes: #1187
closed 2017-08-07 by NeoRaider