freifunk-gluon/gluon #1188: tunneldigger dies somtimes

Sunz3r, 2017-07-11:
If internet lost connection (like reboot of router or plug off lan-cable) then the tunneldigger sometime dies.

After strace the tunneldigger i found that a assert is triggered: https://github.com/wlanslovenija/tunneldigger/blob/master/client/asyncns.c#L884

https://pastebin.com/R6CQmGXr

NeoRaider, 2017-07-11:
The right place to report this is the Tunneldigger bug tracker at https://dev.wlan-si.net/report/15. I'll still keep this ticket open to track it for Gluon.

mitar, 2017-07-13:
cc @kostko

Sunz3r, 2017-07-19:
I was not able to report this bug on there site ...

this bug happens when dns-resolution failed and despite sending data through a socket:
`td-client: Failed to send() control packet (errno=89, type=1)!`
when dns-resolution suddenly works in this situation then the tunneldigger will crash.

I think there is a bug in the broker selection. The selector should wait for successful address-resolution.

To workaround this bug you can use ip's instead of names.

Adorfer, 2017-07-19:
> To workaround this bug you can use ip's instead of names.
thans for the hint. 
perhaps add a hint in the site.conf example.doc in case of tunneldigger/l2tp use.

mitar, 2017-07-19:
Interesting. We always use IPs, so this is why we didn't notice this before.

charlemagnelasse, 2017-11-20:
@mitar: How can i report bugs at https://dev.wlan-si.net/newticket?component=tunneldigger ? The registration link automatically moves me to the wlan-slovenia registration but the created account doesn't work on dev.wlan-si.net. And without an account, I will only see the message "TICKET_CREATE privileges are required to perform this operation"

Btw. the bug I see is with newer kernels (

Only one session (tunnel interface) can be created. All other sessions create a "Failed to setup tunnel with id 102!".

The test kernel is linux-image-4.9.0-4-amd64. And I saw some changes which might be related:

    - l2tp: fix race in l2tp_recv_common()
    - l2tp: ensure session can't get removed during pppol2tp_session_ioctl()
    - l2tp: fix duplicate session creation
    - l2tp: hold session while sending creation notifications
    - l2tp: take a reference on sessions used in genetlink handlers

kostko, 2017-11-20:
Report them in the [GitHub repository](https://github.com/wlanslovenija/tunneldigger).

kostko, 2017-11-20:
BTW, there is already a fix for the session ID issue you mention. It is in https://github.com/wlanslovenija/tunneldigger/pull/64 (not yet merged into master).

rotanid, 2018-01-10:
we will close this issue here, as tunneldigger has recently been updated to the latest version.
you may continue to use this for discussion, but it should rather take place upstream.

closed 2018-01-10 by rotanid