freifunk-gluon/gluon #1190: Error with building minimal Local-mesh

jerix777, 2017-07-14:
I am trying to build the target=ar71xx-generic for TP-Link WA801ND with this command:

`"make GLUON_TARGET=ar71xx-generic V=s"`

But it issues the error below on trying to build **local-Mesh** with minimal site.conf. This is the error:
> ...
...
make[3] -C tools/libressl compile
make -r tools/install: build failed. Please re-run make with -j1 V=s to see what's going on
/home/jerix/gluon/lede/include/toplevel.mk:198: recipe for target 'tools/install' failed
make[1]: *** [tools/install] Error 1
make[1]: Leaving directory '/home/jerix/gluon/lede'
Makefile:113 : recipe for target  « lede/staging_dir/hostpkg/bin/lua » failed
make: *** [lede/staging_dir/hostpkg/bin/lua] Error 2

Here is my site.conf:
```
{
  single_as_lan= true 
  site_name = 'Local Hotspot' 
  site_code = 'hnet',
  prefix4 = '192.168.2.0/24',
  prefix6 = 'fdca::ffee:babe:1::/64'
  timezone = 'GMT0',
  ntp_servers = {'ntp1.local.lan'},
  regdom = 'FR',
  wifi24 = {
    channel = 1,
    supported_rates = {6000, 9000, 12000, 18000, 24000, 36000, 48000, 54000},
    basic_rate = {6000, 9000, 18000, 36000, 54000},

    ap = {
      ssid = 'Local Hotspot',
      -- disabled = true, (optional)
    },

    mesh = {
      id = 'hnet-link',
      mcast_rate = 12000,
      -- disabled = true, (optional)
    },
  },

   mesh = {
     batman_adv = { 
       gw_sel_class = 1,
     },
   },

   roles = {
     default = 'node',
     list = {
       'node',
       'test',
       'backbone',
       'service',
     },
   },

   setup_mode = {
    skip = true,
   },

   config_mode = { 
     geo_location = {
       show_altitude = true,
     },
     owner = {
       obligatory = true
     },
   },
}
```
And my site.mk:
```
GLUON_SITE_PACKAGES := \
	gluon-alfred \
	gluon-respondd \
	gluon-mesh-batman-adv-15 \
	gluon-lock-password \
	gluon-authorized-keys \
	gluon-status-page \
	haveged \
	iwinfo

DEFAULT_GLUON_RELEASE := 0.6+exp$(shell date '+%d%m%Y')
GLUON_RELEASE ?= $(DEFAULT_GLUON_RELEASE)
GLUON_PRIORITY ?= 0
GLUON_REGION ?= us
GLUON_LANGS ?= fr en
```

jerix777, 2017-07-14:
Oops, i forgot to give the version. I'm on v2017.1.1

rubo77, 2017-07-14:
and did you re-run make with -j1 V=s to see what's going on?

you can clone our site.conf from kiel, there we have a branch lede: https://github.com/freifunk-kiel/site-ffki/tree/lede and adapt it to your needs.

But This issuetracker is not the right place to ask for support, please com e to IRC #freifunk to get some more help: https://wiki.freifunk.net/Freifunk-Chat

jerix777, 2017-07-15:
@rubo77  , Thanks a lot  !

NeoRaider, 2017-07-15:
Hmm, the error message doesn't sound like a issue in site.conf/site.mk, please provide a full build log.

jerix777, 2017-07-15:
Ok this is the error log with -j1 V=s option

> make[4]: Leaving directory '/home/jerix/gluon/lede/build_dir/host/mtools-4.0.18'
> touch /home/jerix/gluon/lede/build_dir/host/mtools-4.0.18/.built
> make[3]: Leaving directory '/home/jerix/gluon/lede/tools/mtools'
> make[3]: Entering directory '/home/jerix/gluon/lede/tools/mtools'
> install -m0755 /home/jerix/gluon/lede/build_dir/host/mtools-4.0.18/mcopy /home/jerix/gluon/lede/staging_dir/host/bin/
> install -m0755 /home/jerix/gluon/lede/build_dir/host/mtools-4.0.18/mmd /home/jerix/gluon/lede/staging_dir/host/bin/
> mkdir -p /home/jerix/gluon/lede/staging_dir/host/stamp
> touch /home/jerix/gluon/lede/build_dir/host/mtools-4.0.18/.built
> touch /home/jerix/gluon/lede/staging_dir/host/stamp/.mtools_installed
> make[3]: Leaving directory '/home/jerix/gluon/lede/tools/mtools'
> make[3]: Entering directory '/home/jerix/gluon/lede/tools/dosfstools'
> mkdir -p /home/jerix/gluon/lede/dl
> SHELL= flock /home/jerix/gluon/lede/tmp/.dosfstools-4.0.tar.xz.flock -c '       /home/jerix/gluon/lede/scripts/download.pl "/home/jerix/gluon/lede/dl" "dosfstools-4.0.tar.xz" "9037738953559d1efe04fc5408b6846216cc0138f7f9d32de80b6ec3c35e7daf" "" "https://github.com/dosfstools/dosfstools/releases/download/v4.0/" "http://fossies.org/linux/misc"    '
> Can't exec "curl": No such file or directory at /home/jerix/gluon/lede/scripts/download.pl line 77.
> --2017-07-15 01:51:14--  https://github.com/dosfstools/dosfstools/releases/download/v4.0/dosfstools-4.0.tar.xz
> Resolving github.com (github.com)... 192.30.253.113, 192.30.253.112
> Connecting to github.com (github.com)|192.30.253.113|:443... connected.
> HTTP request sent, awaiting response... 302 Found
> Location: https://github-production-release-asset-2e65be.s3.amazonaws.com/26462150/9b0ee7a0-133c-11e6-8bfd-194226d6e058?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20170715%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20170715T015115Z&X-Amz-Expires=300&X-Amz-Signature=dd3838135c28e03b119b9c6ab10477e1bce6e441d85edab2d34b046927c0bac9&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3Ddosfstools-4.0.tar.xz&response-content-type=application%2Foctet-stream [following]
> --2017-07-15 01:51:15--  https://github-production-release-asset-2e65be.s3.amazonaws.com/26462150/9b0ee7a0-133c-11e6-8bfd-194226d6e058?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20170715%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20170715T015115Z&X-Amz-Expires=300&X-Amz-Signature=dd3838135c28e03b119b9c6ab10477e1bce6e441d85edab2d34b046927c0bac9&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3Ddosfstools-4.0.tar.xz&response-content-type=application%2Foctet-stream
> Resolving github-production-release-asset-2e65be.s3.amazonaws.com (github-production-release-asset-2e65be.s3.amazonaws.com)... 52.216.18.248
> Connecting to github-production-release-asset-2e65be.s3.amazonaws.com (github-production-release-asset-2e65be.s3.amazonaws.com)|52.216.18.248|:443... connected.
> HTTP request sent, awaiting response... 200 OK
> Length: 157564 (154K) [application/octet-stream]
> Saving to: 'STDOUT'
> 
> -                                         100%[=====================================================================================>] 153.87K   139KB/s    in 1.1s
> 
> 2017-07-15 01:51:18 (139 KB/s) - written to stdout [157564/157564]
> 
> . /home/jerix/gluon/lede/include/shell.sh; xzcat /home/jerix/gluon/lede/dl/dosfstools-4.0.tar.xz | tar -C /home/jerix/gluon/lede/build_dir/host/dosfstools-4.0/.. -xf -
> [ ! -d ./src/ ] || cp -fpR ./src/* /home/jerix/gluon/lede/build_dir/host/dosfstools-4.0
> 
> Applying ./patches/0001-mkfs-Default-to-64-32-heads-sectors-for-targets-smal.patch using plaintext:
> patching file src/mkfs.fat.c
> 
> Applying ./patches/0002-Switch-to-AC_CHECK_LIB-for-iconv-library-linking.patch using plaintext:
> patching file configure.ac
> touch /home/jerix/gluon/lede/build_dir/host/dosfstools-4.0/.prepared2dc40ee6c54176d6456380cb2b591d80
> (cd /home/jerix/gluon/lede/build_dir/host/dosfstools-4.0; rm -f aclocal.m4; if [ -f ./configure.ac ] || [ -f ./configure.in ]; then [ -d ./autom4te.cache ] && rm -rf autom4te.cache; [ -e ./config.rpath ] || ln -s /home/jerix/gluon/lede/scripts/config.rpath ./config.rpath; touch NEWS AUTHORS COPYING ABOUT-NLS ChangeLog; AUTOM4TE=/home/jerix/gluon/lede/staging_dir/host/bin/autom4te AUTOCONF=/home/jerix/gluon/lede/staging_dir/host/bin/autoconf AUTOMAKE=/home/jerix/gluon/lede/staging_dir/host/bin/automake ACLOCAL=/home/jerix/gluon/lede/staging_dir/host/bin/aclocal AUTOHEADER=/home/jerix/gluon/lede/staging_dir/host/bin/autoheader LIBTOOLIZE=/home/jerix/gluon/lede/staging_dir/host/bin/libtoolize LIBTOOL=/home/jerix/gluon/lede/staging_dir/host/bin/libtool M4=/home/jerix/gluon/lede/staging_dir/host/bin/m4 AUTOPOINT=true /home/jerix/gluon/lede/staging_dir/host/bin/autoreconf -v -f -i -s -B /home/jerix/gluon/lede/staging_dir/host/share/aclocal -I m4 -I . . || true; fi; );
> autoreconf: Entering directory `.'
> autoreconf: configure.ac: not using Gettext
> autoreconf: running: /home/jerix/gluon/lede/staging_dir/host/bin/aclocal -I m4 -I . --force
> autoreconf: configure.ac: tracing
> autoreconf: configure.ac: not using Libtool
[... cut ...]
> make[2]: Leaving directory '/home/jerix/gluon/lede'
> make[1]: Leaving directory '/home/jerix/gluon/lede'
> lede/staging_dir/hostpkg/bin/lua: scripts/site_config.lua:9: [string "site.conf"]:4: '}' expected (to close '{' at line 2) near 'site_name'
> stack traceback:
>         [C]: in function 'assert'
>         scripts/site_config.lua:9: in main chunk
>         [C]: ?
> Your site configuration did not pass validation.
> Makefile:122 : recipe for target « all » failed
> make: *** [all] Error 1
> jerix@ordix:~/gluon$

MTRNord, 2017-07-15:
in the first, second and fith line are missing trailing commas (site.conf). Thats why it can't compile

jerix777, 2017-07-15:
@MTRNord Ok thank. I'm going to try that.

Adorfer, 2017-07-15:
> in the first, second and fith line are missing trailing commas (site.conf). 

hjson would help a lot many people.
But that's just my point of view.

rubo77, 2017-07-16:
You can also add the .travis file and the `tests` folder of our ffki-site folder, then you would see the validation of the site.conf right here in github

jerix777, 2017-07-16:
@MTRNord , You were right. So, the compilation went on all night until this morning. But, it just issues another error message as follows:
```
libtool: warning: relinking 'libnl-cli-3.la'
libtool: install: (cd /home/jerix/gluon/lede/build_dir/target-mips_24kc_musl-1.1.16/libnl-3.2.29/src/lib; /bin/bash "/home/jerix/gluon/lede/build_dir/target-mips_24kc_musl-1.1.16/libnl-3.2.29/libtool"  --tag CC --mode=relink mips-openwrt-linux-musl-gcc -Wall -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremap /home/jerix/gluon/lede/build_dir/target-mips_24kc_musl-1.1.16/libnl-3.2.29:libnl-3.2.29 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -ffunction-sections -fpic -version-info 224:0:24 -Wl,--version-script=../../libnl-cli-3.sym -L/home/jerix/gluon/lede/staging_dir/target-mips_24kc_musl-1.1.16/usr/lib -L/home/jerix/gluon/lede/staging_dir/target-mips_24kc_musl-1.1.16/lib -L/home/jerix/gluon/lede/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl-1.1.16/usr/lib -L/home/jerix/gluon/lede/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl-1.1.16/lib -znow -zrelro -o libnl-cli-3.la -rpath /usr/lib utils.lo addr.lo ct.lo link.lo neigh.lo rule.lo route.lo tc.lo qdisc.lo class.lo cls.lo exp.lo ../../lib/libnl-3.la ../../lib/libnl-route-3.la ../../lib/libnl-nf-3.la ../../lib/libnl-genl-3.la -L../../lib -ldl -lpthread -lm -inst-prefix-dir /home/jerix/gluon/lede/build_dir/target-mips_24kc_musl-1.1.16/libnl-3.2.29/ipkg-install)
libtool: relink: mips-openwrt-linux-musl-gcc -shared  -fPIC -DPIC  .libs/utils.o .libs/addr.o .libs/ct.o .libs/link.o .libs/neigh.o .libs/rule.o .libs/route.o .libs/tc.o .libs/qdisc.o .libs/class.o .libs/cls.o .libs/exp.o   -L/home/jerix/gluon/lede/build_dir/target-mips_24kc_musl-1.1.16/libnl-3.2.29/lib/.libs -L/home/jerix/gluon/lede/staging_dir/target-mips_24kc_musl-1.1.16/usr/lib -L/home/jerix/gluon/lede/staging_dir/target-mips_24kc_musl-1.1.16/lib -L/home/jerix/gluon/lede/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl-1.1.16/usr/lib -L/home/jerix/gluon/lede/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl-1.1.16/lib -L/home/jerix/gluon/lede/build_dir/target-mips_24kc_musl-1.1.16/libnl-3.2.29/ipkg-install/usr/lib -L/usr/lib -lnl-nf-3 -lnl-route-3 -lnl-genl-3 -lnl-3 -L../../lib -ldl -lpthread -lm  -Os -mno-branch-likely -mips32r2 -mtune=24kc -msoft-float -mips16 -minterlink-mips16 -fstack-protector -Wl,-z -Wl,now -Wl,-z -Wl,relro -Wl,--version-script=../../libnl-cli-3.sym   -Wl,-soname -Wl,libnl-cli-3.so.200 -o .libs/libnl-cli-3.so.200.24.0
libtool: install: /usr/bin/install -c .libs/libnl-cli-3.so.200.24.0T /home/jerix/gluon/lede/build_dir/target-mips_24kc_musl-1.1.16/libnl-3.2.29/ipkg-install/usr/lib/libnl-cli-3.so.200.24.0
libtool: install: (cd /home/jerix/gluon/lede/build_dir/target-mips_24kc_musl-1.1.16/libnl-3.2.29/ipkg-install/usr/lib && { ln -s -f libnl-cli-3.so.200.24.0 libnl-cli-3.so.200 || { rm -f libnl-cli-3.so.200 && ln -s libnl-cli-3.so.200.24.0 libnl-cli-3.so.200; }; })
libtool: install: (cd /home/jerix/gluon/lede/build_dir/target-mips_24kc_musl-1.1.16/libnl-3.2.29/ipkg-install/usr/lib && { ln -s -f libnl-cli-3.so.200.24.0 libnl-cli-3.so || { rm -f libnl-cli-3.so && ln -s libnl-cli-3.so.200.24.0 libnl-cli-3.so; }; })
libtool: install: /usr/bin/install -c .libs/libnl-cli-3.lai /home/jerix/gluon/lede/build_dir/target-mips_24kc_musl-1.1.16/libnl-3.2.29/ipkg-install/usr/lib/libnl-cli-3.la
libtool: install: /usr/bin/install -c .libs/libnl-cli-3.a /home/jerix/gluon/lede/build_dir/target-mips_24kc_musl-1.1.16/libnl-3.2.29/ipkg-install/usr/lib/libnl-cli-3.a
libtool: install: chmod 644 /home/jerix/gluon/lede/build_dir/target-mips_24kc_musl-1.1.16/libnl-3.2.29/ipkg-install/usr/lib/libnl-cli-3.a
libtool: install: mips-openwrt-linux-musl-gcc-ranlib /home/jerix/gluon/lede/build_dir/target-mips_24kc_musl-1.1.16/libnl-3.2.29/ipkg-install/usr/lib/libnl-cli-3.a
libtool: warning: remember to run 'libtool --finish /usr/lib'
make[8]: Nothing to be done for 'install-data-am'.
make[8]: Leaving directory '/home/jerix/gluon/lede/build_dir/target-mips_24kc_musl-1.1.16/libnl-3.2.29/src/lib'
make[7]: Leaving directory '/home/jerix/gluon/lede/build_dir/target-mips_24kc_musl-1.1.16/libnl-3.2.29/src/lib'
make[7]: Entering directory '/home/jerix/gluon/lede/build_dir/target-mips_24kc_musl-1.1.16/libnl-3.2.29/src'
make[8]: Entering directory '/home/jerix/gluon/lede/build_dir/target-mips_24kc_musl-1.1.16/libnl-3.2.29/src'
 /bin/mkdir -p '/home/jerix/gluon/lede/build_dir/target-mips_24kc_musl-1.1.16/libnl-3.2.29/ipkg-install/usr/bin'
  /bin/bash ../libtool   --mode=install /usr/bin/install -c genl-ctrl-list idiag-socket-details nf-ct-add nf-ct-list nf-exp-add nf-exp-delete nf-exp-list nf-log nf-monitor nf-queue nl-addr-add nl-addr-delete nl-addr-list nl-class-add nl-class-delete nl-classid-lookup nl-class-list nl-cls-add nl-cls-delete nl-cls-list nl-fib-lookup nl-link-enslave nl-link-ifindex2name nl-link-list nl-link-name2ifindex nl-link-release nl-link-set nl-link-stats nl-list-caches nl-list-sockets nl-monitor nl-neigh-add nl-neigh-delete nl-neigh-list nl-neightbl-list nl-pktloc-lookup nl-qdisc-add nl-qdisc-delete nl-qdisc-list nl-route-add nl-route-delete nl-route-get nl-route-list nl-rule-list nl-tctree-list nl-util-addr '/home/jerix/gluon/lede/build_dir/target-mips_24kc_musl-1.1.16/libnl-3.2.29/ipkg-install/usr/bin'
[... cut ...]
mv -f ../common/.deps/joold-file.Tpo ../common/.deps/joold-file.Po
mips-openwrt-linux-musl-gcc -DPACKAGE_NAME=\"Jool\" -DPACKAGE_TARNAME=\"jool\" -DPACKAGE_VERSION=\"3.5.2\" -DPACKAGE_STRING=\"Jool\ 3.5.2\" -DPACKAGE_BUGREPORT=\"jool@nic.mx\" -DPACKAGE_URL=\"\" -DPACKAGE=\"jool\" -DVERSION=\"3.5.2\" -DSTDC_HEADERS=1 -DHAVE_SYS_TYPES_H=1 -DHAVE_SYS_STAT_H=1 -DHAVE_STDLIB_H=1 -DHAVE_STRING_H=1 -DHAVE_MEMORY_H=1 -DHAVE_STRINGS_H=1 -DHAVE_INTTYPES_H=1 -DHAVE_STDINT_H=1 -DHAVE_UNISTD_H=1 -DHAVE_ARPA_INET_H=1 -DHAVE_STDLIB_H=1 -DHAVE_STRING_H=1 -DHAVE_UNISTD_H=1 -

jerix777, 2017-07-16:
Libpcap-1.8.1.tar.gz seems missing from openwrt sources repositories. We have only version 1.5.3.

NeoRaider, 2017-07-16:
```
--2017-07-16 07:09:09--  http://www.tcpdump.org/release/libpcap-1.8.1.tar.gz
Resolving www.tcpdump.org (www.tcpdump.org)... failed: Connection timed out.
wget: unable to resolve host address 'www.tcpdump.org'
Download failed.
```

It seems you had connectivity problems: both www.tcpdump.org and the first fallback sources.lede-project.org could not be resolved correctly. The "not found" messages for the openwrt servers are completely normal and expected: they are tried as last resort fallback for all downloads, but most files have not been mirrored on them.

jerix777, 2017-07-16:
ok thank. But why does the build can't resume ? it rebuild all process from begining. So it take a while :-1:

NeoRaider, 2017-07-16:
As long as you don't `make clean` or anything similar, the build will continue where it left off. Finding out what has already been done may take a short while (and print a lot of messages with `V=s`).

jerix777, 2017-07-16:
ok thank @NeoRaider

christf, 2017-08-03:
so is this resolved then?

jerix777, 2017-08-03:
@christf Yes problem where solved. I'm now on another side of my work plan.

Thanks to all helpers ! specilly to @NeoRaider that I give 5-Thanks hands

closed 2017-08-03 by jerix777