freifunk-gluon/gluon #1193: v2017.1.1: Resets configuration of OpenMesh devices during autoupdate

ecsv, 2017-07-19:
Just noticed that v2017.1.1 seems to remove my configuration on each autoupdate run on my OpenMesh OM2P-HSv3. This seems to be a regression compared to v2016.2.6. I've tested the update on an Ubiquiti Nanostation Loco M2 and didn't experience this kind of problem.

[screenlog.txt](https://github.com/freifunk-gluon/gluon/files/1158397/screenlog.txt)

The log shows the run of the autoupdate, three(?) reboots and then the device with its default configuration.

I find it a little bit odd that the devices now reboots three times and that the reboots with the new firmware image happen before the jffs2 is ready.

Configuration changes involved are:

 * Name was set
 * lat/lon was set
 * download and upload limits were set (5000/1000)

I've also recorded the upgrade with 2016.2.6

[screenlog-2016.2.6.txt](https://github.com/freifunk-gluon/gluon/files/1158486/screenlog-2016.2.6.txt)

Here we see only one reboot. Could this be related to your staged upgrade patches? Does target/linux/ar71xx/base-files/lib/upgrade/openmesh.sh must now do something special to get the config restore during upgrade working again?

The instructions how the image were build can be found under https://github.com/FreifunkVogtland/site-ffv

ecsv, 2017-07-19:
Just tested the upgrade again with LEDE 65eec8bd5f6337956b972d07fde49eb5db9cb4a0 (which is used by gluon 2017.1.1). The sysupgrade works perfectly fine with it. The config is restored after the jffs2 was created + mounted (which doesn't happen with gluon).

----

I've then tried to upgrade to the most recent v2017.1.x version and it still showed the same symptoms.

----

Last step was to apply the LEDE patches from gluon 2017.1.1 on LEDE 65eec8bd5f6337956b972d07fde49eb5db9cb4a0 and test with this image. The sysupgrade worked perfectly fine and the configuration was correctly restored

NeoRaider, 2017-07-19:
Hmm, the reboot is probably caused by `setup_mode.skip=true` (it should only reboot once though...)

It's possible that the configuration loss is caused by missing EOFS markers in the upgrade images; in this case this would have been solved in https://git.lede-project.org/f6907dcc7968a81ebbe75b630dde919468cc8025 , so the fix will reach Gluon soon.

ecsv, 2017-07-19:
Looks like the problem can still be recreated with LEDE 65eec8bd5f6337956b972d07fde49eb5db9cb4a0 when using following config (based on the config generated by gluon):

    CONFIG_TARGET_ar71xx=y
    CONFIG_TARGET_ar71xx_generic=y
    CONFIG_TARGET_MULTI_PROFILE=y
    CONFIG_TARGET_DEVICE_ar71xx_generic_DEVICE_OM2P=y
    CONFIG_TARGET_DEVICE_PACKAGES_ar71xx_generic_DEVICE_OM2P=" "
    CONFIG_BUSYBOX_CUSTOM=y
    CONFIG_TARGET_PER_DEVICE_ROOTFS=y

Problem seems to be that the image contains a init file with following content:

    #!/bin/sh
    # Copyright (C) 2006 OpenWrt.org
    export INITRAMFS=1
    exec /sbin/init

And afaik, the INITRAMFS=1 should disable the restore of the config data. Disabling CONFIG_TARGET_PER_DEVICE_ROOTFS=y seems to work around this problem

ecsv, 2017-07-19:
@NeoRaider thanks for the link. The commit you've mentioned will rebuild the root and remove the /init file by doing that. I've therefore tested it here and can confirm that this is now working perfectly fine with the commit on LEDE.

NeoRaider, 2017-07-19:
Thanks for testing, updated to the latest lede-17.01.

closed 2017-07-19 by NeoRaider